UnconnectedBedna/shrink-backup #14: Backup fails when system is not on first mmcblk device

nyanmisaka, 2023-11-01:
**Describe the bug**

Hi! I have an arm64 SBC with onboard eMMC running [ubuntu-rockchip 22.04.3 LTS](https://github.com/Joshua-Riek/ubuntu-rockchip).
For the convenience of testing kernel driver, I am currently using SD card instead of the eMMC.

I happened to find this project can be an alternative to the `rpi-clone` when I move to other SBCs or system. But when I tried to backup the system on SD card (mmcblk1), the script **did not skip unmounted empty disks** (mmcblk0/eMMC), which caused `dd` and `sfdisk` to fail. After I manually modify the script to specify `HD_DEV_PATH="/dev/mmcblk1"`, it works.

```
# main branch

2023-11-01 12:46:19 [INFO] - Using dd to create bootsector, bs=512 count=255
------------------------------------------------------------------------------
2023-11-01 12:46:19 [DEBUG] - Running: dd bs=512 count=255 if=/dev/mmcblk0
/dev/mmcblk0boot0
/dev/mmcblk0boot1
/dev/mmcblk1 of=/mnt/sda/backup.img conv=noerror,sync status=progress
dd: failed to open '/dev/mmcblk0'$'
''/dev/mmcblk0boot0'$'
''/dev/mmcblk0boot1'$'
''/dev/mmcblk1': No such file or directory
------------------------------------------------------------------------------
2023-11-01 12:46:21 [INFO] - Using truncate to resize img file to 40211MB
2023-11-01 12:46:21 [DEBUG] - Running: truncate --size=42164379136 /mnt/sda/backup.img
2023-11-01 12:46:22 [INFO] - Using losetup -P to loop /mnt/sda/backup.img
2023-11-01 12:46:22 [DEBUG] - Running: losetup -P /dev/loop0 /mnt/sda/backup.img
2023-11-01 12:46:23 [INFO] - Using sfdisk to remove root partition
2023-11-01 12:46:23 [DEBUG] - 1 partition detected, setting LOOP0=/dev/loop0p1
------------------------------------------------------------------------------
2023-11-01 12:46:23 [ERROR] - SFDISK FAILED:
sfdisk: /dev/loop0: partition 1: failed to delete
------------------------------------------------------------------------------
2023-11-01 12:46:23 [ERROR] - Cleanup function called with non zero exit code: exit 1
2023-11-01 12:46:23 [DEBUG] - Removing loop0 in cleanup function: losetup -d /dev/loop0
2023-11-01 12:46:23 [INFO] - Elapsed time: 00.07
2023-11-01 12:46:23 [DEBUG] - Exiting script

# testing branch

2023-11-01 15:24:11 [INFO]  - Using dd to create bootsector
------------------------------------------------------------------------------
2023-11-01 15:24:11 [DEBUG] - Running: dd bs=512 count=255 if=/dev/mmcblk0
/dev/mmcblk0boot0
/dev/mmcblk0boot1
/dev/mmcblk1 of=/mnt/sda/backup1.img conv=noerror,sync status=progress
dd: failed to open '/dev/mmcblk0'$'
''/dev/mmcblk0boot0'$'
''/dev/mmcblk0boot1'$'
''/dev/mmcblk1': No such file or directory
------------------------------------------------------------------------------
2023-11-01 15:24:13 [INFO]  - Using truncate to resize img file' to 40046MB
2023-11-01 15:24:13 [DEBUG] - Running: truncate --size=41991589376 /mnt/sda/backup1.img
2023-11-01 15:24:14 [INFO]  - Looping img file
2023-11-01 15:24:14 [DEBUG] - Running function: do_loop
2023-11-01 15:24:15 [DEBUG] - Running: losetup -P /dev/loop0 /mnt/sda/backup1.img
2023-11-01 15:24:16 [INFO]  - Using sfdisk to remove root partition
2023-11-01 15:24:16 [DEBUG] - Running: sfdisk --delete -f /dev/loop0
------------------------------------------------------------------------------
2023-11-01 15:24:16 [ERROR] - SFDISK FAILED:
sfdisk: failed to parse partition number: ''
------------------------------------------------------------------------------
2023-11-01 15:24:16 [ERROR] - Cleanup function called with non zero exit code: exit 1
2023-11-01 15:24:17 [DEBUG] - Removing loop0 in cleanup function: losetup -d /dev/loop0
2023-11-01 15:24:17 [INFO]  - Elapsed time: 00.08
2023-11-01 15:24:17 [DEBUG] - Exiting script
```

**Harware (please complete the following information):**
 - OS: Ubuntu (preinstalled image)
 - Version: 22.04.3 LTS aarch64

**Additional context**
[shrink-backup.log.txt](https://github.com/UnconnectedBedna/shrink-backup/files/13224226/shrink-backup.log.txt)

```console
$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0     31:0    0    16M  0 disk <= onboard SPI flash, not used
mmcblk0      179:0    0 230.5G  0 disk <= onboard eMMC flash, no system, not mounted
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
mmcblk1      179:96   0 238.3G  0 disk <= SD card, system
├─mmcblk1p1  179:97   0   512M  0 part /boot/firmware
└─mmcblk1p2  179:98   0 237.8G  0 part /

$ df -h
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.6G  3.4M  1.6G   1% /run
/dev/mmcblk1p2  234G   37G  188G  17% /
tmpfs           7.8G     0  7.8G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
/dev/mmcblk1p1  512M  101M  412M  20% /boot/firmware
tmpfs           1.6G   76K  1.6G   1% /run/user/129
tmpfs           1.6G   68K  1.6G   1% /run/user/1000

$ fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mtdblock0: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mmcblk0: 230.47 GiB, 247463936000 bytes, 483328000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mmcblk1: 238.3 GiB, 255869321216 bytes, 499744768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 33EC80F7-D4F2-4D42-A2EE-964C3CBCCFF6

Device           Start       End   Sectors   Size Type
/dev/mmcblk1p1   32768   1081343   1048576   512M Linux extended boot
/dev/mmcblk1p2 1081344 499744734 498663391 237.8G Linux filesystem
```

UnconnectedBedna, 2023-11-01:
Hello and thank you for the report! <3

I think I see what is happening there.

```2023-11-01 12:46:19 [DEBUG] - Running: dd bs=512 count=255 if=/dev/mmcblk0
/dev/mmcblk0boot0
/dev/mmcblk0boot1
/dev/mmcblk1 of=/mnt/sda/backup.img conv=noerror,sync status=progress
dd: failed to open '/dev/mmcblk0'$'
''/dev/mmcblk0boot0'$'
''/dev/mmcblk0boot1'$'
''/dev/mmcblk1': No such file or directory```

The variable seems to contain more than one device, the "running dd...." should be on one line, but the variables seems to contain more than just the device path.
I will look into this in the coming days, I am insanely busy atm but put this high on the prio list.

UnconnectedBedna, 2023-11-01:
I took a quick look during my lunch break and made changes to the testing branch, I _think_ it is fixed, but I sadly have no hardware to test this, might I ask you to help me out? <3

UnconnectedBedna, 2023-11-01:
I found this: https://forums.justlinux.com/showthread.php?153881-3TB-hard-disk-used-as-external-USB-connection-or-internal-Sata-connection

It seems sfdisk has trouble with the blocksize sometimes.
Lets try with parted instead (when/if you have time) :)

Testing branch updated.

UnconnectedBedna, 2023-11-01:
I did some testing on the hardware I have available, and parted did not want to play nice with the --script option in combination with rm and the partition.
Well, how about this line then???
printf 'Ignore\n'$IMG_ROOT_PARTN | parted $LOOP rm $IMG_ROOT_PARTN ---pretend-input-tty

It works here, hopefully it will also solve your problem. :)

UnconnectedBedna, 2023-11-05:
I am going to mark this as done/invalid, simply because op left me hanging with a report I have no possibility to test. Pretty sad.

The first part of the issue is fixed, the selection of root and boot WAS bugged was fixed pretty fast.

About the second "bug" involving sfdisk.
I have a tiny feeling there was never anything wrong with the img file in the first place but rather the errors was cloned from source filesystem, but I have no way of testing that. Spent WAY to many hours trying to "fix" this. :(

I have been holding the next release because of this, does not feel good.
So the solution is to mark it as wontfix and move on, I cant wait forever.
There will be an update in stable in the coming week. :)

jsm174, 2024-01-23:
I think I'm having the same issue. Just adding my info if someday there ever is a fix.

I made the mistake of using a 128gb card on a armbian based system that really only takes up about 10gb.

```
orangepi5:shrink-backup:% cat /etc/os-release                                                                                                                                                             PRETTY_NAME="Ubuntu 22.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.2 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

```
2024-01-23 07:56:37 [INFO]  - Debugging requested, writing to log file ./shrink-backup.log
2024-01-23 07:56:37 [INFO]  - ext4 root filesystem detected
2024-01-23 07:56:37 [DEBUG] - FSTYPE=ext4
2024-01-23 07:56:37 [DEBUG] - Update existing img file, UPDATE=false
2024-01-23 07:56:37 [DEBUG] - Requesting size from resize2fs, RESIZE2FS_RUN=true
2024-01-23 07:56:37 [DEBUG] - Prompt for user confirmation, PROMPTS=true
2024-01-23 07:56:37 [DEBUG] - Auto expansion, AUTOEXPAND=true
2024-01-23 07:56:37 [INFO]  - -a selected by user
2024-01-23 07:56:37 [DEBUG] - ADDED_SPACE=0
2024-01-23 07:56:37 [INFO]  - Executing make_img function
------------------------------------------------------------------------------
2024-01-23 07:56:37 [INFO]  - Running function: get_dev_variables
2024-01-23 07:56:37 [DEBUG] - LOCAL_DEV_PTUUID=1303ea25-a12a-7546-9cd5-d81c74ef55fe | LOCAL_DEV_PATH=/dev/mmcblk1
2024-01-23 07:56:37 [INFO]  - Separate boot partition detected
2024-01-23 07:56:37 [DEBUG] - LOCAL_DEV_BOOT_PATH=/dev/mmcblk1p1 | LOCAL_DEV_ROOT_PATH=/dev/mmcblk1p2
2024-01-23 07:56:37 [DEBUG] - LOCAL_ROOT_PARTN=2
2024-01-23 07:56:37 [INFO]  - Calculating size for dd to cover bootsector in blocks (512B block size) and adding 256 blocks to overlap into root (only used in img creation)
2024-01-23 07:56:37 [DEBUG] - Running: fdisk -lo start /dev/mmcblk1 | tail -1
2024-01-23 07:56:38 [DEBUG] - LOCAL_ROOT_START=557056 Blocks | LOCAL_BOOTSECTOR=557055 Blocks | LOCAL_DDBOOTSECTOR=557311 Blocks
2024-01-23 07:56:38 [DEBUG] - LOCAL_ROOT_START=285212672 Bytes | LOCAL_BOOTSECTOR=285212160 Bytes
2024-01-23 07:56:38 [DEBUG] - LOCAL_RESIZE2FS_MIN=9032294400 Bytes
2024-01-23 07:56:38 [INFO]  - Setting TOTAL (space needed for files on root) to size calculated by resize2fs
2024-01-23 07:56:38 [DEBUG] - TOTAL=9032294400 Bytes
2024-01-23 07:56:38 [INFO]  - Calculating .img file size by adding LOCAL_BOOTSECTOR to TOTAL (only used in img creation)
2024-01-23 07:56:38 [DEBUG] - TRUNCATE_TOTAL=9317506560 Bytes
2024-01-23 07:56:38 [INFO]  - Running function: get_shared_variables
2024-01-23 07:56:38 [DEBUG] - LOOP=/dev/loop0
2024-01-23 07:56:38 [INFO]  - Separate boot partition detected
2024-01-23 07:56:38 [DEBUG] - Running: cat /etc/fstab | grep '/boot' | awk '{print $2}'
2024-01-23 07:56:38 [DEBUG] - BOOT_PATH=/boot | IMG_DEV_BOOT_PATH=/dev/loop0p1 | IMG_DEV_ROOT_PATH=/dev/loop0p2
2024-01-23 07:56:39 [INFO]  - Do you want to continue? [y/n]
2024-01-23 07:56:39 [INFO]  - Y or y pressed to confirm
------------------------------------------------------------------------------
2024-01-23 07:56:39 [WARNING] - /media/mpf/a83cc37b-21d6-4ed5-9b0c-304cba9dba18/mpf_small.img ALREADY EXISTS!
2024-01-23 07:56:40 [WARNING] - Do you want to overwrite? [y/n]
2024-01-23 07:56:40 [WARNING] - Overwrite confirmed by user
2024-01-23 07:56:40 [WARNING] - Removing: /media/mpf/a83cc37b-21d6-4ed5-9b0c-304cba9dba18/mpf_small.img
2024-01-23 07:56:43 [INFO]  - Using dd to create bootsector
------------------------------------------------------------------------------
2024-01-23 07:56:47 [DEBUG] - Running: dd bs=512 count=557311 if=/dev/mmcblk1 of=/media/mpf/a83cc37b-21d6-4ed5-9b0c-304cba9dba18/mpf_small.img conv=noerror,sync status=progress
557311+0 records in
557311+0 records out
285343232 bytes (285 MB, 272 MiB) copied, 4.79493 s, 59.5 MB/s
------------------------------------------------------------------------------
2024-01-23 07:56:49 [INFO]  - Using truncate to resize img file to 8885MB
2024-01-23 07:56:49 [DEBUG] - Running: truncate --size=9317506560 /media/mpf/a83cc37b-21d6-4ed5-9b0c-304cba9dba18/mpf_small.img
2024-01-23 07:56:50 [DEBUG] - Running function: do_loop
2024-01-23 07:56:51 [DEBUG] - Running: losetup -P /dev/loop0 /media/mpf/a83cc37b-21d6-4ed5-9b0c-304cba9dba18/mpf_small.img
2024-01-23 07:56:53 [INFO]  - Using sfdisk to remove root partition
2024-01-23 07:56:53 [DEBUG] - Running: sfdisk --delete -f /dev/loop0 2
------------------------------------------------------------------------------
2024-01-23 07:56:53 [ERROR] - SFDISK FAILED:
GPT PMBR size mismatch (250347519 != 18198254) will be corrected by write.
sfdisk: /dev/loop0: partition 2: failed to delete
------------------------------------------------------------------------------
2024-01-23 07:56:53 [ERROR] - Cleanup function called with non zero exit code: exit 1
2024-01-23 07:56:53 [DEBUG] - Removing loop in cleanup function: losetup -d /dev/loop0
2024-01-23 07:56:53 [INFO]  - Elapsed time: 00.16
2024-01-23 07:56:53 [DEBUG] - Exiting script
```

UnconnectedBedna, 2024-01-23:
Thank you so much for reporting this!

The problem with the initial post here was that I could not confirm certain things, more specifically, that the error was not already present when creating the backup.

Please confirm this by running `sudo sfdisk /dev/????` (on your root device on your system) to make sure this is not the case before we go forward.
Please read this for info: https://arstech.net/gpt-pmbr-size-mismatch/

Because if the error is there on the filesystem you are trying to back up, it will also show on the img file you create, and I do not run any checks on the system you back up, only the img you back up to.

I am very keen on trying to figure this out, but since I can not reproduce the error... Maybe you can help? <3

jsm174, 2024-01-23:
I will get right on it. I'm putting the image back on a fresh sd card, and will report back shortly.

jsm174, 2024-01-23:
Okay, so the OPI 5, is running the 128gb image. I have a 250gb USB SSD plugged in where I could write the "shrinked" image to.

Here is the output of `sfdisk`:

```
sudo sfdisk /dev/mmcblk1                                                                                                                               <main ✗>
[sudo] password for mpf:

Welcome to sfdisk (util-linux 2.37.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Checking that no-one is using this disk right now ... FAILED

This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.

sfdisk: Use the --force flag to overrule all checks.
```

Here is the output of `parted -l`:

```
parted -l                                                                                                                               <main ✗>
Model: PNY CS90 0 240GB SSD (scsi)
Disk /dev/sda: 240GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End    Size   Type     File system  Flags
 1      1049kB  240GB  240GB  primary  ntfs

Model: Unknown (unknown)
Disk /dev/zram1: 52.4MB
Sector size (logical/physical): 4096B/4096B
Partition Table: loop
Disk Flags:

Number  Start  End     Size    File system  Flags
 1      0.00B  52.4MB  52.4MB  ext2

Error: /dev/mtdblock0: unrecognised disk label
Model: Unknown (unknown)
Disk /dev/mtdblock0: 16.8MB
Sector size (logical/physical): 512B/512B
Partition Table: unknown
Disk Flags:

Model: SD GD2S5 (sd/mmc)
Disk /dev/mmcblk1: 128GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start   End    Size   File system  Name    Flags
 1      16.8MB  285MB  268MB  fat16        bootfs  bls_boot
 2      285MB   127GB  127GB  ext4

Model: Unknown (unknown)
Disk /dev/zram0: 4031MB
Sector size (logical/physical): 4096B/4096B
Partition Table: loop
Disk Flags:

Number  Start  End     Size    File system     Flags
 1      0.00B  4031MB  4031MB  linux-swap(v1)
 ```
 
Do you want me to force the `sfdisk`?

UnconnectedBedna, 2024-01-23:
Just to be clear, you are running the `sfdisk` command towards the filesystem you are BACKING UP correct? Not the filesystem where you saved the img file?

No, don't force it, but you might want to umount the drive and try again.

But it does not look like there is any errors from `parted` at least, but if you could redo the `sfdisk` without it mounted that would be awesome!

The important part is the tests are done on the filesystem you are backing up FROM.
Could be a bit tricky if you only own one linux device, otherwise, if it is a sd-card and you have another linux system, just insert the sd-card without mounting it and run the `sfdisk` command there instead. :)

BUT, I do see a few things that _might_ be connected:
- Your backup drive (sda) is in ntfs AND in DOS, should probably be in GPT. NTFS is not great to use on linux either, the drivers are closed source. If you want to share a drive with windows, make it exfat instead (those drivers are open source). But the DOS instead of GPT might be trouble, not sure. _Should_ not matter, I use a samba mount for my images, so a ntfs _should_ be fine, but you never know.
- You mmcblk contains a boot partition. I have not reinstalled armbian in a while, and I only have an opiPC2 (way older) and all armbians (orangepi, bananapi, rpi etc.) I have used has always used U-boot and MBR partitioning, but that could be different since you are running way newer orangepi hardware. Just saw it thought it might be worth mentioning.

Let me know if you are able to figure all that out. Ask questions if needed, I am VERY GRATEFUL for you helping me out here! <3

jsm174, 2024-01-23:
Just to add some context:

- A year ago I helped getting Mission Pinball Framework running on Orange Pi 5 to power a Cuphead pinball machine
- I immediately made an image of the drive and stored it on a NAS
- We need to test some updates and not take a chance at ruining the existing drive

- I restored the 128gb image to a 128gb sd card, put it in my Orange Pi 5 and was successfully able to boot
- Now I need to shrink and image the 128gb drive so I can realistically send it to another developer for testing

- I've tried Pishrink countless times, with no luck (image shrinks but is not bootable), and then stumbled onto your awesome project

- I'm not a linux expert, but I didn't think you could image a drive while running from the drive. But looking at the readme and wiki, I'm assuming I do run shrink-backup directly on the system?  If not, do I mount the drive in another computer running ubuntu, and run shrink-backup there? 

- I only added the usb hard drive as a place to store the saved image

UnconnectedBedna, 2024-01-23:
> I'm not a linux expert, but I didn't think you could image a drive while running from the drive. But looking at the readme and wiki, I'm assuming I do run shrink-backup directly on the system? If not, do I mount the drive in another computer running ubuntu, and run shrink-backup there?

You are correct in the assumption that you should avoid it if possible, but to be honest, the backups I run fail extremely seldom, but you are correct, it IS a risk backing up from a running system.
The script is intended to be ran on a "live" system.

> Now I need to shrink and re-image the 128gb drive so I can realistically send it to another developer for testing

To be honest, if you have that file, and it boots, have you tried compressing it? Even if an img file is big AF, if it contains lots of "empty space" the data should be compressable pretty efficiently.

Maybe I should not have mentioned the boot partition, I was just a bit surprised when I saw it, but since you CAN boot from the fill img if you "burn" it to something, the filesystem is "ok". The question is, is it ok at a block level, or was the boot process put together with tape and glue. I can sadly not answer that for you.

I DO have a sollution I _think_ would work, deleteing the partition with parted instead. The problem with that is that it made Raspberry pi os complain about this exact thing instead, so I am a bit reluctant to inplement that in the script, but if you want to I can give you the lines you have to change and what command to change with if you want to try?

jsm174, 2024-01-23:
I'm willing to try anything. I have an image so I have nothing to lose 😄. I'm also a dev, so whatever you need me to change, I'm good. 

Btw, one thing to mention, another reason I wanted a smaller image, regardless of zipping, is sometimes a 128gb sd card isn't really 128gb. We've found some drives that fudge the numbers, ie 127gb. So then etcher won't let you write it.

UnconnectedBedna, 2024-01-23:
Noice! Then I don't need to hold your hand editing it. :D

I am referencing shrink-backup v0.9.4 here.
Line 746
```diff
-  # Remove partition
-  echo '## Removing root partition...'
-  sleep 1
-  debug 'INFO' 'Using sfdisk to remove root partition'
-  debug 'DEBUG' "Running: sfdisk --delete -f $LOOP $LOCAL_ROOT_PARTN"
-  #parted -s "$LOOP" rm "$IMG_ROOT_PARTN" # does not work, still asking for user confirmation even though --script is used
-  #debug 'INFO' 'Using parted to remove root partition'
-  #debug 'DEBUG' "Running: printf 'Ignore\\\n'$LOCAL_ROOT_PARTN | parted $LOOP rm $LOCAL_ROOT_PARTN ---pretend-input-tty"
-  if ! output=$(sfdisk --delete -f "$LOOP" "$LOCAL_ROOT_PARTN" 2>&1); then # seems to fail if img size is very big
-  #if ! output=$(printf 'Ignore\n'$LOCAL_ROOT_PARTN | parted $LOOP rm $LOCAL_ROOT_PARTN ---pretend-input-tty 2>&1); then # fails on raspberry pi os
-    echo -e "$output\n## SFDISK FAILED!!!"
-    #echo -e "$output\n## PARTED FAILED!!!"
-    debug 'BREAK'
-    debug 'ERROR' "SFDISK FAILED:\n$output\n------------------------------------------------------------------------------"
-    #debug 'ERROR' "PARTED FAILED:\n$output\n------------------------------------------------------------------------------"
-    exit 1
-  fi
-  sleep 1
-----------------------------------------------------------------
+  # Remove partition
+  echo '## Removing root partition...'
+  sleep 1
+  #debug 'INFO' 'Using sfdisk to remove root partition'
+  #debug 'DEBUG' "Running: sfdisk --delete -f $LOOP $LOCAL_ROOT_PARTN"
+  #parted -s "$LOOP" rm "$IMG_ROOT_PARTN" # does not work, still asking for user confirmation even though --script is used
+  debug 'INFO' 'Using parted to remove root partition'
+  debug 'DEBUG' "Running: printf 'Ignore\\\n'$LOCAL_ROOT_PARTN | parted $LOOP rm $LOCAL_ROOT_PARTN ---pretend-input-tty"
+  #if ! output=$(sfdisk --delete -f "$LOOP" "$LOCAL_ROOT_PARTN" 2>&1); then # seems to fail if img size is very big
+  if ! output=$(printf 'Ignore\n'$LOCAL_ROOT_PARTN | parted $LOOP rm $LOCAL_ROOT_PARTN ---pretend-input-tty 2>&1); then # fails on raspberry pi os
+    #echo -e "$output\n## SFDISK FAILED!!!"
+    echo -e "$output\n## PARTED FAILED!!!"
+    debug 'BREAK'
+    #debug 'ERROR' "SFDISK FAILED:\n$output\n------------------------------------------------------------------------------"
+    debug 'ERROR' "PARTED FAILED:\n$output\n------------------------------------------------------------------------------"
+    exit 1
+  fi
+  sleep 1
```

As you can see I was clever enough to keep the code in there because I had a feeling it sooner or later would become an issue for _someone_. :D

Make sure you disable autoexpansion in the script for now (`-e` tag) since it also relies on `sfdisk`, so just to disable it for now and if this works we can try with autoexpansion.

Thank you so much for this! <3

jsm174, 2024-01-23:
```
orangepi5:shrink-backup:% pwd
/media/mpf/test/shrink-backup
orangepi5:shrink-backup:% sudo ./shrink-backup -l -a -e ./mpf_small.img
# ----------------------------------------------------------------------------
# Write to logfile: true
# Resize2fs decide size: true
# Autoexpand filesystem at boot: false
# Use exclude.txt: false
# Bootsector size: 271MB
# Estemated root usage: 8555MB
# Resize2fs decide minimum (root partition): 8583MB
# Total img size: 8855MB
##############################################################################
Do you want to continue? [y/n] y
## Creating bootsector...
279431680 bytes (279 MB, 266 MiB) copied, 47 s, 5.9 MB/s
557311+0 records in
557311+0 records out
285343232 bytes (285 MB, 272 MiB) copied, 47.822 s, 6.0 MB/s
## Resizing img file...
## Looping img file...
## Removing root partition...
Error: Invalid argument during seek for read on /dev/loop0
Retry/Ignore/Cancel? Ignore
Error: The backup GPT table is corrupt, but the primary appears OK, so that will be used.
OK/Cancel? 2
parted: invalid token: 2
OK/Cancel?
## PARTED FAILED!!!
## Cleanup function called with non zero exit code, something went wrong!!!
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.57
```

I think it didn't know how to answer the OK question?

```
##############################################################################
2024-01-23 16:44:40 [INFO]  - -a selected by user
2024-01-23 16:44:40 [DEBUG] - ADDED_SPACE=0
2024-01-23 16:44:40 [INFO]  - Debugging requested, writing to log file shrink-backup.log...
2024-01-23 16:44:40 [INFO]  - Update existing img file, UPDATE=false
2024-01-23 16:44:40 [INFO]  - Requesting size from resize2fs, RESIZE2FS_RUN=true
2024-01-23 16:44:40 [INFO]  - Prompt for user confirmation, PROMPTS=true
2024-01-23 16:44:40 [INFO]  - Auto expansion, AUTOEXPAND=false
2024-01-23 16:44:40 [DEBUG] - Executing make_img function
2024-01-23 16:44:40 [DEBUG] - Running function: get_dev_variables
2024-01-23 16:44:40 [DEBUG] - LOCAL_DEV_PTUUID=1303ea25-a12a-7546-9cd5-d81c74ef55fe | LOCAL_DEV_PATH=/dev/mmcblk1
2024-01-23 16:44:40 [INFO]  - Separate boot partition detected
2024-01-23 16:44:40 [DEBUG] - LOCAL_DEV_ROOT_PATH=/dev/mmcblk1p2 | LOCAL_DEV_BOOT_PATH=/dev/mmcblk1p1
2024-01-23 16:44:40 [DEBUG] - LOCAL_ROOT_PARTN=2
2024-01-23 16:44:40 [INFO]  - Calculating size for dd to cover bootsector in blocks (512B block size) and adding 256 blocks to overlap into root (only used in img creation)
2024-01-23 16:44:40 [DEBUG] - Running: fdisk -lo start /dev/mmcblk1 | tail -1
2024-01-23 16:44:40 [DEBUG] - LOCAL_ROOT_START=557056 Blocks | LOCAL_BOOTSECTOR=557055 Blocks | LOCAL_DDBOOTSECTOR=557311 Blocks
2024-01-23 16:44:40 [DEBUG] - LOCAL_ROOT_START=285212672 Bytes | LOCAL_BOOTSECTOR=285212160 Bytes
2024-01-23 16:44:41 [DEBUG] - LOCAL_RESIZE2FS_MIN=9000660992 Bytes
2024-01-23 16:44:41 [INFO]  - Setting TOTAL (space needed for files on root) to size calculated by resize2fs
2024-01-23 16:44:41 [DEBUG] - TOTAL=9000660992 Bytes
2024-01-23 16:44:41 [INFO]  - Calculating .img file size by adding LOCAL_BOOTSECTOR to TOTAL (only used in img creation)
2024-01-23 16:44:41 [DEBUG] - TRUNCATE_TOTAL=9285873152 Bytes
2024-01-23 16:44:41 [DEBUG] - Running function: get_shared_variables
2024-01-23 16:44:41 [DEBUG] - LOOP=/dev/loop0
2024-01-23 16:44:41 [INFO]  - Separate boot partition detected
2024-01-23 16:44:41 [INFO]  - Fetching boot mount path from fstab
2024-01-23 16:44:41 [DEBUG] - Running: cat /etc/fstab | grep '/boot' | awk '{print $2}'
2024-01-23 16:44:41 [DEBUG] - BOOT_PATH=/boot | IMG_DEV_BOOT_PATH=/dev/loop0p1 | IMG_DEV_ROOT_PATH=/dev/loop0p2
2024-01-23 16:44:42 [INFO]  - Do you want to continue? [y/n]
2024-01-23 16:44:42 [USER_INPUT] - Y or y pressed to confirm
2024-01-23 16:44:43 [INFO]  - Using dd to create bootsector
------------------------------------------------------------------------------
2024-01-23 16:45:31 [DEBUG] - Running: dd bs=512 count=557311 if=/dev/mmcblk1 of=./mpf_small.img conv=noerror,sync status=progress
557311+0 records in
557311+0 records out
285343232 bytes (285 MB, 272 MiB) copied, 47.822 s, 6.0 MB/s
------------------------------------------------------------------------------
2024-01-23 16:45:33 [INFO]  - Using truncate to resize img file' to 8855MB
2024-01-23 16:45:33 [DEBUG] - Running: truncate --size=9285873152 ./mpf_small.img
2024-01-23 16:45:34 [INFO]  - Looping img file
2024-01-23 16:45:34 [DEBUG] - Running function: do_loop
2024-01-23 16:45:35 [DEBUG] - Running: losetup -P /dev/loop0 ./mpf_small.img
2024-01-23 16:45:36 [INFO]  - Using parted to remove root partition
2024-01-23 16:45:36 [DEBUG] - Running: printf 'Ignore\n'2 | parted /dev/loop0 rm 2 ---pretend-input-tty
------------------------------------------------------------------------------
2024-01-23 16:45:37 [ERROR] - PARTED FAILED:
Error: Invalid argument during seek for read on /dev/loop0
^M                                                                          ^M^M                                                                          ^MRetry/Ignore/Cancel? Ignore
Error: The backup GPT table is corrupt, but the primary appears OK, so that will be used.
^M                                                                          ^M^M                                                                          ^MOK/Cancel? 2
parted: invalid token: 2
^M                                                                          ^MOK/Cancel?
------------------------------------------------------------------------------
2024-01-23 16:45:37 [ERROR] - Cleanup function called with non zero exit code: exit 1
2024-01-23 16:45:37 [DEBUG] - Removing loop0 in cleanup function: losetup -d /dev/loop0
2024-01-23 16:45:37 [INFO]  - Elapsed time: 00.57
2024-01-23 16:45:37 [DEBUG] - Exiting script
##############################################################################
```

UnconnectedBedna, 2024-01-23:
> Error: The backup GPT table is corrupt, but the primary appears OK, so that will be used.

I'm getting more and more sure that it is your actual partition table and how it is setup that neither sfdisk nor parted likes.

If you take a look at the partitions we can see that the start is a very strange size, 16,8M into the partition. That does NOT look correct to me on a gpt partition table.
```
Number  Start   End    Size   File system  Name    Flags
 1      16.8MB  285MB  268MB  fat16        bootfs  bls_boot
 2      285MB   127GB  127GB  ext4
```

But I DO see the dd is JUST on the edge of not covering the boot partition, we could try to increase the overlap a bit, but I do not think that would help. But now that I think about it, I might have assumed in the code that  you have a normal start of the partitions and therefore the dd does not actually cover the complete boot partition and overlap the root.

On row 202 the "overlap" on the dd is hardcoded with 256 blocks, increase this to, idk, 2048 or smthn to make dd grab a bigger portion of the disk before reformating root.

`  LOCAL_DDBOOTSECTOR=$(( LOCAL_BOOTSECTOR + 2048 ))`

Try this and maybe, but otherwise, I am afraid the freak partition setup might be very hard to overcome with the script. :(

jsm174, 2024-01-24:
>  LOCAL_DDBOOTSECTOR=$(( LOCAL_BOOTSECTOR + 2048 ))

I did this, but when the script runs, I still get this:

```
Error: The backup GPT table is corrupt, but the primary appears OK, so that will be used.
OK/Cancel? 2
parted: invalid token: 2
```

Is there a way to send "OK"?

> `sudo fdisk -lo start /dev/mmcblk0 | tail -1 | awk '{print $1}'`

```
orangepi5:shrink-backup:% sudo fdisk -lo start /dev/mmcblk0 | tail -1 | awk '{print $1}'    
fdisk: cannot open /dev/mmcblk0: No such file or directory
orangepi5:shrink-backup:% sudo fdisk -lo start /dev/mmcblk1 | tail -1 | awk '{print $1}'
557056
```

> I recall reading something about gpt being a few blocks bigger and that fdisk can't see that, would make sense.
Let's use both parted and fdisk to give us the positions.

```
orangepi5:shrink-backup:% dumpe2fs -h /dev/mmcblk1p1 | grep "Block size" | awk '{print $3}'
dumpe2fs 1.46.5 (30-Dec-2021)
dumpe2fs: Bad magic number in super-block while trying to open /dev/mmcblk1p1
```
```
orangepi5:shrink-backup:% dumpe2fs -h /dev/mmcblk1p2 | grep "Block size" | awk '{print $3}'
dumpe2fs 1.46.5 (30-Dec-2021)
4096
```
```
orangepi5:shrink-backup:% fdisk -l /dev/mmcblk1
Disk /dev/mmcblk1: 119.38 GiB, 128177930240 bytes, 250347520 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 1303EA25-A12A-7546-9CD5-D81C74EF55FE

Device          Start       End   Sectors   Size Type
/dev/mmcblk1p1  32768    557055    524288   256M Linux extended boot
/dev/mmcblk1p2 557056 247824384 247267329 117.9G Linux filesystem
```
```
orangepi5:shrink-backup:% parted /dev/mmcblk1 unit b print                                                                                                       <main ✗>
WARNING: You are not superuser.  Watch out for permissions.
Model: SD GD2S5 (sd/mmc)
Disk /dev/mmcblk1: 128177930240B
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start       End            Size           File system  Name    Flags
 1      16777216B   285212671B     268435456B     fat16        bootfs  bls_boot
 2      285212672B  126886085119B  126600872448B  ext4
 ```

UnconnectedBedna, 2024-01-24:
You could try to use line 464 instead and add the if statement and an `-f`.
ie: `if ! output=$(sfdisk --delete -f "$LOOP" "$IMG_ROOT_PARTN" 2>&1); then`.

Don't forget to comment out the other line.

I have had very few successes with this line even though it clearly states in the manual that:

```
parted - a partition manipulation program
=========================================

      -s, --script
              never prompts for user intervention

      -f, --fix
              automatically answer "fix" to exceptions in script mode
```

Maybe you are lucky. f is fix, not actually force, but you can play around with it, I think you can figure out the line.

Otherwise you can also try this line:
`if ! output=$(printf 'Ignore\n'$IMG_ROOT_PARTN | parted $LOOP rm -s -f $IMG_ROOT_PARTN ---pretend-input-tty 2>&1); then`

But I think I have tried that one in the past, and it does not accept it.
You could also try without the `-s` and/or `-f` in there as well.

You can also maybe squeeze in a "OK" after the newline in the printf (I cant see you screen, but I think the "ignore" is also used before this?), ie something like `printf 'Ignore\nOK\n'$IMG_ROOT_PARTN etc etc etc`

I am a bit at loss here, but the time is also 2 am here, so I will sleep on it and hopefully come up with something we could try if none of above works.

jsm174, 2024-01-24:
Okay, so I'm guessing I should go back to the original version of the script.

I did try this:

`Ignore\nOK\n`

```
## Removing root partition...
Error: Invalid argument during seek for read on /dev/loop0
Retry/Ignore/Cancel? Ignore
Error: The backup GPT table is corrupt, but the primary appears OK, so that will be used.
OK/Cancel? OK
## PARTED FAILED!!!
## Cleanup function called with non zero exit code, something went wrong!!!
```

```
------------------------------------------------------------------------------
2024-01-23 20:48:15 [DEBUG] - Running: dd bs=512 count=559103 if=/dev/mmcblk1 of=./mpf_small.img conv=noerror,sync status=progress
559103+0 records in
559103+0 records out
286260736 bytes (286 MB, 273 MiB) copied, 40.83 s, 7.0 MB/s
------------------------------------------------------------------------------
2024-01-23 20:48:17 [INFO]  - Using truncate to resize img file' to 8855MB
2024-01-23 20:48:17 [DEBUG] - Running: truncate --size=9285987840 ./mpf_small.img
2024-01-23 20:48:18 [INFO]  - Looping img file
2024-01-23 20:48:18 [DEBUG] - Running function: do_loop
2024-01-23 20:48:19 [DEBUG] - Running: losetup -P /dev/loop0 ./mpf_small.img
2024-01-23 20:48:20 [INFO]  - Using parted to remove root partition
2024-01-23 20:48:20 [DEBUG] - Running: printf 'Ignore\n'2 | parted /dev/loop0 rm 2 ---pretend-input-tty
------------------------------------------------------------------------------
2024-01-23 20:48:20 [ERROR] - PARTED FAILED:
Error: Invalid argument during seek for read on /dev/loop0
^M                                                                          ^M^M                                                                          ^MRetry/Ignore/Cancel? Ignore
Error: The backup GPT table is corrupt, but the primary appears OK, so that will be used.
^M                                                                          ^M^M                                                                          ^MOK/Cancel? OK
------------------------------------------------------------------------------
2024-01-23 20:48:20 [ERROR] - Cleanup function called with non zero exit code: exit 1
2024-01-23 20:48:20 [DEBUG] - Removing loop0 in cleanup function: losetup -d /dev/loop0
2024-01-23 20:48:20 [INFO]  - Elapsed time: 00.50
2024-01-23 20:48:20 [DEBUG] - Exiting script
##############################################################################

```

UnconnectedBedna, 2024-01-24:
You know what, remove the if statement and it might actually work in your case.
Would be interesting to know if this is a gpt thing that always will happen or if this is because the start is so far off...

There is probably an error code other than 0 sent so the script immediately fails.

I should really go to bed.

jsm174, 2024-01-24:
Thanks for sticking with this. After allowing it to continue:

```
## Recreating root partition...
Error: Invalid argument during seek for read on /dev/loop0
```

```
2024-01-24 08:44:19 [DEBUG] - Running: parted -s -a none /dev/loop0 unit B mkpart primary ext4 285212672 100%
```

UnconnectedBedna, 2024-01-24:
"Good news" everyone!

I think I have found the cause at least.

I could not reproduce the setup you have with gpt with any of the sbc hardware I own so I installed Manjaro on a usb stick using gpt and when trying to back it up:

```
## Removing root partition...
GPT PMBR size mismatch (125045423 != 22894006) will be corrected by write.
sfdisk: /dev/loop0: partition 2: failed to delete
## SFDISK FAILED!!!
```

I now have something to work with here and will see if I can figure out a way to force delete the partition even though the warning is there.

UnconnectedBedna, 2024-01-24:
I got it working, I am typing this from a restored img of the Manjaro installation. :)
I will not implement this in a hotfix but rather do this update in the next version.
I will implement a check, and if the partition table is GPT another way will be utilized to remove/create/format the ext4 system before backing up because without even trying, I have a feeling other systems like raspberry pi os will not like this.

The problem IS gpt, it does not allow hacky stuff like mbr does, there are more checks that needs to be "ignored" and bypassed, but it DOES work.

I have to succumb to asking the users to install one more program though, `gdisk`, or rather `sgdisk` that is part of the package, at least on arch. Because no matter what I try, using parted does NOT like this.
So you have to install `gdisk` before running the backup. I have no debian based os spinned up right now, but you prob figure it out.

So. I made a gist you can just copy/paste into your `shrink-backup`.
Please test it and come back with results. <3
RUN IT WITHOUT AUTOEXPANSION ENABLED!!!

https://gist.github.com/UnconnectedBedna/8ef639c933a7616d65403025f03e65f9

jsm174, 2024-01-24:
You are my hero! It worked! 

When it booted and I opened gparted it worked! 

I will try with the autoexpansion next!

jsm174, 2024-01-24:
Autoexpansion worked! 

[image]

Truly amazing work! Thank you so much!

UnconnectedBedna, 2024-01-24:
Thank YOU! :heart: 

This is great information!

But there are still 618MiB unallocated at the end... :thinking: 
Maybe it is by your design?

I will leave the issue open until I actually implement the code in the main branch.

Thank you so much and I am happy you got it working for you.
And good luck with your pinball endeavors! :video_game:

UnconnectedBedna, 2024-01-28:
This should now be in main branch so I am closing this.
If additional bugs appear, feel free to report them!

closed 2024-01-28 by UnconnectedBedna