UnconnectedBedna/shrink-backup #25: Script fails if Snap is installed

FluffyPanda69, 2024-01-10:
**Describe the bug**
The script simply assumes `/dev/loop0` is unused, which is not always the case, particularly on systems where `snap` packages are installed.
```shell
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0  93.3M  1 loop /snap/core/16204
loop2         7:2    0  59.7M  1 loop /snap/core20/2107
loop3         7:3    0  43.2M  1 loop /snap/certbot/3567
sda           8:0    0 447.1G  0 disk 
└─sda1        8:1    0 447.1G  0 part /mnt/ssd
mmcblk0     179:0    0 119.1G  0 disk 
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0 118.6G  0 part /

``` 
Changing the `LOOP` variable to something like `/dev/loop999` along with setting `IMG_DEV_BOOT_PATH`, `IMG_DEV_ROOT_PATH` and cleanup parameters accordingly does fix the problem for me, but perhaps you'd like to implement a more elegant solution.

**Harware (please complete the following information):**
 - OS: RaspberryPiOS/Debian
 - Version: 12 aarch64

**Additional context**
[shrink-backup.log](https://github.com/UnconnectedBedna/shrink-backup/files/13888109/shrink-backup.log)

UnconnectedBedna, 2024-01-10:
Thank you fore creating an issue! :)

This has actually been on my mind since I created the script, it is just that nobody has ever complained about it so I never dealt with it, the loop is hardcoded.

Let me look if I can just do a normal losetup that sets the next free loop instead and set that to the variable.
I will do these changes on the testing branch and will let you know when they are ready for you.

UnconnectedBedna, 2024-01-10:
So, I still recommend you do the changes yourself until I have pushed them to stable instead of using testing as of now (there might be bugs since I mess around with btrfs there)

Line 52:
```diff
-    if losetup /dev/loop0 &>/dev/null; then
+    if losetup "$LOOP" &>/dev/null; then
-      losetup -d /dev/loop0
+      losetup -d "$LOOP"
-      debug 'DEBUG' 'Removing loop0 in cleanup function: losetup -d /dev/loop0'
+      debug 'DEBUG' "Removing loop in cleanup function: losetup -d $LOOP"
    fi
-    if losetup /dev/loop1 &>/dev/null; then
-      losetup -d /dev/loop1
-      debug 'DEBUG' 'Removing loop1 in cleanup function: losetup -d /dev/loop1'
-    fi
```

Line 267:
```diff
-  LOOP='/dev/loop0'
+  LOOP=$(losetup -l)
  debug 'DEBUG' "LOOP=$LOOP"

  if [ $(lsblk | grep -c 'boot') -ne 0 ]; then
    debug 'INFO' 'Separate boot partition detected'
    debug 'DEBUG' "Running: cat /etc/fstab | grep '/boot' | awk '{print \$2}'"
    BOOT_PATH=$(cat /etc/fstab | grep '/boot' | awk '{print $2}')
-    IMG_DEV_BOOT_PATH='/dev/loop0p1'
+    IMG_DEV_BOOT_PATH="${LOOP}p1"
-    IMG_DEV_ROOT_PATH='/dev/loop0p2'
+    IMG_DEV_ROOT_PATH="${LOOP}p2"
    debug 'DEBUG' "BOOT_PATH=$BOOT_PATH | IMG_DEV_BOOT_PATH=$IMG_DEV_BOOT_PATH | IMG_DEV_ROOT_PATH=$IMG_DEV_ROOT_PATH"
  else
    debug 'INFO' 'No boot partition detected'
-    IMG_DEV_ROOT_PATH='/dev/loop0p1'
+    IMG_DEV_ROOT_PATH="${LOOP}p1"
    debug 'DEBUG' "IMG_DEV_ROOT_PATH=$IMG_DEV_ROOT_PATH"
  fi
```

Please let me know if it works. :)

FluffyPanda69, 2024-01-10:
Thanks for the quick answer and fix, this works perfectly!

closed 2024-01-10 by FluffyPanda69