UnconnectedBedna/shrink-backup #34: Run shrink-backup stops with error when running within Webmin

bmw-biker, 2024-06-08:
Hi

First of all I appretiate your script very much. It's working like a charm and easy to handle.

From SSH (XShell) shrink-backup does work totally fine, high speed, from SD card to a /mnt mounted network drive.
I use the script like click and go with the -y feature.

Now, on a Raspberry I like to use Debian Lite (without desktop) but together with Webmin web UI for some system maintenance.
Within Webmin I e.g. can configure some user commands to be started on button click.

Configuring the same command as Webmin user command shrink-backup fails with `error 256` / last error: `DD TO LOCAL_BOOTSECTOR FAILED`. Although, ~512MB will be written...

I don't think this is the real ERROR because from XShell all works fine. 
Also, there is another error output some lines before which doesn't appear in XShell console: `tee: /dev/tty: No such device or address`.

So, here my question: 
Does shrink-backup need /dev/tty for IO and probably in the Webmin web fake console it is missing ?
If I try `tty` in Webmin it also says `not a tty`.
Unfortunately tty cannot be redirected so easily to e.g. stdout to make output visible in Webmin.
By `mount --bind /dev/null /dev/tty?` it just can be nulled...

Within shrink-backup script I see some `tee /dev/tty` in if clauses where some shrink/backup fault should be catched. In my case it seems that also a missing tty is being catched ?
Would you think you could improve shrink-backup to run in non console environment ?

```
Output from cd /home/pi/shrinkBackup ; sudo ./shrink-backup -a -y /mnt/dsRaspberryBackup/pi/fhem-pi-2-$(date +"%Y%m%d%H%M%S").img ..

dumpe2fs 1.47.0 (5-Feb-2023)
resize2fs 1.47.0 (5-Feb-2023)
##############################################################################
# DISABLE PROMPTS SELECTED (-y), NO WARNINGS ABOUT DELETION!!!
# A backup will be created at /mnt/dsRaspberryBackup/pi/fhem-pi-2-20240608131404.img
# ----------------------------------------------------------------------------
# Write to logfile: false
# Resize2fs decide size: true
# Autoexpand filesystem at boot: true
# Use exclude.txt: false
# Bootsector size: 515MB
# Estemated root usage: 5917MB
# Resize2fs decide minimum (root partition): 7351MB
# Total img size: 7867MB
# PRESS CTRL+C WITHIN 5s TO CANCEL!
##############################################################################
## Creating bootsector...
tee: /dev/tty: No such device or address

45013504 bytes (45 MB, 43 MiB) copied, 1 s, 45.0 MB/s
90249728 bytes (90 MB, 86 MiB) copied, 2 s, 45.1 MB/s
135683584 bytes (136 MB, 129 MiB) copied, 3 s, 45.2 MB/s
181313536 bytes (181 MB, 173 MiB) copied, 4 s, 45.3 MB/s
227771392 bytes (228 MB, 217 MiB) copied, 5 s, 45.6 MB/s
274211328 bytes (274 MB, 262 MiB) copied, 6 s, 45.7 MB/s
320479744 bytes (320 MB, 306 MiB) copied, 7 s, 45.8 MB/s
366617088 bytes (367 MB, 350 MiB) copied, 8 s, 45.8 MB/s
407773696 bytes (408 MB, 389 MiB) copied, 9 s, 45.3 MB/s
454144512 bytes (454 MB, 433 MiB) copied, 10 s, 45.4 MB/s
500572672 bytes (501 MB, 477 MiB) copied, 11 s, 45.5 MB/s
1057023+0 records in
1057023+0 records out
541195776 bytes (541 MB, 516 MiB) copied, 44.9402 s, 12.0 MB/s
## DD TO LOCAL_BOOTSECTOR FAILED!!!
## Cleanup function called with non zero exit code, something went wrong!!!
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.53

Command failed with exit status 256
```

Best regards

UnconnectedBedna, 2024-06-08:
Hello!

It always makes me very glad to see others find use for the script. :slightly_smiling_face: 

You are 100% correct in your assumption and what has to be fixed.
The reason the tty is there is because, well, if you are on console it's pretty nice to see what is going on..
But as you correctly identified, if there is no tty, it fails.
I simply did not think about this. (Might also make it work running with crontab for those who likes that)

Since you seem ok with digging in the code a bit, U would really appreciate if I would not have to commit changes to testing branch but maybe I could ask you to run the tests first?

If not, I will make commits in testing branch and you can pull the updates.
But since the `tee /dev/tty` exists on more places than just for the `dd` it would be nice if you could change the script on your machine directly to test if this is enough first. :slightly_smiling_face: 

On the dd line, line 972 (if you are using the version on main, if you are using the script on testing branch, it is on line 1045)

Make following changes:
```diff
- if ! output=$(dd bs=512 count=$LOCAL_DDBOOTSECTOR if="$LOCAL_DEV_PATH" of="$IMG_FILE" conv=noerror,sync status=progress 2>&1 | tee /dev/tty); then
+ if [ -z /dev/tty ]; then
+   TTY_AVAILABILITY='/dev/null'
+ else
+   TTY_AVAILABILITY='/dev/tty'
+ fi
+ if ! output=$(dd bs=512 count=$LOCAL_DDBOOTSECTOR if="$LOCAL_DEV_PATH" of="$IMG_FILE" conv=noerror,sync status=progress 2>&1 | tee "${TTY_AVAILABILITY}"); then
```

The script will still fail further down, because of other `/dev/tty` redirects.
But if you can add the `-l` tag and provide the logfile here after trying, and we see that the dd goes through correctly, I can pretty easily change the script on all the other instances where `/dev/tty` is used.

Please let me know.
(since you do not care about the console output, you might as well also include the `-z` option so the script runs as fast as possible. :slightly_smiling_face:)

UnconnectedBedna, 2024-06-09:
There, the fix is now in the testing branch and will be pushed to main in the next update (that is hopefully not to far away now).

I tried it and it works BUT you will miss out on what is going on with a few operations, but the log file will still provide all info on what has transpired incl dd and rsync reports. :slightly_smiling_face: 

I will keep this issue open until I push to main branch in case something that I did not think of would come up.

If you are up to it, feel free to test the script on testing branch.

bmw-biker, 2024-06-10:
Hey there...

Thanks so much for your light speed response. Sorry, I only just now was able to do some testing.
You've been very busy since V1.0.0...  (I tried to compare and understand what changed 😄 )
And - it looks very good !

Now even without tty the script runs through without error (at least on one Raspberry, but this issue afterwards).
Even speed more then doubles up from ~11min (from XShell console over local WLAN) to 4:40 (in web UI, without tty).
Of course in web UI I cannot see the files running through - I just have to wait at `## Backing up files...` - so hopefully everything has been copied 🤣 I didn't test a SD card restore yet 😆 
So another finding: At least via network the console traffic is slowing down the script very much.

Thank you sooo much for your very quick solution !!!

Just for info - this is the output which I can see at the web fake console, I used your standard exclude.txt:
```
Output from cd /home/pi/shrink-backup-test ; sudo ./shrink-backup -a -y -z -t -l /mnt/dsRaspberryBackup/pi/proxy-pi-2-$(date +"%Y%m%d%H%M%S").img ..

## Debugging requested, writing to log file ./shrink-backup.log
## Zoom speed requested...
## Scanning filesystem and calculating...
##############################################################################
# DISABLE PROMPTS SELECTED (-y), NO WARNINGS ABOUT DELETION!!!
# A backup will be created at /mnt/dsRaspberryBackup/pi/proxy-pi-2-20240610124014.img
# ext4 filesystem detected on root
# ----------------------------------------------------------------------------
# Write to logfile: true
# Zoom speed requested: true
# Autocalculate img root partition size: true
# Autoexpand filesystem at boot: true
# Use exclude.txt: true
# Bootsector size: 512MiB
# Estemated root usage: 8824MiB
# Auto calculated size (root partition): 10598MiB
# Total img size: 11110MiB
## Creating bootsector...
## Resizing img file...
## Looping img file...
## Removing root partition...
## Recreating root partition...
## Formatting filesystem...
## Checking img filesystem...
rootfs: 11/678608 files (0.0% non-contiguous), 69239/2712074 blocks
## Creating temp directory...
## Mounting img root partition...
## Mounting img boot partition...
## Backing up files...
## Rsync done...
## Please stand by...
## Finalizing filesystem...
rootfs: /lost+found not found.  CREATED.

      210429 inodes used (31.01%, out of 678608)
          77 non-contiguous files (0.0%)
         149 non-contiguous directories (0.1%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 203863/13
     1651464 blocks used (60.89%, out of 2712074)
           0 bad blocks
           1 large file

      190202 regular files
       13561 directories
           0 character device files
           0 block device files
           0 fifos
         319 links
        6656 symbolic links (6544 fast symbolic links)
           0 sockets
------------
      210738 files
[... 172 lines cut ...]
2024-06-10 12:44:53 [DEBUG] - Running: mount /dev/loop0p2 /tmp/backup-Zlm
2024-06-10 12:44:53 [INFO]  - Separate boot partition detected, mounting boot inside root
2024-06-10 12:44:53 [DEBUG] - Running: mount /dev/loop0p1 /tmp/backup-Zlm/boot/firmware
2024-06-10 12:44:53 [INFO]  - Checking OS for autoexpand function
2024-06-10 12:44:53 [INFO]  - Raspberry pi detected
2024-06-10 12:44:53 [INFO]  - Running function: autoexpansion_rpi
2024-06-10 12:44:53 [DEBUG] - Systemd basic.target.wants directory does not exist, running: mkdir /tmp/backup-Zlm/etc/systemd/system/basic.target.wants
2024-06-10 12:44:53 [DEBUG] - Creating systemd unit expand-fs.service
2024-06-10 12:44:53 [DEBUG] - Enabling systemd service by creating symlink: ln -s /etc/systemd/system/expand-fs.service /tmp/backup-Zlm/etc/systemd/system/basic.target.wants/expand-fs.service
2024-06-10 12:44:53 [DEBUG] - Creating expansion script /tmp/backup-Zlm/expand-fs.sh
2024-06-10 12:44:53 [DEBUG] - Making /expand-fs.sh executable
2024-06-10 12:44:53 [INFO]  - Raspberry pi filesystem autoresizing at boot
2024-06-10 12:44:53 [INFO]  - /mnt/dsRaspberryBackup/pi/proxy-pi-2-20240610124014.img is 11110MiB with a root partition of 10598MiB
2024-06-10 12:44:53 [INFO]  - Img file created
2024-06-10 12:44:53 [INFO]  - Backup done
------------------------------------------------------------------------------
2024-06-10 12:44:53 [DEBUG] - Cleanup function called with exit 0
2024-06-10 12:44:54 [DEBUG] - Unmounting boot partition in cleanup function: umount /tmp/backup-Zlm/boot/firmware
2024-06-10 12:44:54 [DEBUG] - Unmounting root partition in cleanup function: umount /tmp/backup-Zlm
2024-06-10 12:44:54 [DEBUG] - Removing loop in cleanup function: losetup -d /dev/loop0
2024-06-10 12:44:54 [DEBUG] - Removing temp directory in cleanup function: rm -rf /tmp/backup-Zlm
2024-06-10 12:44:54 [DEBUG] - Removing temp file in cleanup function: rm /tmp/rsync-j4t
2024-06-10 12:44:54 [INFO]  - Elapsed time: 04.40
2024-06-10 12:44:54 [DEBUG] - Exiting script
```

bmw-biker, 2024-06-10:
Now I have two things left:

- An issue: I also recognized with V1.0.0 (but not with 0.9.4) that one of my Raspberrys gets stuck with an error:
Could we have a look into this as well please ?  Should I open a new issue ?
```
mount: /tmp/backup-m3Y: wrong fs type, bad option, bad superblock on , missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
!! ROOT MOUNT FAILED!!!
```

- And a puzzle 😉 :
At first I did a shrink-backup without exclude.txt file. It seems, the default is without excluding /var/swap, right ?
Now, when I exclude /var/swap the time for copying of course will be shortened - but the resulting file size will stay same.
I have a 2 GB swap file the place of which will be occupied on my storage for nothing 😉 🤣 
This doesn't really matter, but would you like me to open a new feature request, just as nice to have backlog ?

Best regards and have a nice week, Torsten

UnconnectedBedna, 2024-06-10:
Naa, no need to open another issue.
It's a small project, let's do it here.

A log file from the failing rpi would be great.

About the `/var/swap`, I completely missed adding the path if not using the exclude file, thank you for pointing that out, will fix immediately.

The size is calculated with resize2fs (if using ext4) and I am not sure how it does the calculations, I use it because it is WAY faster than running a `du` or something similar. I would suspect it uses `df` or `stat` somehow, because when using that to calculate, it goes fast, but also about the exact same size (or very close to) that resize2fs recommends.

That size is only measured upon root straight up, it disregards whatever is in the exclude file, that file is only used during the `rsync` operation.

I HAVE been thinking (and have tried running code) that uses another applications to calculate sizes and then also exclude things in the exclude file, but whatever application used, it takes way to much time imho so I ended up just relying on resize2fs instead.
The idea of the script is to be fast AND small, and in this situation, the size I "saved" was not worth the time it took just to calculate.
That is ONE of the reasons `[extra space]` exists, in case it becomes WAY to big or if it fails even though it _should_ have enough space.

I have also been thinking of enabling an option where the user chooses to use `du` instead of `resize2fs`, but honestly, it works just as well to just use `[extra space]` and only added a TON of extra code that probably won't be utilized by more than 0.1% anyway.

If you have ideas, I am open ears. :slightly_smiling_face:

bmw-biker, 2024-06-10:
Ok, I understand you already did some deeper digging into the exclude puzzle. You're right, other solutions make no sense when take too long.

Now the "root mount failed" issue...
Please find the logs of the problem Raspberry of working V0.9.4 and error V1.0.1-beta attached (from tty console). Additionally I attached the V1.0.4-beta log of the other Raspberry where it's working.
[shrink-backup-RP1-ok-V0.9.4.log](https://github.com/user-attachments/files/15774422/shrink-backup-RP1-ok-V0.9.4.log)
[shrink-backup-RP1-error-V1.0.1-beta.log](https://github.com/user-attachments/files/15774423/shrink-backup-RP1-error-V1.0.1-beta.log)
[shrink-backup-RP2-ok-V1.0.1-beta.log](https://github.com/user-attachments/files/15774421/shrink-backup-RP2-ok-V1.0.1-beta.log)

From what I can see on the problem Pi:
V0.9.4 does `Running: mkfs.ext4...` and then `Discarding device blocks...`. Then continous copying files and exits without error.
V1.0.1 after `Running: mkfs.ext4...` says `The file  does not exist and no size was specified.`.
Oh, it before says `IMG_DEV_BOOT_PATH= | IMG_DEV_ROOT_PATH=`  ???  instead of `IMG_DEV_BOOT_PATH=/dev/loop0p1 | IMG_DEV_ROOT_PATH=/dev/loop0p2`. So it somehow cannot / doesn't set the partion paths ?

The other Raspberry works well with either version and with or without tty.

Any idea ?

UnconnectedBedna, 2024-06-10:
> Oh, it before says `IMG_DEV_BOOT_PATH= | IMG_DEV_ROOT_PATH=` ??? instead of `IMG_DEV_BOOT_PATH=/dev/loop0p1 | IMG_DEV_ROOT_PATH=/dev/loop0p2`. So it somehow cannot / doesn't set the partion paths ?

Oh noo.... Not that one.. :cold_sweat: 

Let me take a look at the logs, this bug has puzzled me for quite some time now, and it seems to be a bug outside of my script somehow.
Let me see what I can find, I DID lower a sleep value from 2 to 1 in latest updates, could be that... sigh...

bmw-biker, 2024-06-10:
Yes, you're right.
The pause press enter did it. It continued running.
Although, now the script totally crashed somehow. The console link closed. Webmin, `$ free`, `$ top -i`, ... showed different system results. A `$ reboot` didn't change anything.
And another start of shrink-backup got stuck so that I had to power of the Raspberry.

During the backup of course I saw huge memory usage.
On the other hand maybe I'm doing wrong: I'll do some test now with hopefully all main processes being stopped. As I wrote earlier the problem device handles smarthome.....

Ok, now it went through.
I'll do another test with this sleep set back to 1sec without pause. Just for test...

Oh man, from your comment abouf those lines I see, you've got lot's of insights meanwhile during lots of time consuming testing 😬 
Not to mention those days probably where you totally mix up windows. Like me... Had consoles to both Raspberrys open. And wondered that my change to the script wouldn't work - when I changed the wrong one.

UnconnectedBedna, 2024-06-10:
Change it from 1 to 2 seconds, but make sure to do it in all places needed...

Or you can just pull latest from testing, I updated the script so you can just do that if you want.

Also added some coloring. :)

bmw-biker, 2024-06-10:
So, after another test with sleep 1sec, without pause and all processes stopped:
The script again shows the 'The file  does not exist and no size was specified.' error.
=> For now I will leave sleep 5sec at line 525. This won't matter for ~15min backup.
Ok, I'll try your latest from testing tomorrow.....  Thanks very much !

Oh yeah, after it finally works... About delay and speed... there could be a similar reason:
The "good" Raspberry runs with 1GBit Ethernet connection to the NAS drive.
The problem one does WLAN. He says 5GHz ~200MBit. But compared to the other he seems also to somehow take longer breaks while file copying. 
Problem Wifi Pi: backup file ~8GB, ~15min
Ethernet Pi: backup file ~11GB, <5min

UnconnectedBedna, 2024-06-10:
Hmmmm.

That is not a bad clue. Might have something to do with network and the loop...

You know what, I do have one more idea I have been cooking on, and that is simply to loop the script until it finds the the path, because at that point, the script already knows how many partitions to look for (if there is boot or not).

I am a bit adamant though, because if that is not the reason, but the uuid will never show up, the scipt ends up in an enless loop, but I think 3 retrys, with 5s sleep between each, it _SHOULD_ be fine right.. riiiiight?!?!

Since you give up for today, so will I, but I will implement a loop function that makes sure everything goes through 3 tries before failing. if it works immediately, great! Otherwise 2 more tires over 10 seconds hopefully is enough (I will still keep the 2s before anyway).

bmw-biker, 2024-06-11:
👍  This sounds ilke a good compromise - and it works, at least on both of my Raspberrys !

I stop all services now while backup. If backup (also longer lasting) once in a while runs at night that doesn't matter.
Maybe (my?) Wifi has really some delays when it comes to file transmission etc - today the 8GB backup lasted 22min 😬.
4k streaming via same AP is no issue.

So, I am happy now... 👍 
If you would like me to test anything else feel free to send a message.

Again, thanks very much for the effort you spend/t in this script !

UnconnectedBedna, 2024-06-11:
> If you would like me to test anything else feel free to send a message.

Yes please, could you check in the log file of the one that failed if it did loop and retry, or if the lsblk went through immediately (I added stuff to include that in the log file in case it had to retry after 5 seconds).

Also, if you can make updates of the backups (`-U`), especially on the one that failed earlier.

Thank you so much for helping out!

bmw-biker, 2024-06-11:
No, I cannot see it looping, catched the first run: [shrink-backup.log](https://github.com/user-attachments/files/15788550/shrink-backup.log)

I'll come back soon with result on updates. Now I'll have to do some shopping - wife will arrive from holidays soon 🤣

UnconnectedBedna, 2024-06-11:
No worries.

That means the code I put into place was not actually ran, the 2s sleep was enough.

But the fact you pointed out, that the img file is on a network mounted location is actually something I will keep in mind if this bug presents itself again.

Only ext4 and new backups has this "protection".

I will leave this be, and if I don't hear about any problems arising, this issue will be open until the next update (merge testing ==> main) on main branch.

bmw-biker, 2024-06-13:
ok, here I am again.

Backup update had no timing issue at my Wifi "problem" Raspberry.
But, the backup stopped with error in Webmin (without tty).

Attached the log file with 4 cycles.
1. I stumbled over the fact that -U doesn't create a file if not there.
2. Creation of first backup (no tty).
3. Backup update with error (no tty).
4. Backup update working from XShell.
[shrink-backup.log](https://github.com/user-attachments/files/15827237/shrink-backup.log)

UnconnectedBedna, 2024-06-13:
This is great feedback!

From the first backup:
> 2024-06-13 19:49:08 [WARNING] - LOOP paths can not be set, retrying in 5 seconds
2024-06-13 19:49:08 [WARNING] - IMG_DEV_BOOT_PATH= | IMG_DEV_ROOT_PATH=

Here we can see the "protection" worked, and the script slept for 5s and retried, it detected none of the UUID's (that is actually what is happening)

Second backup:
> 2024-06-13 20:20:18 [DEBUG] - IMG_DEV_BOOT_PATH= | IMG_DEV_ROOT_PATH=/dev/loop0p2

That leads to:
> 2024-06-13 20:20:18 [DEBUG] - Running: mount  /tmp/backup-1Sg/boot/firmware
2024-06-13 20:20:18 [ERROR] - BOOT MOUNT FAILED:
mount: /tmp/backup-1Sg/boot/firmware: wrong fs type, bad option, bad superblock on , missing codepage or helper program, or other error.

If you look at the mount line, it's missing the path.
Also VERY interesting is that it finds root but not boot, that is also a new one.

It WAS expected to fail with the backup, because I did not implement the protection part there before I knew it actually worked.

I will update the script in testing branch.

bmw-biker, 2024-06-15:
Wonderful, now "update" runs through as well, within ~3min.
Looks like it needed a second try.
Log attached, fake web console showed:
```
## Debugging requested, writing to log file ./shrink-backup.log
## Zoom speed requested...
## Scanning filesystem and calculating...
## Looping img file...
!! [0;33mLOOP paths can not be set, retrying in 5 seconds...[0m
## [0;32mLOOP paths found, resuming backup...[0m
##############################################################################
# DISABLE PROMPTS SELECTED
# Updating /mnt/dsRaspberryBackup/pi/fhem-pi-2.img
# ext4 filesystem detected on root
...
```
[shrink-backup.log](https://github.com/user-attachments/files/15847228/shrink-backup.log)

Have a nice weekend...

UnconnectedBedna, 2024-06-20:
In preparation for merging of testing ==> main I close this issue.

Version 1.0.1 will drop in a short while.

closed 2024-06-20 by UnconnectedBedna