UnconnectedBedna/shrink-backup #38: Orange Pi 5 success

jerry3904, 2024-07-02:
**REPORT**: this app works very well with the Orange Pi 5 of which I'm just preparing to upload an MX Linux respin (Xfce) where I locate it in the main Menu along with a help doc.<br>

https://github.com/jerry3904/OrangePi_respin<br>

The significance of this success lies in the fact that no other method I've tried or found mentioned online has worked. I would think you might want to add that to your "Experimental" section at least.

UnconnectedBedna, 2024-07-02:
Thank you so much for the feedback!
I will see if I can squeeze it in the readme somewhere.

Although I think I would have preferred it not created as an issue but rather in the discussions section, in the [Feedback tread](https://github.com/UnconnectedBedna/shrink-backup/discussions/1) for example. :wink: 

But thank you again, I am really glad others find the script as useful as I do. :heart:

jerry3904, 2024-07-02:
That's the simple help document I drafted today for the Menu