UnconnectedBedna/shrink-backup #39: LOOP paths can not be set // umounting of /boot/firmware

sneumeister, 2024-07-20:
Hello,

What I did:
Downloaded latest version (1.1) of shrink-backup.
The shring-backup-folder is located on my NAS mounted unter `/mnt/netbackup/`.

```
-rw-r--r-- 1 1024 users 9377861120 Jun  4 22:31 backup_george_bookworm.img
-rw-r--r-- 1 1024 users         83 Aug  1  2023 exclude.txt
-rw-r--r-- 1 1024 users       1539 Jun 25 17:20 LICENSE
-rw-r--r-- 1 1024 users      18192 Jun 25 17:00 README.md
-rwxr-xr-x 1 1024 users      99071 Jun 25 16:58 shrink-backup
-rw-r--r-- 1 1024 users      91117 Jul 20 23:06 shrink-backup.log
```

Ok, so I started upadte `-U` the above visible *.img file
` sudo /mnt/netbackup/shrink-backup/shrink-backup -Uz /mnt/netbackup/shrink-backup/backup_george_bookworm.img`
So far so good, output of program:
```
## Zoom speed requested...
## Scanning filesystem and calculating...
## Looping img file...
#####################################################################################
# Updating /mnt/netbackup/shrink-backup/backup_george_bookworm.img                  #
# f2fs filesystem detected on root                                                  #
# Autoexpand filesystem at boot not available for f2fs                               #
# Resize operations not available for f2fs                                          #
# --------------------------------------------------------------------------------- #
# Write to logfile: false                                                           #
# Zoom speed requested: true                                                        #
# Autocalculate img root partition size: false                                      #
# Autoexpand filesystem at boot: false                                              #
# Use exclude.txt: false                                                            #
# Bootsector size: 515MiB                                                           #
# Estemated root usage: 10661MiB                                                    #
# Total img size: 8943MiB                                                           #
#####################################################################################
## Do you want to continue? [y/n] n
!! Aborting...
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.20
```
Since I discovered logfile was disables I stopped script by entering `n`.
So I restarted backup with additional `l` option:
` sudo /mnt/netbackup/shrink-backup/shrink-backup -Ulz /mnt/netbackup/shrink-backup/backup_george_bookworm.img`
But since than the script shows errors....
```
## Debugging requested, writing to log file: /mnt/netbackup/shrink-backup/shrink-backup.log
## Zoom speed requested...
## Scanning filesystem and calculating...
lsblk: : not a block device
## Looping img file...
!! LOOP paths can not be set, retrying in 5 seconds...
!! LOOP paths can not be set, retrying in 5 seconds...
!! LOOP paths can not be set, retrying in 5 seconds...
^C
!! Script stopped by user...
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.21
```
And this repeat everytime I restart the script.
Logfile:
[shring-backup.log](https://github.com/user-attachments/files/16322380/shring-backup.log)

`losetup --list`
Gives no output.

Output of lsblk:
```
 $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0 119.1G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part
└─mmcblk0p2 179:2    0 118.6G  0 part /
```
This is the effect we have seen in some older issues:
The `mmcblk0p1` was un-mount for what ever reason.
Re-mounting it again to `/boot/firmware`  makes the script run as it should.

Since you search in `/proc/mounts` on your script I was wondering if this unmount-behaviour may be influenced by the order of mounts.

So I rebooted, started `shrink-backup` again and stopped with no (not continue).
The first time nothing happened (checked with `lsblk`).
Starting `shrink-backup` again and stopping with no (not continue) then brought the result with the unmounting of the `/boot/firmware` partition!

Remounting `/boot/firmware` and starting / stopping the script several times then never umounts `/boot/firmware`.

Repeating with reboot, mount `/mnt/netbackup` starting backup uns stopping ist with entering no this time immediately leads to an unmounted `/boot/firmware` ....
Re-mount `/boot/firmware` manually and starting / stopping the script several times then never umounts anything.

Strange behavior... not able to grab the root cause.

UnconnectedBedna, 2024-07-21:
Hi again my best QA tester! xD

Extremely interesting!

So it is still doing that for you huh... (unmounting boot)
I have since you had that problem also noticed this behavior myself, ONCE, but I could not for the life of me figure out WHY my boot was suddenly unmounted, and it has not happened since, but you might have stumbled upon a workflow where we can force it and maybe finally figure out why and when it happens. And if it happens in the cleanup function (witch should not be possible) it should say so in the log file.

Also strange that the script does not warn you about your boot not mounted, I implemented a check for that IIRC, so the script should have exited with that information instead of doing the loop 3 times.
But maybe I forgot to implement that function when using f2fs, I will have to check that again.

But I need the entire log file though.
I can see the variable looking strange af:
```
IMG_DEV_BOOT_PATH=PATH
/dev/loop0
/dev/loop0p1
/dev/loop0p2 | IMG_DEV_ROOT_PATH=/dev/loop0p2
```

That is supposed to point at your boot partitions loop.
So I really need the entire logfile or it won't mean anything for me, I need to track backwards all the way from the start.

sneumeister, 2024-07-21:
Hello,

OK, my root partition is in f2fs format.
So no big changes in `fstab` only changed filesystem to f2fs; the rest was setup when installing bookworm:
```
$ cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=655d03f1-01  /boot/firmware  vfat    defaults          0       2
PARTUUID=655d03f1-02  /               f2fs    defaults,noatime  0       1
diskstation2:/volume1/NetBackup /mnt/netbackup nfs defaults,noauto 0 0

# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```
To complete the information: my `cmdline.txt` looks like that:
```
$ cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=655d03f1-02 rootfstype=f2fs fsck.repair=yes rootwait

```

Here is the logfile from every action on 2024-07-20:
[shrink-backup.log](https://github.com/user-attachments/files/16325295/shrink-backup.log)

I was wondering if the order of mounts in `/proc/mounts` in connection with the `-qs` options somehow is responsible for the trouble. But just guessing..... would take more time to understand your script... ;-)

UnconnectedBedna, 2024-07-21:
Ok..

I analyzed the log file, and I can not see it getting unmounted because you did not include the "pressed N and after that it fails". I can only see that it fails a lot of time. Then I see you pressing "n" to cancel multiple times, THEN it fails, but not from unmounted boot partition, but because (I am assuming here) you use exclude.txt without anything inside it, because the exclude in the rsync line is empty apart from the img directory and temp file and directory created by the script.

Try running the backup either with editing exclude.txt and include virtual directories, or run script without `-t`.

So unless you can show with a logfile that pressing n really unmounts your root, there is not anything I can do.

sneumeister, 2024-07-22:
OK. You want it, you get it..... ;-)

My impression: the 'unmounting' occures when system is freshly rebooted.
At least after that running shrink-backup something like 2 times the unmounting happens.

But let me describe my doing:
First I reboot my RaspberryPi.
After that SSHing to the system.
Manual mounting of `/mnt/netbackup` by `sudo mount /mnt/netbackup`. 
Removing `shrink-backup.log` to have a clean documentation.
Shring-backup is located in folder `/mnt/netbackup/shrink-backup/` .
I start shrink-backup from pwd `~/` as user `pi`. 
The `/proc/mounts` at this point looks like this:
```
pi@george:~ $ cat /proc/mounts
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
proc /proc proc rw,relatime 0 0
udev /dev devtmpfs rw,nosuid,relatime,size=82784k,nr_inodes=20696,mode=755 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
tmpfs /run tmpfs rw,nosuid,nodev,noexec,relatime,size=43820k,mode=755 0 0
/dev/mmcblk0p2 / f2fs rw,lazytime,noatime,background_gc=on,nogc_merge,discard,discard_unit=block,no_heap,user_xattr,inline_xattr,acl,inline_data,inline_dentry,noflush_merge,barrier,extent_cache,mode=adaptive,active_logs=6,alloc_mode=default,checkpoint_merge,fsync_mode=posix,memory=normal,errors=continue 0 0
securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
cgroup2 /sys/fs/cgroup cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot 0 0
pstore /sys/fs/pstore pstore rw,nosuid,nodev,noexec,relatime 0 0
bpf /sys/fs/bpf bpf rw,nosuid,nodev,noexec,relatime,mode=700 0 0
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct 0 0
mqueue /dev/mqueue mqueue rw,nosuid,nodev,noexec,relatime 0 0
debugfs /sys/kernel/debug debugfs rw,nosuid,nodev,noexec,relatime 0 0
tracefs /sys/kernel/tracing tracefs rw,nosuid,nodev,noexec,relatime 0 0
fusectl /sys/fs/fuse/connections fusectl rw,nosuid,nodev,noexec,relatime 0 0
configfs /sys/kernel/config configfs rw,nosuid,nodev,noexec,relatime 0 0
ramfs /run/credentials/systemd-sysctl.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
ramfs /run/credentials/systemd-sysusers.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
ramfs /run/credentials/systemd-tmpfiles-setup-dev.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
ramfs /run/credentials/systemd-tmpfiles-setup.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,nosuid,nodev,noexec,relatime 0 0
sunrpc /run/rpc_pipefs rpc_pipefs rw,relatime 0 0
tmpfs /run/user/1000 tmpfs rw,nosuid,nodev,relatime,size=43816k,nr_inodes=10954,mode=700,uid=1000,gid=1000 0 0
diskstation2:/volume1/NetBackup /mnt/netbackup nfs rw,relatime,vers=3,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=192.168.178.211,mountvers=3,mountport=892,mountproto=udp,local_lock=none,addr=192.168.178.211 0 0
pi@george:~ $
```

   
After 2nd run of shrink-backup the `/boot/firmware` disappeared visible by `lsblk` outpt.
Relevant  snapshot from bash:

```
pi@george:~ $ sudo /mnt/netbackup/shrink-backup/shrink-backup -Utlz /mnt/netbackup/shrink-backup/backup_george_bookworm.img
## Debugging requested, writing to log file: /mnt/netbackup/shrink-backup/shrink-backup.log
## Zoom speed requested...
## Scanning filesystem and calculating...
## Looping img file...
#####################################################################################
# Updating /mnt/netbackup/shrink-backup/backup_george_bookworm.img                  #
# f2fs filesystem detected on root                                                  #
# Autoexpand filesystem at boot not available for f2fs                               #
# Resize operations not available for f2fs                                          #
# --------------------------------------------------------------------------------- #
# Write to logfile: true                                                            #
# Zoom speed requested: true                                                        #
# Autocalculate img root partition size: false                                      #
# Autoexpand filesystem at boot: false                                              #
# Use exclude.txt: true                                                             #
# Bootsector size: 515MiB                                                           #
# Estemated root usage: 10446MiB                                                    #
# Total img size: 8943MiB                                                           #
#####################################################################################
## Do you want to continue? [y/n] n
[... 70 lines cut ...]
tmpfs /run tmpfs rw,nosuid,nodev,noexec,relatime,size=43820k,mode=755 0 0
/dev/mmcblk0p2 / f2fs rw,lazytime,noatime,background_gc=on,nogc_merge,discard,discard_unit=block,no_heap,user_xattr,inline_xattr,acl,inline_data,inline_dentry,noflush_merge,barrier,extent_cache,mode=adaptive,active_logs=6,alloc_mode=default,checkpoint_merge,fsync_mode=posix,memory=normal,errors=continue 0 0
securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
[... cut ...]
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,nosuid,nodev,noexec,relatime 0 0
sunrpc /run/rpc_pipefs rpc_pipefs rw,relatime 0 0
tmpfs /run/user/1000 tmpfs rw,nosuid,nodev,relatime,size=43816k,nr_inodes=10954,mode=700,uid=1000,gid=1000 0 0
diskstation2:/volume1/NetBackup /mnt/netbackup nfs rw,relatime,vers=3,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=192.168.178.211,mountvers=3,mountport=892,mountproto=udp,local_lock=none,addr=192.168.178.211 0 0
pi@george:~ $

```

UnconnectedBedna, 2024-07-22:
So, It REALLY looks like the script is unmounting your partition, but it _might_ be a red herring.
Unless you copy pasted the wrong output, look at your first `cat /proc/mounts` that you pasted, THERE IS NO BOOT MOUNTED, so I am VERY confused that `lsblk` returns it as mounted.
I AM dyslectic, but I have read the output for about 5 mins, and unless you accidentally pasted the wrong output, the boot is already gone at that point somehow.

Looking at the logs, it only removes the loop, witch is correct because nothing else is done at that point.
If any unmounting were to happen, it would be logged.

framps, 2024-07-23:
Just a note from my side on this issue: I was just able to [create a backup of RaspbianOS 64 bit lite](https://github.com/UnconnectedBedna/shrink-backup/issues/41) on a RPi3 and executed -Ulz without any issue.  I just added one file in /home/pi and the update of the backup finished quite fast :+1:

sneumeister, 2024-07-23:
OK, just saw you updated testing 1h ago..... let's do another try with a freshly rebooted system.

This time I needed to start the script more often but finally again `/boot/firmware` was unmunted.
Full log from terminal window to not mix up things:

```
pi@george:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0 119.1G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2 179:2    0 118.6G  0 part /
pi@george:~ $ cat /proc/mounts
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
proc /proc proc rw,relatime 0 0
udev /dev devtmpfs rw,nosuid,relatime,size=82784k,nr_inodes=20696,mode=755 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
tmpfs /run tmpfs rw,nosuid,nodev,noexec,relatime,size=43820k,mode=755 0 0
/dev/mmcblk0p2 / f2fs rw,lazytime,noatime,background_gc=on,nogc_merge,discard,discard_unit=block,no_heap,user_xattr,inline_xattr,acl,inline_data,inline_dentry,noflush_merge,barrier,extent_cache,mode=adaptive,active_logs=6,alloc_mode=default,checkpoint_merge,fsync_mode=posix,memory=normal,errors=continue 0 0
securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
cgroup2 /sys/fs/cgroup cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot 0 0
pstore /sys/fs/pstore pstore rw,nosuid,nodev,noexec,relatime 0 0
bpf /sys/fs/bpf bpf rw,nosuid,nodev,noexec,relatime,mode=700 0 0
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct 0 0
mqueue /dev/mqueue mqueue rw,nosuid,nodev,noexec,relatime 0 0
debugfs /sys/kernel/debug debugfs rw,nosuid,nodev,noexec,relatime 0 0
tracefs /sys/kernel/tracing tracefs rw,nosuid,nodev,noexec,relatime 0 0
fusectl /sys/fs/fuse/connections fusectl rw,nosuid,nodev,noexec,relatime 0 0
configfs /sys/kernel/config configfs rw,nosuid,nodev,noexec,relatime 0 0
ramfs /run/credentials/systemd-sysctl.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
ramfs /run/credentials/systemd-sysusers.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
ramfs /run/credentials/systemd-tmpfiles-setup-dev.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
/dev/mmcblk0p1 /boot/firmware vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro 0 0
ramfs /run/credentials/systemd-tmpfiles-setup.service ramfs ro,nosuid,nodev,noexec,relatime,mode=700 0 0
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,nosuid,nodev,noexec,relatime 0 0
sunrpc /run/rpc_pipefs rpc_pipefs rw,relatime 0 0
tmpfs /run/user/1000 tmpfs rw,nosuid,nodev,relatime,size=43816k,nr_inodes=10954,mode=700,uid=1000,gid=1000 0 0
diskstation2:/volume1/NetBackup /mnt/netbackup nfs rw,relatime,vers=3,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=192.168.178.211,mountvers=3,mountport=892,mountproto=udp,local_lock=none,addr=192.168.178.211 0 0
pi@george:~ $ sudo /mnt/netbackup/shrink-backup/shrink-backup-testing -Utlz /mnt/netbackup/shrink-backup/backup_george_bookworm.img
## Debugging requested, writing to log file: /mnt/netbackup/shrink-backup/shrink-backup.log
## Zoom speed requested...
## Scanning filesystem and calculating...
## Looping img file...
!! LOOP paths can not be set, retrying in 5 seconds...
## LOOP paths found, resuming backup...
#####################################################################################
# Updating /mnt/netbackup/shrink-backup/backup_george_bookworm.img                  #
# f2fs filesystem detected on root                                                  #
# Autoexpand filesystem at boot not available for f2fs                               #
# Resize operations not available for f2fs                                          #
# --------------------------------------------------------------------------------- #
# Write to logfile: true                                                            #
# Zoom speed requested: true                                                        #
# Autocalculate img root partition size: false                                      #
# Autoexpand filesystem at boot: false                                              #
# Use exclude.txt: true                                                             #
# Bootsector size: 515MiB                                                           #
# Estemated root usage: 10408MiB                                                    #
# Total img size: 8943MiB                                                           #
#####################################################################################
## Do you want to continue? [y/n] n
!! Aborting...
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.24
pi@george:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0 119.1G  0 disk
[... 105 lines cut ...]
# --------------------------------------------------------------------------------- #
# Write to logfile: true                                                            #
# Zoom speed requested: true                                                        #
# Autocalculate img root partition size: false                                      #
# Autoexpand filesystem at boot: false                                              #
[... cut ...]
#####################################################################################
## Do you want to continue? [y/n] n
!! Aborting...
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.38
pi@george:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0 119.1G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part
└─mmcblk0p2 179:2    0 118.6G  0 part /
pi@george:~ $
```
The corresponding log file:
[shrink-backup.log](https://github.com/user-attachments/files/16353715/shrink-backup.log)

framps, 2024-07-23:
Do you have any automount services active on your system? I've seen these kind of issues already multiple times cause issues for backup tools. Do you use systemd automount for example?

sneumeister, 2024-07-24:
@framps:

My RaspberryPi 1B runs headless in cellar.
Because shring-backup did not work with original Debian/Buster I installed Bookworm ~3 months ago.
On the system I run [https://github.com/SBFspot/SBFspot](https://github.com/SBFspot/SBFspot), Node-Red and Mosquitto MQTT.
The net-share I mount manually.
So actively I do not have configured any automount.
My `/etc/fstab` does also contain no `x-systemd.automount` option........
```
pi@george:~ $ cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=655d03f1-01  /boot/firmware  vfat    defaults          0       2
PARTUUID=655d03f1-02  /               f2fs    defaults,noatime  0       1
diskstation2:/volume1/NetBackup /mnt/netbackup nfs defaults,noauto 0 0

# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```

systemd meint dazu...
```
pi@george:~ $ sudo systemctl status *.automount
● proc-sys-fs-binfmt_misc.automount - Arbitrary Executable File Formats File System Automount Point
     Loaded: loaded (/lib/systemd/system/proc-sys-fs-binfmt_misc.automount; static)
     Active: active (running) since Mon 2024-07-01 10:23:59 CEST; 3 weeks 2 days ago
   Triggers: ● proc-sys-fs-binfmt_misc.mount
      Where: /proc/sys/fs/binfmt_misc
       Docs: https://docs.kernel.org/admin-guide/binfmt-misc.html
             https://www.freedesktop.org/wiki/Software/systemd/APIFileSystems

Jul 23 20:42:08 george systemd[1]: proc-sys-fs-binfmt_misc.automount: Got automount request for /proc/sys/fs/binfmt_misc, triggered by>
Notice: journal has been rotated since unit was started, output may be incomplete.
```

So I would say: no automount is not configured.

framps, 2024-07-24:
It's just a guess. I know automount and devmon automount partitions. 

Please check the output of `apt list --installed | egrep "mount|devmon"`

That's what I get on my system and shows no automount is active:
```
pi@raspberrypi-bookworm32-light:~ $ apt list --installed | egrep "mount|devmon"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libmount1/stable,now 2.38.1-5+deb12u1 armhf [installed]
mount/stable,now 2.38.1-5+deb12u1 armhf [installed]
```

sneumeister, 2024-07-24:
@framps:
My output:
```
pi@george:~ $ sudo apt list --installed | egrep "mount|devmon"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libmount1/stable,now 2.38.1-5+deb12u1 armhf [installed]
mount/stable,now 2.38.1-5+deb12u1 armhf [installed]
```
So same out as yours.

framps, 2024-07-24:
Ok. So there's some other reason why you face this issue - unfortunately.

UnconnectedBedna, 2024-07-25:
It is very strange, but since I can not find any proof that the script is actually causing this, there is not much I can do.
If it was the script, it would happen every time, not just sometimes. It would also be included in the log file, but the code for unmounting is not even run if exiting the script that early so it makes absolutely zero sense. :(

I was 100% sure I implemented a check that boot is actually mounted if it exists in fstab, I must have dreamt that....
Updated testing branch. Now if boot is not mounted, you will get info and the script will exit.
Tested and confirmed it worked. But it does not help with your boot becoming unmounted, but at least you get a warning about it now.

sneumeister, 2024-07-25:
OK, all very strange.
Tried now  to repeat this topic with your 7h old "testing" version ... reboot... restart script several time.... after 15th try `/boot/firmware` was still mounted and no issue..... 

Then looking into `dmesg` (nothing unusual) I wanted to really backup......... (at this time `/boot/firmware` was mounted ) but I got this..............
```
pi@george:~ $ sudo /mnt/netbackup/shrink-backup/shrink-backup-testing -atlz /mnt/netbackup/shrink-backup/backup_george_bookworm.img
## Debugging requested, writing to log file: /mnt/netbackup/shrink-backup/shrink-backup.log
## Zoom speed requested...
## Scanning filesystem and calculating...
#####################################################################################
# A backup will be created at /mnt/netbackup/shrink-backup/backup_george_bookworm.img #
# f2fs filesystem detected on root                                                  #
# Autoexpand filesystem at boot not available for f2fs                               #
# --------------------------------------------------------------------------------- #
# Write to logfile: true                                                            #
# Zoom speed requested: true                                                        #
# Autocalculate img root partition size: true                                       #
# Autoexpand filesystem at boot: false                                              #
# Use exclude.txt: true                                                             #
# Bootsector size: 515MiB                                                           #
# Estemated root usage: 10306MiB                                                    #
# Auto calculated size (root partition): 10818MiB                                   #
# Total img size: 11334MiB                                                          #
#####################################################################################
## Do you want to continue? [y/n] y

!! WARNING!!! WARNING!!! WARNING!!!
!! /mnt/netbackup/shrink-backup/backup_george_bookworm.img
!! FILE ALREADY EXISTS!!!
!! Do you want to overwrite? [y/n] y

!! Removing old img file...
## Creating bootsector...
545249280 bytes (545 MB, 520 MiB) copied, 165 s, 3.3 MB/s
1067007+0 records in
1067007+0 records out
546307584 bytes (546 MB, 521 MiB) copied, 165.518 s, 3.3 MB/s
## Resizing img file...
## Looping img file...
## Removing root partition...
## Recreating root partition...
## Formatting filesystem...

    F2FS-tools: mkfs.f2fs Ver: 1.15.0 (2022-05-13)

Info: Disable heap-based policy
Info: Debug level = 0
Info: Label = system
Info: Trim is enabled
        /dev/loop0p2 appears to contain an existing filesystem (f2fs).
Info: Segments per section = 1
Info: Sections per zone = 1
Info: sector size = 512
Info: total sectors = 22156375 (10818 MB)
Info: zone aligned segment0 blkaddr: 512
Info: format version with
  "Linux version 6.6.31+rpt-rpi-v6 (serge@raspberrypi.com) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian) 2.40) #1 Raspbian 1:6.6.31-1+rpt1 (2024-05-29)"
Info: [/dev/loop0p2] Discarding device
Info: This device doesn't support BLKSECDISCARD
Info: This device doesn't support BLKDISCARD
Info: Overprovision ratio = 1.930%
Info: Overprovision segments = 212 (GC reserved = 111)
Info: format successful
## Creating temp directory...
## Mounting img root partition...
!! BOOT PARTITION NOT MOUNTED ON SYSTEM, PLEASE MOUNT BOOT PARTITION AND RE-RUN THE SCRIPT!!!
!! Cleanup function called with non zero exit code, something went wrong!!!
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 04.06
pi@george:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0 119.1G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part
└─mmcblk0p2 179:2    0 118.6G  0 part /
```
Boot Partition not mounted.............  ?!?
Log file reduced to today..... you see the 15 mentioned trials.... the 16th run is of the output been seen above.:
[shrink-backup.log](https://github.com/user-attachments/files/16383391/shrink-backup.log)

I do not know what's going on... 😕 

Does  `dmesg --time-format iso` help anything?
```
2024-07-25T22:04:28,212530+02:00 loop0: detected capacity change from 0 to 23213143
2024-07-25T22:04:28,441338+02:00  loop0: p1 p2
2024-07-25T22:04:28,444714+02:00 loop0: p2 size 248680448 extends beyond EOD, truncated
2024-07-25T22:04:32,180913+02:00  loop0: p1
2024-07-25T22:04:32,473185+02:00  loop0: p1
2024-07-25T22:04:34,565674+02:00  loop0: p1 p2
2024-07-25T22:05:16,052655+02:00  loop0: p1 p2
2024-07-25T22:05:16,456035+02:00 F2FS-fs (loop0p2): Found nat_bits in checkpoint
2024-07-25T22:05:16,989008+02:00 F2FS-fs (loop0p2): Mounted with checkpoint version = d467831
pi@george:~ $
```

UnconnectedBedna, 2024-07-26:
THIS is VERY interesting.

Please do not give up, but also be aware that you might dig yourself so deep down the hole everything turns to darkness. xD

If this was ran with the latest version from testing branch...
I implemented another check on the boot partition that runs almost immediately when starting the script on line 2036:
```
# Make sure boot is mounted if it exists in fstab
if grep -q 'boot' /etc/fstab && ! mount | grep -q 'boot'; then
  echo '!! Boot found in fstab but partition not mounted, please mount and retry script...'
  debug 'ERROR' 'Boot found in fstab but partition not mounted, exit 3'
  exit 3
fi
```

Then after you have pressed Y and all that jazz, it does another check of the boot in the do_mount function on line 722:
```
# Make sure boot is mounted if it exists in fstab
if grep -q 'boot' /etc/fstab && ! mount | grep -q 'boot'; then
  echo '!! Boot found in fstab but partition not mounted, please mount and retry script...'
  debug 'ERROR' 'Boot found in fstab but partition not mounted, exit 3'
  exit 3
fi
```

Now, if someone can explain THAT, I would be REALLY grateful, because I find no logic about any of this... :melting_face: 
The second check, is NOT NEW, it has been there for a long time, but the new FIRST check goes through without a hitch, and it also finds the paths that was the initial error for you when you made your first issue.

Keep going, do not feel you are a burden or anything.
Again YOU ARE MY BEST QA TESTER! :heart:

sneumeister, 2024-07-27:
I will make other trials.
But this weekend first I will upgrade my NAS... 😃

messageexplorer, 2024-09-17:
Hello, I have same issue.

Can it be issue with grep utility?
 
raspi@raspi:~$ cat /etc/fstab
LABEL=writable  /       ext4    defaults        0       1
LABEL=system-boot       /boot/firmware  vfat    defaults        0       1
raspi@raspi:~$ mount | grep -q 'boot'
raspi@raspi:~$ mount | grep 'boot'
/dev/sda1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
raspi@raspi:~$ grep 'boot' /etc/fstab
LABEL=system-boot       /boot/firmware  vfat    defaults        0       1
raspi@raspi:~$

UnconnectedBedna, 2024-10-18:
I am planning a new release, maybe even this weekend.

Any news on this?

Otherwise I will close this issue since it seems to be related to something outside the script.

It can always be opened again if something were to turn up.

closed 2024-10-20 by UnconnectedBedna