UnconnectedBedna/shrink-backup #41: Boot from a backup created from an RPI3 running RaspbianOS lite fails with "ALERT! PARTUUID=a1f2b077-02 does not exist. Dropping to a shell!"

framps, 2024-07-22:
I created a backup successfully for the system mentioned in the issue title.

When I started the restored backup (restored with dd) I get

```
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
done.
Gave up waiting for root file system device. 
...
ALERT! PARTUUID=a1f2b077-02 does not exist. Dropping to a shell!
```
I checked the backup image and get 
```
sudo blkid
/dev/sdc1: UUID="BDDC-31AC" TYPE="vfat" PARTUUID="966ad0d3-01"
/dev/sdc2: UUID="b4f05dfa-8a60-47e8-a6f2-f923d6d339dc" TYPE="ext4" PARTUUID="966ad0d3-02"
```
There is indeed no PARTUUID a1f2b077-02. 

[shrink-backup.log](https://github.com/user-attachments/files/16339746/shrink-backup.log)
In the debug log `PARTUUID a1f2b077-02` is used. But the created backup image has `PARTUUID="966ad0d3-02"`.

UnconnectedBedna, 2024-07-22:
Please read https://github.com/UnconnectedBedna/shrink-backup/issues/40#issuecomment-2243840867 and continue conversation here about this topic. :slightly_smiling_face:

framps, 2024-07-23:
> If the content of `cmdline.txt` is `a1f2b077-02` you must have done some trickery,

Yes, in cmdline.txt and /etc/fstab of tha restored system a1f2b077-02 is used but the partuuid of the SD card is 966ad0d3-02 :thinking: 

Ok. Now I execute a backup and restore again and document in detail what I do:

framps, 2024-07-23:
Debug log of the backup:
[shrink-backup.log](https://github.com/user-attachments/files/16350049/shrink-backup.log)

framps, 2024-07-23:
```
pi@raspberrypi-bookworm32-light:~ $ lsblk -f -o +partuuid,ptuuid
NAME        FSTYPE FSVER LABEL UUID                                 FSAVAIL FSUSE% MOUNTPOINTS    PARTUUID                             PTUUID
sda                                                                                                                                    89094449-425c-49b1-a0f8-901795f76635
├─sda1      ext4   1.0         94a61e86-27fa-450e-a93f-81f85d7ffa4a  140.4G    23% /backup        1a95df5f-2644-46ef-9182-a9b6ced248f4 89094449-425c-49b1-a0f8-901795f76635
└─sda2      ext4   1.0         946b0b1c-bdba-45a0-ac32-56354fdd69bb                               6ea82550-5bda-45b3-b4bd-1bfe96e83b4d 89094449-425c-49b1-a0f8-901795f76635
sdb                                                                                                                                    
mmcblk0                                                                                                                                a1f2b077
├─mmcblk0p1 vfat   FAT32       BDDC-31AC                             413.7M    19% /boot/firmware a1f2b077-01                          a1f2b077
└─mmcblk0p2 ext4   1.0         b4f05dfa-8a60-47e8-a6f2-f923d6d339dc    3.8G    39% /var/hdd.log   a1f2b077-02                          a1f2b077                                                                                   /                           
```

/dev/sdb is the SD card which will get the restored image from /backup/shrinkBackup.img.

```
pi@raspberrypi-bookworm32-light:~ $ sudo fdisk -l /dev/sdb
Disk /dev/sdb: 7.28 GiB, 7816085504 bytes, 15265792 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```
Restoring the backup

```
pi@raspberrypi-bookworm32-light:~ $ sudo dd if=/backup/shrinkBackup.img of=/dev/sdb bs=2MiB
```

framps, 2024-07-23:
Check of the restored image:
```
pi@raspberrypi-bookworm32-light:~ $ sudo blkid
/dev/mmcblk0p2: UUID="b4f05dfa-8a60-47e8-a6f2-f923d6d339dc" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="a1f2b077-02"
/dev/mmcblk0p1: UUID="BDDC-31AC" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="a1f2b077-01"
/dev/sda2: UUID="946b0b1c-bdba-45a0-ac32-56354fdd69bb" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="6ea82550-5bda-45b3-b4bd-1bfe96e83b4d"
/dev/sda1: UUID="94a61e86-27fa-450e-a93f-81f85d7ffa4a" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="1a95df5f-2644-46ef-9182-a9b6ced248f4"
/dev/sdb2: UUID="b4f05dfa-8a60-47e8-a6f2-f923d6d339dc" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="8b3caecd-02"
/dev/sdb1: UUID="BDDC-31AC" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="8b3caecd-01"
```

/dev/sdb uses PARTUUID 8b3caecd-01

```
pi@raspberrypi-bookworm32-light:~ $ sudo mount /dev/sdb1 /mnt
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
pi@raspberrypi-bookworm32-light:~ $ cat /mnt/cmdline.txt 
console=serial0,115200 console=tty1 root=PARTUUID=a1f2b077-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE video=HDMI-A-1:1280x800M@60
```
cmdline.txt used PARTUUID a1f2b077-02 which is the PARTUUID of the active OS and used to create and restore the backup. 

Same for /etc/fstab ...

```
pi@raspberrypi-bookworm32-light:~ $ sudo mount /dev/sdb2 /mnt
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
pi@raspberrypi-bookworm32-light:~ $ cat /mnt/etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=a1f2b077-01  /boot/firmware  vfat    defaults,ro          0       2
PARTUUID=a1f2b077-02  /               ext4    defaults,noatime  0       1

# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```

So somehow the PARTUUID of the restored image doesn't match the PARTUUID of the system which was saved and restored.

```
pi@raspberrypi-bookworm32-light:/backup $ sudo losetup -P /dev/loop0 shrinkBackup.img
pi@raspberrypi-bookworm32-light:/backup $ sudo blkid
/dev/mmcblk0p2: UUID="b4f05dfa-8a60-47e8-a6f2-f923d6d339dc" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="a1f2b077-02"
/dev/mmcblk0p1: UUID="BDDC-31AC" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="a1f2b077-01"
/dev/sda2: UUID="946b0b1c-bdba-45a0-ac32-56354fdd69bb" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="6ea82550-5bda-45b3-b4bd-1bfe96e83b4d"
/dev/sda1: UUID="94a61e86-27fa-450e-a93f-81f85d7ffa4a" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="1a95df5f-2644-46ef-9182-a9b6ced248f4"
/dev/sdb2: UUID="b4f05dfa-8a60-47e8-a6f2-f923d6d339dc" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="8b3caecd-02"
/dev/sdb1: UUID="BDDC-31AC" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="8b3caecd-01"
/dev/loop0p2: UUID="b4f05dfa-8a60-47e8-a6f2-f923d6d339dc" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="8b3caecd-02"
/dev/loop0p1: UUID="BDDC-31AC" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="8b3caecd-01"
```
As you can see the backup image (/dev/loop0) already uses PARTUUID 8b3caecd-02. 

I now will execute the same procedure with a 64bit lite image to check whether it's an issue with the 32bit version.

UnconnectedBedna, 2024-07-23:
Reading the log file:
`2024-07-22 22:19:36 [DEBUG] - LOCAL_BOOT_UUID=BDDC-31AC | LOCAL_ROOT_UUID=b4f05dfa-8a60-47e8-a6f2-f923d6d339dc | LOCAL_ROOT_PARTUUID=a1f2b077-02`

The correct PARTUUID according to the lsblk you provided so there is nothing wrong with the script.

I would suspect the reason you get a new PARTUUID is because you dd on the same machine as you made the backup from, so linux maybe detects the conflict and changes it for you because you can not have 2 of the same unique identifiers in your devices. ESPECIALLY not the same as the root you are currently running.

Redo the dd on another machine and retry.

framps, 2024-07-23:
> I would suspect the reason you get a new PARTUUID is because

Sounds reasonable. But I just executed the same procedure with the 64 bit OS and get the identical PARTUUID on the backup (/dev/loop0).

```
pi@raspberrypi-boorkworm64-lite:~ $ sudo losetup -P /dev/loop0 /backup/shrinkBackup64.img
pi@raspberrypi-boorkworm64-lite:~ $ sudo blkid
/dev/mmcblk0p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="91FE-7499" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="f8b36121-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="56f80fa2-e005-4cca-86e6-19da1069914d" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="f8b36121-02"
/dev/sdb2: UUID="b4f05dfa-8a60-47e8-a6f2-f923d6d339dc" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="8b3caecd-02"
/dev/sdb1: UUID="BDDC-31AC" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="8b3caecd-01"
/dev/loop0p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="91FE-7499" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="f8b36121-01"
/dev/loop0p2: LABEL="rootfs" UUID="56f80fa2-e005-4cca-86e6-19da1069914d" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="f8b36121-02"
/dev/sda2: UUID="946b0b1c-bdba-45a0-ac32-56354fdd69bb" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="6ea82550-5bda-45b3-b4bd-1bfe96e83b4d"
/dev/sda1: UUID="94a61e86-27fa-450e-a93f-81f85d7ffa4a" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="1a95df5f-2644-46ef-9182-a9b6ced248f4"
```
So I think the restored image now will boot. Why this doesn't work on the 32 bit OS ... I don't know.

Will execute the restore now of the 64 bit OS.

UnconnectedBedna, 2024-07-23:
Try rebooting the rpi3 if you haven't, then make a new img and loop that img ON ANOTHER MACHINE and check the partuuid. As you figured out, you don't need to write the img to see the uuid.

I hope you are aware you can use `shrink-backup` for this.
`sudo shrink-backup --loop /path/to/backup.img` :smiley:

framps, 2024-07-23:
The restored 64 image boots successfully even I used the same OS which I saved with shrink-backup to restore the backup. So it looks like there is some issue with shrink-backup and the 32 bit RaspbianOS lite. I still have some RPi2 in production which can run the 32 bit OS only.

> I hope you are aware you can use `shrink-backup` for this.
> `sudo shrink-backup --loop /path/to/backup.img`

No. I wasn't because I frankly didn't understand what you meant with `looping` in the help text. Maybe you should write something like `mount image to a loop device` or similar to make this clear :wink:

UnconnectedBedna, 2024-07-23:
> No. I wasn't because I frankly didn't understand what you meant with `looping` in the help text. Maybe you should write something like `mount image to a loop device` or similar to make this clear 😉

No, I can't do that, because it doesn't get mounted, it gets looped.
Mount you will have to do yourself afterwards if you want... xD

I don't think I can make it clearer than "Loop img file and exit"

And here comes the sucky part.
This is not a bug, because rpi3 is not supported, and I am not sure I want to either, since I have no reliable way of constantly testing it reliably.

framps, 2024-07-23:
> No, I can't do that, because it doesn't get mounted, it gets looped.

:smile: Ok. You won :+1: Strictly speaking it's not a mount. I checked the losetup man page and it says

"losetup  is  used to associate loop devices with regular files or block devices,"

A loop device is usually used to access the partitions of an image file and thus you usually use losetup together with mount and that's why a lot of people use mount instead of associate. That's what rpi-clone for example does at the end of the backup image creation: Associate the backup image with a loop device and mount the boot and root partition so they can be easily accessed in a second window. Maybe that's a convenience enhancement for shrink-backup :wink: 

> This is not a bug, because rpi3 is not supported, and I am not sure I want to either, since I have no way to continuously testing it reliably.

It's a bug as long it's not documented RPI3 is not supported :wink: I think it's not an RPi3 issue but a 32 bit RaspbianOS issue. I'm quite sure it's possible to reproduce the issue on a RPi4 or RPi5 when you use RaspianOS 32. Frankly it doesn't make sense to run the 32 bit OS on this HW. But from a testing point of you you can execute tests and maybe modify shrink-backup to support also 32 bit OS. Maybe it's a low hanging fruit.

UnconnectedBedna, 2024-07-23:
> It's a bug as long it's not documented RPI3 is not supported

Amiga and C64 is also not mentioned, should they be considered supported? xD
What is mentioned in the README.MD is what is supported, nothing else.
It might work on other things, for example I _can_ use it on my btrfs arch install on my deskotp and laptop, but it's not very useful for that.

If people come in with what they think is a bug, like you do, if they make a good case, I might implement support for whatever they are using. Done that for both dietpi and and special usecase in webmin, and then added that information on the readme.
But since I have no way of even trying to find out why it fails on something that DEMANDS 32bit, I consider this a non supported issue.

Actually, now that I read it, I see that it is incorrect too, since f2fs does NOT support autoexpansion, I have to rephrase that and probably squeeze in 64bit somewhere.
Something like:

Supported for 64-bit versions of Raspberry Pi os (bookworm and older), Armbian, Manjaro-arm, DietPi & ArchLinuxARM for rpi with ext4 or [f2fs](https://github.com/UnconnectedBedna/shrink-backup#f2fs) root partition but probably works on most SBC distributions. Autoexpansion works on all except `f2fs`.

> Maybe it's a low hanging fruit.

Quite the opposite actually, for 2 reasons.
1. I don't think there is hardly anybody using 32-bit on a 64-bit processors, not even when emulating nowdays.
2. I have enough testing on each release as it is, I don't need more work unless it is actually useful for users...

But again, if someone finds out what this is about, feel free to inform me and if it's an easy fix, sure...

You never did what I asked though, witch was to reboot, make a fresh img, bring that img to another system, loop it and check the PARTUUID. Becasue I STILL believe there is something strange with your system, not the script, the script READS the PARTUUID correctly, and a dd, well, you know what a dd does, and after that, the UUID is decided and the script never touches anything around it.

framps, 2024-07-23:
> Supported for 64-bit versions of Raspberry Pi os (bookworm and older), Armbian, Manjaro-arm, DietPi & ArchLinuxARM for rpi with ext4 or [f2fs](https://github.com/UnconnectedBedna/shrink-backup#f2fs) [...]

:+1: 

> 1. I don't think there is hardly anybody using 32-bit on a 64-bit processors, not even when emulating nowdays.

I wrote it's useless but I wanted to point out you can debug the 32 bit issue even if you don't have a 32 bit RPi - if you want support the old 32 bit RPis.

>     2. I have enough testing on each release as it is, I don't need more work unless it is actually useful for users...

I see your point. That's why I support RPis in raspiBackup only and added an `--unsupported flag` for users using another OS or HW.

> You never did what I asked though, witch was to reboot, make a fresh img, bring that img to another system, loop it and check the PARTUUID.

Will do now :wink:

UnconnectedBedna, 2024-07-23:
> I wrote it's useless but I wanted to point out you can debug the 32 bit issue even if you don't have a 32 bit RPi - if you want support the old 32 bit RPis.

No, I would not feel comfortable calling that supported.
Supported would be if I had ran tests of the script on hardware that requires 32-bit, the can very well be differences.

I do own a rpi2, but it has been "lended out" (read indefinately) so I can't exactly rely on that for testing, because knowing myslef, after a few days I would come back with "aaaactually I did some testing after all...".. xD
And my rpi3 is a B+, so it's also 64 bit..

framps, 2024-07-23:
> Will do now

Bullseye :smile: 

I checked the 32 bit image backup PARTUUID on my Linux desktop and it's identical to the PARTUUID of the system backed up. Then I restored the backup with dd on my Linux desktop to a SD card and finally booted the restored image on my RPi3 ... success :clap: 

Looks like some mysterious things happen when there are identical PARTUUIDs on a system when a 32 bit OS is used.

I saw you asked for feedback: I got lot of issues with raspiBackup when I kept the PARTUUID in the backup identical to the source. A lot of folks created a backup and plugged in the backup on the system which was backed up just to test the backup. As you already wrote - it's dangerous to have the same PARTUUIDs multiple times on a system and creates a lot of headache. That's why I finally decided to change the new PARTUUID generation default in the backup which was initially off to on.

UnconnectedBedna, 2024-07-23:
So there never was anything wrong huh..? xD

> Looks like some mysterious things happen when there are identical PARTUUIDs on a system when a 32 bit OS is used.

No, you did something strange, nothing has changed in regards to the rpi3 issue you posted about. It just "suddenly started working" witch means you did something different.
If the user does strange things like that, it's on the user. Sorry... :innocent: 

Maybe you misunderstand what the script is.
- The script is not: "here, have a classifier and let me hold your hand for you, idiot proof - kinda script".
- The script Is: "a pretty simple way of making bootable img files from your system in a fraction of the tame it takes on other backup software, with the possibility of advanced usage".

Suddenly changing partuuid and things like that is absolutely NOT what the script is about.
If that is something you are looking for, I suggest you turn towards something else because it will NEVER happen on this script, it totally goes against the mentality of cross platform and simplicity.

When I asked the devs of rpiOS what the reason was, it was stated exactly like you did: "We got tons and tons of "bug reports" where users did not understand that they can not boot with 2 devices with the same PARTUUID connected at the same time, so we decided to force implement a change when using the autoexpansion in raspi-config". Ie, the users were dumb, we created a way to hide that fact by changing things without letting the user know.
**That kind of mentality will NEVER be acceptable in shrink-backup. THAT IS A PLEDGE.**

Might sound strange, but to me it's a mentality thing, about shrink-backup never ever patting the user on the head because _the user is dumb_, even if I sometimes sometimes might think so.. :melting_face: 

If the script doesn't work, the user is more than welcome to come here, read the README on the github to educate themselves, reading the wiki, or even ask a question in the discussions section. I will always be welcoming and trying to please, sometimes even implementing changes to make things easier to achieve.
But the script will NEVER hold the users hand due to lack of knowledge.

framps, 2024-07-23:
> It just "suddenly started working" witch means you did something different.

Nope. I did not. Looks like you're not interested in any feedback :-(. I'm over and out now.

UnconnectedBedna, 2024-07-23:
I am, but I am not willing to change things that will break things for others.

I'm sorry you feel that way.

closed 2024-07-23 by UnconnectedBedna

framps, 2024-07-24:
> I'm sorry you feel that way, that giving you hours of my time trying to help was not enough for you.

I think we both misunderstand each other and tried to explain why [here](https://github.com/UnconnectedBedna/shrink-backup/issues/40#issuecomment-2248727695)

I also felt very uncomfortable with this issue because it's a very nasty issue. I agree it's not a good idea to have the same partitions with the same PARTUUID multiple times on different devices. But it finally worked on the 64 OS. So I tried again to backup and restore the 32 bit image on my RPi3 - and the PARTUUIDs were identical and the restores system booted successfully :smile: 

So it looks like you are right I did something wrong when I tested shrink-backup with the 32 bit OS. Frankly I have no clue what I may have done wrong. Unfortunately I'm not able to reproduce the issue :cry: 

Sorry you had to spent your space time on this invalid issue.