UnconnectedBedna/shrink-backup #45: Bug in shrink-backup (!! PARTED FAILED!!!)

mandresve, 2024-08-01:
Hi, thank you for this amazing tool. I've encountered an error:

## Creating bootsector...

18431+0 records in
18431+0 records out
9436672 bytes (9,4 MB, 9,0 MiB) copied, 0,439452 s, 21,5 MB/s
## Resizing img file...
## Looping img file...
## Removing root partition...
## Recreating root partition...
**Error: The location 4194304 is outside of the device /dev/loop0.
!! PARTED FAILED!!!
!! Cleanup function called with non zero exit code, something went wrong!!!**
## Exiting and cleaning up...
## Please stand by...
## Done.

mandresve, 2024-08-01:
It seems that -a is miscalculating the size by exactly 512. So I disabled -a and added 10MB to the end of the image and seems to be working right.

UnconnectedBedna, 2024-08-02:
Hello and thank you for creating this bug report!

Pure luck I noticed this, for some reason, github decided to not send an email notification about this! :open_mouth: 

Reading through your log I find:
```
2024-08-01 15:53:11 [INFO]  - Calculating recommended root size
2024-08-01 15:53:11 [DEBUG] - BLOCKSIZE= bytes
```
Witch obv leads to errors in the calculations:
```
2024-08-01 15:53:11 [INFO]  - ext4 filesystem detected, using resize2fs to set recommended root size
2024-08-01 15:53:11 [DEBUG] - LOCAL_AUTORESIZE_MIN=0 bytes
2024-08-01 15:53:11 [INFO]  - Setting TOTAL (space needed for files on root) to autocalculated size
2024-08-01 15:53:11 [DEBUG] - TOTAL=0 bytes
<snip>
2024-08-01 15:53:22 [INFO]  - Using truncate to resize img file to 3MiB
```
So it's a _little bit_ too smaller than 512KiB.. xD

The script also says you have no boot partition, is that correct?
Are you running an armbian release or somtehing similar without a boot partition?

Please provide output of `stat -f / | grep -i 'Block size' | awk '{print $3}'`

mandresve, 2024-08-02:
Thank you for your response.

Indeed, I was trying to backup an embedded Debian Bookworm system. Funny enough, the system seems to miscalculate the total of the system's size in the loop image. The boot sector size is around 3 MiB, the root size is around 2288 MiB, but the automatic size estimation falls short below this for around 200 ~300 MiB.

Anyway, the output of `stat -f /` is below:

`Block size: 4096`

I don't know if my analysis is correct, but in the end the I managed to make a correct and bootable copy using something like `shrink-backup backup.img 300 --fix` even if when restoring the copy I have to do a manual resize of the root because the script fails to do the autorezisisng.

UnconnectedBedna, 2024-08-02:
Your analysis is kinda correct. And the reason it works if you remove the `-a` option is because then it does not care about block size.

Since the command is returning the correct value of 4096 I have no idea why it does not work. :open_mouth: 
Are you running the latest version of the script? (v1.1)
Because I have made changes to the script in this regard.

If the line I provided returned 4096, it should not be an empty variable in the script.
```
2024-08-01 15:53:11 [DEBUG] - BLOCKSIZE= bytes
```

That variable is set by using the exact line I posted so it is very strange.

Can you confirm you are running the latest version?
`head 4 /path/to/shrink-backup`

mandresve, 2024-08-02:
Yes, I'm running the latest version 1.1. 

I think I have found the reason why `[DEBUG] - BLOCKSIZE= bytes` is empty. If shrink-backup calculates the block value based on the exact response from `stat -f / | grep -i 'Block size' | awk '{print $3}'`, on my system this fails because I have Spanish language configured, and not English, so `grep -i 'Block size'` returns nothing, instead it expects to look for "Tam. Bloque", which would be what actually appears in console.

UnconnectedBedna, 2024-08-02:
GREAT FEEDBACK!!!
Thank you!!
I need to make sure the script is using UTF-8 or something similar, this has not even crossed my mind!
Or use a different method of getting block size, I used to use `dumpe2fs` but that is not always included in all distros, so I changed to `stat` instead...

Let me bite my teeth into this because quite frankly, I do not know how to do this.

An alternative is to remove the block size check completely, it's mostly a leftover from when I first calculated the boot partition and those can be in 512 OR 4096. But I have to make sure other things does not break when doing that.

Let me get back to you on this one, or if you have ideas I am more than open to suggestions.

mandresve, 2024-08-02:
You're welcome! If you need further testing count on me. Love to debug things. 
Maybe you can use `stat -fc %s .`? Returns blocksize no matter the language :3

UnconnectedBedna, 2024-08-02:
Can you try that out?

On line 390 edit and change:
```diff
- BLOCKSIZE=$(stat -f / | grep -i 'Block size' | awk '{print $3}') # bytes
+ BLOCKSIZE=$(stat -fc %s /) # bytes
```

mandresve, 2024-08-02:
I can confirm that it works this way. It seems that the image is a bit bigger than it should be, but it's not a big problem.
[shrink-backup.log](https://github.com/user-attachments/files/16470660/shrink-backup.log)

UnconnectedBedna, 2024-08-02:
It is supposed to be a bit bigger so you can update the img by using `-U`.

It uses resize2fs recommended size. If you want the absolute [smallest img possible](https://github.com/UnconnectedBedna/shrink-backup?tab=readme-ov-file#smallest-image-possible), use manual space set to 0.

I will include this change on the testing branch that will be merged in the v1.2 update.
Thank you for the easy solution!

mandresve, 2024-08-02:
I am glad to have contributed to the solution of this issue, which fortunately was simpler than expected. Thank you for making this tool, I work all the time with SBCs, and only yesterday I found this software. I will continue to use it whenever I get the chance, thank you.

closed 2024-08-02 by mandresve