UnconnectedBedna/shrink-backup #48: Testing branch: shrink-backup.conf <> exclude.txt

sneumeister, 2024-08-06:
Hello,

Downloaded the latest `shrink-backup` from `tesing` branch to continue some tests regarding [umount issue](https://github.com/UnconnectedBedna/shrink-backup/issues/39).

Started shrink-backup with option `-t` which means
```
  -t            Use exclude.txt in same folder as script to set excluded directories
                  One directory per line: "/dir" or "/dir/*" to only exclude contents
``` 
The shrink-backup folder looks like this (`exclude.txt` exists):
```
-rw-r--r-- 1 1024 users 11885129216 Jul 25 23:59 backup_george_bookworm.img
-rw-r--r-- 1 1024 users          83 Aug  1  2023 exclude.txt
-rw-r--r-- 1 1024 users        1539 Jun 25 17:20 LICENSE
-rw-r--r-- 1 1024 users       18192 Jun 25 17:00 README.md
-rwxr-xr-x 1 1024 users       99071 Jun 25 16:58 shrink-backup
-rw-r--r-- 1 1024 users        3342 Aug  6 21:51 shrink-backup.log
-rwxr-xr-x 1 1024 users      100836 Aug  6 21:44 shrink-backup-testing
```

But shrink-backup drops an error:
```
pi@george:/mnt/netbackup/shrink-backup $ sudo /mnt/netbackup/shrink-backup/shrink-backup-testing -atlz /mnt/netbackup/shrink-backup/backup_george_bookworm.img
## Debugging requested, writing to log file: /mnt/netbackup/shrink-backup/shrink-backup.log
## Zoom speed requested...
## Scanning filesystem and calculating...
!! ERROR! -f selected but shrink-backup.conf does not exist!
```
I did not select an `-f` option.

Log-File:
```
2024-08-06 22:01:12 [INFO]  - Debugging requested, writing to log file /mnt/netbackup/shrink-backup/shrink-backup.log
2024-08-06 22:01:12 [INFO]  - Script started with: /mnt/netbackup/shrink-backup/shrink-backup-testing -atlz /mnt/netbackup/shrink-backup/backup_george_bookworm.img
2024-08-06 22:01:12 [INFO]  - Zoom speed requested, setting SLEEPING to empty variable | ZOOM=true
2024-08-06 22:01:12 [INFO]  - -a selected by user, setting ADDED_SPACE to 0 (non-zero value)
2024-08-06 22:01:12 [DEBUG] - ADDED_SPACE=0
2024-08-06 22:01:12 [INFO]  - f2fs root filesystem detected
2024-08-06 22:01:12 [DEBUG] - FSTYPE=f2fs
2024-08-06 22:01:13 [DEBUG] - LOCAL_DEV_PTUUID=655d03f1 | LOCAL_DEV_PATH=/dev/mmcblk0
2024-08-06 22:01:13 [INFO]  - f2fs filesystem or conversion to f2fs requested, disabling autoexpansion
------------------------------------------------------------------------------
2024-08-06 22:01:13 [DEBUG] - IMG_FILE=/mnt/netbackup/shrink-backup/backup_george_bookworm.img
2024-08-06 22:01:13 [DEBUG] - PARTITION_TABLE=msdos
2024-08-06 22:01:13 [DEBUG] - UPDATE=false
2024-08-06 22:01:13 [DEBUG] - AUTORESIZE_RUN=true
2024-08-06 22:01:13 [DEBUG] - PROMPTS=true
2024-08-06 22:01:13 [DEBUG] - EXCLUDE_FILE=true
2024-08-06 22:01:14 [DEBUG] - AUTOEXPAND=false
2024-08-06 22:01:14 [DEBUG] - RSYNC_DELETE=--delete
2024-08-06 22:01:14 [DEBUG] - F2FS=false
2024-08-06 22:01:14 [DEBUG] - TTY_AVAILABILITY=/dev/tty
------------------------------------------------------------------------------
2024-08-06 22:01:14 [INFO]  - -f selected by user, using /mnt/netbackup/shrink-backup/shrink-backup.conf
2024-08-06 22:01:14 [ERROR] - shrink-backup.conf does not exist, exit 3
```
A command 
` $ cp exclude.txt shrink-backup.conf`
solves the problem. 

Is `exclude.txt` now replaced by `shring-backup.conf` ?

UnconnectedBedna, 2024-08-07:
"If using the testing branch, be aware there might be bugs".. xD

If you look at the repo in the testing branch, there are now 2 files, one called exclude.txt and one shrink-backup.conf.

If you keep the branch updated with `git pull` instead of manually downloading the file, you would have gotten both the files and it would have worked.

You can easily switch branch with `git checkout testing` and then back with `git checkout main`. You don't need 2 versions, you can just switch back and forth inside the directory with those 2 commands if you issued a `git clone` when acquiring the script. You can still move the shrink-backup directory by using this method and still issue above commands to change between branches.

But you figured out how to fix it on your system, good job!
You are starting to become a master of this script. :slightly_smiling_face: 

I am working on this. It's because I got a pr with an installation method using curl, and by using that the exclude file needs to be renamed to give the user information on what that file is when located in other directory (/usr/local/etc) than where the script is located.
This will not be the case in the future, so keep both files for now if you want to keep your workflow. :slightly_smiling_face:

I will keep this issue open until I have confirmed all is well with this functionality.

UnconnectedBedna, 2024-10-16:
I have tested the installation script and removed the temporary config file on the testing branch so I am closing this issue.