UnconnectedBedna/shrink-backup #53: image file is double the size of the copied files

MeestorX, 2025-01-02:
**Describe the bug**
image file is over double the size of the disk that it's copying.
The whole size is listed as 2.53G, but the image file is 5349MiB

**To Reproduce**
run the program on a 2.5G disk

**Expected behavior**
2.5G or smaller .img file

**Harware (please complete the following information):**
Rock Pi-S, Armbian 24.8.3

**Additional context**
Here's the last of the log:
```
Number of files: 71,917 (reg: 60,281, dir: 8,382, link: 3,254)
Number of created files: 71,916 (reg: 60,281, dir: 8,381, link: 3,254)
Number of deleted files: 1 (dir: 1)
Number of regular files transferred: 60,277
Total file size: 2.53G bytes
Total transferred file size: 2.53G bytes
Literal data: 2.53G bytes
Matched data: 0 bytes
File list size: 2.23M
File list generation time: 0.005 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 2.53G
Total bytes received: 1.20M

sent 2.53G bytes  received 1.20M bytes  6.91M bytes/sec
total size is **2.53G**  speedup is 1.00
## Rsync done...
## Please stand by...
## Finalizing filesystem...
armbi_root: /lost+found not found.  CREATED.

       71923 inodes used (21.03%, out of 342048)
          33 non-contiguous files (0.0%)
          57 non-contiguous directories (0.1%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 68758/1
      707696 blocks used (51.74%, out of 1367920)
           0 bad blocks
           1 large file

       60277 regular files
        8382 directories
           0 character device files
           0 block device files
           0 fifos
           4 links
        3254 symbolic links (3155 fast symbolic links)
           0 sockets
------------
       71917 files
## Remounting for autoexpansion...
## Mounting img root partition...
## Enabling fs-autoexpand...
## Armbian OS detected...
## Armbian filesystem autoresizing at boot...
## Backup done.
#####################################################################################
## Write to logfile: true                                                           #
## Autoexpand filesystem at boot: true                                              #
## Use exclude.txt: true                                                            #
## /mnt/2025-01-01.img is **5359MiB** with 100MiB [extra space] included                #
## Please wait for the system to reboot after restoring an image with autoexpansion #
#####################################################################################
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 06.59

```

MeestorX, 2025-01-02:
More from the log, Why is LOCAL_AUTORESZIE_MIN set to 6G?

```
2025-01-02 01:33:51 [INFO]  - Running function: get_dev_variables
2025-01-02 01:33:51 [INFO]  - No boot partition detected
2025-01-02 01:33:51 [DEBUG] - LOCAL_DEV_ROOT_PATH=/dev/mmcblk1p1
2025-01-02 01:33:51 [DEBUG] - LOCAL_ROOT_UUID=117b851b-2be0-4bb7-8572-7edce1ef7375 | LOCAL_ROOT_PARTUUID=ec99e6ce-01
2025-01-02 01:33:51 [DEBUG] - LOCAL_ROOT_PARTN=1
2025-01-02 01:33:51 [INFO]  - Calculating size for dd to cover bootsector and adding 5MiB (5242880 bytes) to overlap into root (only used in img creation)
2025-01-02 01:33:51 [DEBUG] - LOCAL_ROOT_START=16777216 bytes | LOCAL_BOOTSECTOR=16776704 bytes | LOCAL_DDBOOTSECTOR=43007 512B blocks
2025-01-02 01:33:51 [INFO]  - Calculating recommended root size
2025-01-02 01:33:51 [DEBUG] - BLOCKSIZE=4096 bytes
2025-01-02 01:33:51 [INFO]  - ext4 filesystem detected, using resize2fs to set recommended root size
2025-01-02 01:33:51 [DEBUG] - LOCAL_AUTORESIZE_MIN=6541316096 bytes
2025-01-02 01:33:51 [INFO]  - Calculating TOTAL (space needed for files on root) by adding LOCAL_USED_SPACE and ADDED_SPACE
2025-01-02 01:33:51 [DEBUG] - LOCAL_USED_SPACE=5363929088 bytes | ADDED_SPACE=104857600 bytes
2025-01-02 01:33:51 [DEBUG] - TOTAL=5468786688 bytes
2025-01-02 01:33:51 [INFO]  - Calculating .img file size by adding LOCAL_BOOTSECTOR to TOTAL
2025-01-02 01:33:51 [DEBUG] - TRUNCATE_TOTAL=5485563392 bytes
2025-01-02 01:33:51 [WARNING] - Adding WIGGLEROOM (128MiB) because manually requested ADDED_SPACE is less than autocaluclated size
2025-01-02 01:33:51 [DEBUG] - AUTORESIZE_WARNING=true | WIGGLEROOM=134217728 bytes | TRUNCATE_TOTAL=5619781120 bytes
2025-01-02 01:33:52 [INFO]  - Do you want to continue? [y/n]
2025-01-02 01:33:54 [INFO]  - Y or y pressed to confirm
```

UnconnectedBedna, 2025-01-02:
Please provide the entire log.

A quick glance tells me the img is 58G, and why `resize2fs` says you need 6G when `df` reports 2.5G used, well, that is a question you have to ask the devs of `resize2fs`. (don't do that, IIRC they have a disclaimer the `-P` function can be hit or miss)
As stated in the log file:
```
2025-01-02 01:33:51 [INFO]  - ext4 filesystem detected, using resize2fs to set recommended root size
2025-01-02 01:33:51 [DEBUG] - LOCAL_AUTORESIZE_MIN=6541316096 bytes
```
 
The exact line is at 398 in shrink-backup:
```
debug 'INFO' 'ext4 filesystem detected, using resize2fs to set recommended root size'
LOCAL_AUTORESIZE_MIN=$(resize2fs -P "$LOCAL_DEV_ROOT_PATH" 2>/dev/null | awk '{print $7}') # blocks
LOCAL_AUTORESIZE_MIN=$(( LOCAL_AUTORESIZE_MIN * BLOCKSIZE )) # bytes
```

You root path gets detected as `2025-01-02 01:33:51 [DEBUG] - LOCAL_DEV_ROOT_PATH=/dev/mmcblk1p1`

So the line used is: `sudo resize2fs -P /dev/mmcblk1p1` (returns blocks) and then calculate how many bytes needed, maybe the variable `BLOCKSIZE` gets set incorrectly, but I don't think so, I think it is `resize2fs` being "safe" and returning a way bigger number than needed.
Try that line on your system and see what it tells you.

I have no control of how `resize2fs` works.
A solution could be to use df or du instead to calculate size (I have to make an option that lets you select), OR you could try manually adding space instead of relying on the automatic calculations, for example: `sudo shrink-backup -tl 2048`
Please see: [Smallest image possible](https://github.com/UnconnectedBedna/shrink-backup?tab=readme-ov-file#smallest-image-possible)

See if the "manually added space" method works first for you and we go from there.

MeestorX, 2025-01-02:
I tried it again earlier, and was able to get an image created. However, what I was looking to do was to "burn" it to the eMMC on a rockpi-s, so without really knowing what I was doing, I dd'd it there, and it didn't boot.
Likely this script isn't really what I'm looking for, which is a way to emulate armbian-install from a commandline.
Sorry for wasting your time!

UnconnectedBedna, 2025-01-02:
That is exactly how it should work, if it doesn't, it IS a bug and you are not wasting my time. :)
But that was not what you reported, you reported the img getting "too big".

The img file you create should be the same files as on the original system. The difference is the partition size gets shrunk.
Then if you do NOT disable the autoresize function (`-e`) when creating/updating the img, the filesystem and partition will be resized at first boot on that img.

If it does not work, I consider that a bug.
So if you want to keep looking at it, I am willing.
But without further information there is nothing I can do to help. :smiley:

MeestorX, 2025-01-02:
Cool. Appreciate the willingness to help!

What I have is a rockpi-s that has a µSD slot and eMMC. I typically use `armbian-install` to copy a working "disk" on the µSD card to the eMMC, which works great. However, I want to automate the process, and `armbian-install` is menu-driven and quite complex.
If I were able to use your program to just do what I need, that would be amazing!
Basically, live-clone a working µSD card while inserted and running on the rockpi-s to its eMMC. If you can let me know how to do that, I will test the shit out of it! :-)

UnconnectedBedna, 2025-01-02:
Oh, NOW I think I understand what you mean.
You change from `mmcblk` to `emmc` as root device?
Hmmmm, I have to look into that.
`shrink-backup` states it supports Armbian, so I should make the time to find out what is going on here.

I saw an email about a comment on armbian forums, was that you? (I need to find credentials to log in there)
I have to make a small disclaimer here, I am not the same person as created the script, that person has sadly passed but was kind enough to prepare and leave credentials for logins and such.
I am the closest relative to him, and his spirit will live on in this script. :heart: 
I just wanted to mention this, because even though I have a TON of documentation he handcrafted and did not put online, it could take a while for me to learn 100% how the script works. Please do not make a thing about it...

MeestorX, 2025-01-02:
No problem. That was me on the forum. I figured there was the place to have a discussion, and just report bugs here. This bug was most likely operator error on my part.
Sorry for your loss, I can empathize and have no issue being patient. I'm currently spending a lot of time trying to strip out everything I DON'T need within `armbian-install` to make it into simple script that will work for me. However, if you're able to make it work with this script, I think it would be more useful as I can also use it to create backup images, which currently is a bit of a PITA.
Let me know if you have something to try and I'll see how it goes. :-)
Happy New Year!

UnconnectedBedna, 2025-01-02:
Thank you! :heart: 

I am trying to find `armbian-install` but all I can find is [armbian-config](https://github.com/armbian/configng/blob/main/bin/armbian-config)
I have no idea what configs I need to change to change boot device on armbian, so if you could provide a link it would probably make things significantly easier for me. :)

MeestorX, 2025-01-02:
Yeah, very odd that `armbian-install` has no repository...
[Here's a link to it](https://www.dropbox.com/scl/fi/7egoot6prweh55sqri503/armbian-install.zip?rlkey=tbodyak5w4munh1ripsbuzq3p&st=mu9ysh2g&dl=1), I assume I'm not breaking any rules by posting it here...

MeestorX, 2025-01-02:
Fair warning, it's REALLY convoluted with not much in the way of commenting.

UnconnectedBedna, 2025-01-02:
```
# Tool to transfer the rootfs of an already running Armbian installation from boot media
# to NAND, eMMC, SATA, NVME or USB storage. In case of eMMC it's also possible to transfer
# the bootloader to eMMC in a single step so from then on running without SD card is
# possible. In case of UEFI it's possible to install it side by side with Windows 10/11 or
# to an empty target.
```

Sounds like what we are looking for. :)

I will look closer at this. But after a first quck glance, this is what you are after right?
Line 888
```
	for choice in $choices
	do
		case $choice in
			1)
				title='MMC (SD/eMMC) boot | USB/SATA/NVMe root install'
				command='Reboot'
				check_partitions
				show_warning "This script will erase your device $DISK_ROOT_PART. Continue?"
				format_disk "$DISK_ROOT_PART"
				create_armbian "" "$DISK_ROOT_PART"
				;;
```

Just a quick question to make sure you actually need all of this.
Have you tried writing the img to emmc WHILE booted with the sd-card, then use `armbian-config` to select emmc as boot device, shutdown the device, remove the sd-card and boot?

I do not own a armbian device with an emmc, only normal sd-card devices so I have never done anything like this.
But as far as I understand it, both sd-cards AND emmc both gets `/dev/mmcblk*` or am I incorrect with that statement?

MeestorX, 2025-01-02:
I'm not sure how you "select emmc as boot device". The way it works is that it boots from the eMMC if there's no SD card inserted. If there's an SD card inserted and it has a bootable image, it boots from that.
The 2 devices cannot have the same UUID, as there will be need to mount the emmc when booting from the SD card, and I think when the UUID match, it puts the mounted file system into read-only.
On the Rockpi-s, mmcblk0 is always the emmc, mmcblk1 is always the SD. However, if you want something more generic, you'll have to determine which is which, even if it has a different name, I would expect...

UnconnectedBedna, 2025-01-02:
Then I'm confused.
What is going wrong if you do what I suggested earlier?
If you write the img to the emmc and remove the sd-card, what goes wrong?

MeestorX, 2025-01-02:
It doesn't boot. But, I never really found out if I was doing the procedure correctly.

1. Use your tool to create an image of the SD card that's currently running on the Rock-Pi, storing it in /mnt/ so it won't get added to the list while running...
2. use `dd if=/mnt/backup.img of=/dev/mmcblk0 BS=8M status=progress` to copy the image file to the emmc
3. Shut down the Rockpi and pull the sd card
4. Power on the RockPi

Does not boot, or at least I cannot SSH to it - not sure what it's doing because it has no monitor. If I knew how to read a `dmesg` from the emmc while booted on the SD card, perhaps it would show if it even tried to boot?

MeestorX, 2025-01-02:
I guess another way to test would be to copy the image file to my computer, use Balena to burn it to an SD card, and then try booting with that SD card. I would assume that would work. Not sure if my `dd` command is the proper way to copy the image to the emmc. Does the image or emmc have to be mounted?

UnconnectedBedna, 2025-01-02:
We have to establish a few things.

- does the backup succeed successfully?
- is the partition and filesystem ok on the emmc after the img is dd:d?
- do you have any way of connecting a monitor to see what is happening at boot?

You should be able to mount the emmc just to check the filesystem, ext4 actually accepts that at runtime, boot, not so much.

Confirming that the img is ok by trying another sd-card and boot from is a very good idea. :)

If it actually does find the emmc and tries to boot but for some reason fails, the logs should still be there if you boot from sd-card and read them.

I again want to emphasize, do the backup with `-e` option to minimize the risks.

MeestorX, 2025-01-02:
Ok, I'll give this a try.
The only way to monitor the boot process is to use a serial logger, and I'm not set up for that. The RockPi-S has no video.
Not sure how to find the boot logs after the fact, but can google how to do that I think.

MeestorX, 2025-01-02:
BTW, the text colouring should be disable-able. It's printing white on white ATM. Lol.
[image]

UnconnectedBedna, 2025-01-02:
Yeah, that looks terrible, good suggestion. I will make sure it happens in the next release.

MeestorX, 2025-01-02:
Ok, using your suggested commandline, I created an image named `/mnt/sd.img`.
I created a test mounting folder called `/media/testmount/`
I cannot mount the image to that folder (I may not be entering the command correctly). I tested mounting the existing emmc which does have a proper file system to /media/testmount/ and it mounted correctly. I then umounted it.
A screenshot:
```
root@rockpi-s:/mnt# ls -l
total 1239324
-rw-r--r-- 1 root root 2106428928 Jan  2 16:51 sd.img

root@rockpi-s:/mnt# ls -l /media/
total 4
drwxr-xr-x 2 root root 4096 Jan  2 16:57 testmount

root@rockpi-s:/mnt# mount /mnt/sd.img /media/testmount
mount: /media/testmount: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.

root@rockpi-s:/mnt# 
```

UnconnectedBedna, 2025-01-02:
Well, you cant mount an img file like that, but you can loop it.
You can use the `--loop` option for that, but that is not necessary.

I added a `--no-color` option in the testing branch btw.
`git checkout testing` to switch if you want.

You should mount the emmc and do a ls to see it looks like your root, and you can check with lsblk and df to see if it looks normal, but I would assume it worked just fine.

I am assuming that `dmesg` is from the system running from the sd-card, and what we want to check is logs from the emmc AFTER you have tried to boot from it.
You might need to invest in a USB TTL to see what is going on at boot, because without that, it will be very hard to figure out.

MeestorX, 2025-01-02:
it would be helpful if you could give me exact commands on how to mount the image so I can check that it's ok - not copying it to the emmc at first. I don't know what `loop` means. If you do, that will save me some googling.

MeestorX, 2025-01-02:
Ok, looked into this.... ran `losetup --find --show /mnt/sd.img`
It only shows the one device, no partition(s).
```
root@rockpi-s:/mnt# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0     2G  0 loop 
mmcblk0     179:0    0   3.6G  0 disk 
└─mmcblk0p1 179:1    0   3.5G  0 part 
mmcblk1     179:32   0  59.5G  0 disk 
└─mmcblk1p1 179:33   0  58.9G  0 part /var/log.hdd
                                      /
zram0       251:0    0 236.8M  0 disk [SWAP]
zram1       251:1    0    50M  0 disk /var/log
zram2       251:2    0     0B  0 disk 
```

UnconnectedBedna, 2025-01-02:
You just mount it normally like any other device.
`lsblk` to find the device then (since I saw you run as root, no sudo needed) `mount /dev/mmcblk?p? /mnt/testmount` (change the mmcblk to the correct you found in `lsblk`)

If you mount the mmc at `/mnt/testmount` you can then use `journalctl -b0 -p3 --file /mnt/testmount/var/log/journal/superlongstringofnumbersandletters/system.journal` to see errors from last boot

MeestorX, 2025-01-02:
Update... I copied the created .img file to my computer, and using Balena, created an SD card and booted with that card, and it works properly, so at least we know the .img is being created properly.

So, just so I'm clear on what you'd like...

1. Copy the .img file to the eMMC. - Please confirm the correct command to do that. I assume it's `dd` but what exactly?
2. Try to boot from the eMMC.
3. If it boots, I'm golden.
4. If it doesn't boot, shut down and reboot with a working SD card, then mount the eMMC and try to find the boot log, based on your instructions above.

Correct?

UnconnectedBedna, 2025-01-02:
The dd line you gave earlier looked good. or maybe 4M BS, but not important.
Otherwise yes, exactly that. :)

MeestorX, 2025-01-02:
Copy. 
BTW, the image I just burned did not auto-expand. Is there a way to expand it's partition while mounted?

MeestorX, 2025-01-02:
Or have it expand when it's being burnt to the SD card within Balena?

UnconnectedBedna, 2025-01-02:
I think you can do it in `armbian-config`, otherwise I linked to the resize service provided by armbian earlier in this thread.

MeestorX, 2025-01-02:
Success!
I took the sd.img file I had created, dd'd it to the eMMC, shut down, rebooted without the SD card and it started up!
So... Something must be wrong with the settings I used previously?

Was it the auto expansion? Why did you suggest to not allow the auto expansion?

UnconnectedBedna, 2025-01-02:
I suggested it to minimize the risk, because I actually do not know if there are things happening on hardware level when you switch from one boot device to another.
Maybe it is that, maybe it is not.
Only way to know is for you to try with an img you create without using the `-e` option. :)

MeestorX, 2025-01-02:
Ok. Certainly having matching UUID's looks to be an issue. It's taking quite a bit of work to try and get the rp to boot from the SD card while it has a matching UUID. Had to run `armbian-install` to fix it.
Is that something you might be able to solve?

MeestorX, 2025-01-02:
Good news! I did an image without the -e and it booted! It's not using ALL of the available space, but it's close, but it's very cool!
If you are able to resolve the duplicate UUID issue, then this could be a winner!

MeestorX, 2025-01-03:
My guess is that you'd have to make a copy of `/boot/armbianEnv.txt`, change the UUID line and use that in the image file. Not sure if `/etc/fstab` would also need to be copied and edited?
Check the `create_armbian()` section of `armbian-install` for some ideas.

UnconnectedBedna, 2025-01-03:
> If you are able to resolve the duplicate UUID issue, then this could be a winner!

Not sure what issue you are referring to, can you clarify?
Sounds to me there never were any issue to begin with but rather some fluke that made it not work the first time.

MeestorX, 2025-01-03:
Oh, I expect it was operator error on my part the first time. The script does what it's supposed to and quite well!
However, for my needs, there's 2 things I still need to figure out:

1. How to use as much of the available space on the device for the partition as possible
2. How to have a different UUID for the image

I understand that these improvements may be out of the scope of what you wish to do with your program, and if so, I'll see if I can figure that out.