UnconnectedBedna/shrink-backup #54: rync failing "out of space"

MeestorX, 2025-01-02:
**Describe the bug**
rsync is failing, complaining to be "out of space", even though the /tmp folder is 8g in size.

**To Reproduce**
run the program

**Expected behavior**
no issues

**Harware (please complete the following information):**
 Rockpi-S, Armbian 24.8.3

**Additional context**
Here's what the log shows:
```
2025-01-02 02:24:27 [DEBUG] - Running: rsync -ahvHAX --exclude-from=./exclude.txt --exclude={/mnt/*,/tmp/backup-ki3,/tmp/rsync-fs3} --info=progress2 --stats --delete --force --partial --delete-excluded / /tmp/backup-ki3
------------------------------------------------------------------------------
2025-01-02 02:30:31 [ERROR] - RSYNC FAILED:
^M          2.46G  97%    6.53MB/s    0:05:59 (xfr#58577, ir-chk=1594/71772)
var/lib/apt/lists/deb.debian.org_debian_dists_bookworm-updates_main_binary-arm64_Packages.diff_Index
^M          2.46G  97%    6.53MB/s    0:05:59 (xfr#58578, ir-chk=1593/71772)
var/lib/apt/lists/deb.debian.org_debian_dists_bookworm-updates_non-free-firmware_binary-arm64_Packages
^M          2.46G  97%    6.53MB/s    0:05:59 (xfr#58579, ir-chk=1592/71772)
var/lib/apt/lists/deb.debian.org_debian_dists_bookworm-updates_non-free_binary-arm64_Packages
^M          2.46G  97%    6.53MB/s    0:05:59 (xfr#58580, ir-chk=1591/71772)
var/lib/apt/lists/deb.debian.org_debian_dists_bookworm_InRelease
^M          2.46G  97%    6.53MB/s    0:05:59 (xfr#58581, ir-chk=1590/71772)
var/lib/apt/lists/deb.debian.org_debian_dists_bookworm_contrib_binary-arm64_Packages
^M          2.46G  97%    6.53MB/s    0:05:59 (xfr#58582, ir-chk=1589/71772)
var/lib/apt/lists/deb.debian.org_debian_dists_bookworm_main_binary-arm64_Packages
^M          2.47G  97%   10.54MB/s    0:00:05  ^M          2.49G  98%    9.35MB/s    0:00:04  rsync: [receiver] write failed on "/tmp/backup-ki3/var/lib/apt/lists/deb.debian.org_debian_dists_bookworm_main_binary-arm64_Packages": No space l>
rsync error: error in file IO (code 11) at receiver.c(380) [receiver=3.2.7]

rsync: [sender] write error: Broken pipe (32)
------------------------------------------------------------------------------
2025-01-02 02:30:31 [ERROR] - Cleanup function called with non zero exit code: exit 1
2025-01-02 02:30:32 [DEBUG] - Unmounting root partition in cleanup function: umount /tmp/backup-ki3
2025-01-02 02:30:33 [DEBUG] - Removing loop in cleanup function: losetup -d /dev/loop0
2025-01-02 02:30:33 [DEBUG] - Removing temp directory in cleanup function: rm -rf /tmp/backup-ki3
2025-01-02 02:30:33 [DEBUG] - Removing temp file in cleanup function: rm /tmp/rsync-fs3
2025-01-02 02:30:33 [INFO]  - Elapsed time: 06.53
2025-01-02 02:30:33 [DEBUG] - Exiting script
##############################################################################
```
df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            164M     0  164M   0% /dev
tmpfs            48M  1.2M   47M   3% /run
/dev/mmcblk1p1   58G  2.5G   55G   5% /
tmpfs           237M     0  237M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           8.0G     0  8.0G   0% /tmp
/dev/zram1       47M  1.1M   43M   3% /var/log
tmpfs            48M     0   48M   0% /run/user/0
```

UnconnectedBedna, 2025-01-02:
You only need to make one issue.
Closing this in favor of #53

closed 2025-01-02 by UnconnectedBedna

UnconnectedBedna, 2025-01-02:
You are correct, I might have been a bit too fast to close it, but I think the problem is that the img file created is 8G, and there can be other sides inside there too.
But I think you said you got it to work.

Could be good to know that the mounts also happen inside of a tmp directory created in `/tmp`, same as some logs from rsync etc.
I would try to create the img file on a physical media instead of a virtual mount.

Create a network server for NFS or samba on another machine and save the img there instead, or just create a directory in `/mnt` or smthn.
https://github.com/UnconnectedBedna/shrink-backup?tab=readme-ov-file#-t-excludetxt
> The folder where the img file is created will ALWAYS be excluded in the backup.
> If -t option is selected, exclude.txt MUST exist (but can be empty) within the directory where the script is located or the script will exit with an error.