UnconnectedBedna/shrink-backup #56: Bug in shrink-backup - Journal is missing after first boot

Awak00m, 2025-01-13:
**Describe the bug**
/var/log/journal link is missing after the first boot, though it is yet in place on burned SD.
/var/log.hdd/journal should be a directory, but it is a link, pointing to itself, but as a directory.
Looks like /var/log/journal is moved to /var/log.hdd/ while the first boot.

**To Reproduce**
Steps to reproduce the behavior:
Just apply the script to existing and correct file system, then burn the image to SD.

**Expected behavior**
A clear and concise description of what you expected to happen.
/var/log/~journal -> /var/log.hdd/journal sub-directory

Instead we get:

/var/log/~journal is absent.
/var/log.hdd/~journal -> /var/log.hdd/journal as a sub-directory.

**Hardware (please complete the following information):**
 - Hardware: Banana Pi M2 zero
 - OS:  armbian-community from [here](https://dl.armbian.com/bananapim2zero/Bookworm_current_minimal) 

**Additional context**
I tied both -e and -az options. No difference.
Screenshots and logs are attached.
[image]
[image]
[image]
[image]
[shrink-backup1.log](https://github.com/user-attachments/files/18401334/shrink-backup1.log)
[shrink-backup2.log](https://github.com/user-attachments/files/18401336/shrink-backup2.log)

UnconnectedBedna, 2025-01-14:
As I said on the Armbian forum, you need to provide proof that the files are missing on the img before you boot.

I did exactly this on a armbian system.
```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0         7:0    0     5G  0 loop 
└─loop0p1   259:0    0     5G  0 part /mnt/temp
mtdblock0    31:0    0     1M  0 disk 
mtdblock1    31:1    0     1M  0 disk 
mmcblk0     179:0    0  14.8G  0 disk 
└─mmcblk0p1 179:1    0  14.6G  0 part /
zram0       252:0    0 492.3M  0 disk [SWAP]
zram1       252:1    0    50M  0 disk /var/log

$ df -h
Filesystem                                         Size  Used Avail Use% Mounted on
udev                                               419M     0  419M   0% /dev
tmpfs                                               99M   13M   87M  13% /run
/dev/mmcblk0p1                                      15G  4.0G  9.8G  29% /
tmpfs                                              493M  736K  492M   1% /dev/shm
tmpfs                                              5.0M  4.0K  5.0M   1% /run/lock
tmpfs                                              493M     0  493M   0% /tmp
/dev/zram1                                          49M   24M   22M  53% /var/log
tmpfs                                               99M     0   99M   0% /run/user/1000

# Is anything mounted from there?
$ findmount | grep /var/log
findmnt | grep /var/log
├─/var/log.hdd                 /dev/mmcblk0p1[/var/log]                          ext4        rw,noatime,errors=remount-ro,commit=600
├─/var/log                     /dev/zram1                                        ext4        rw,relatime,discard

$ ls -l /var/log
total 792
drwxr-xr-x 2 root     root       4096 Jan 14 00:00 apt
-rw-r--r-- 1 root     root          0 Jan 14 00:01 armbian-hardware-monitor.log
-rw-r----- 1 root     adm      136195 Jan 14 01:00 auth.log
-rw------- 1 root     root          0 Jan 14 00:01 boot.log
-rw-rw---- 1 root     utmp          0 Jan 14 00:01 btmp
drwxr-x--- 2 _chrony  _chrony    4096 May 19  2024 chrony
-rw-r----- 1 root     adm      106053 Jan 14 01:00 daemon.log
-rw-r----- 1 root     adm           0 Jan 14 00:01 debug
-rw------- 1 root     root          0 Jan 14 00:01 fail2ban.log
-rw-r----- 1 root     adm         660 Jan 14 00:47 kern.log
-rw-rw-r-- 1 root     utmp     296296 Jan 14 00:54 lastlog
drwxr-x--- 2 www-data www-data   4096 May 19  2024 lighttpd
drwx------ 2 root     root      16384 Jun 21  2024 lost+found
-rw-r----- 1 root     adm        1580 Jan 14 00:47 messages
drwxr-xr-x 2 pihole   pihole     4096 Jan 13 00:00 pihole
lrwxrwxrwx 1 pihole   pihole       23 May 19  2024 pihole-FTL.log -> /var/log/pihole/FTL.log
lrwxrwxrwx 1 pihole   pihole       26 May 19  2024 pihole.log -> /var/log/pihole/pihole.log
drwx------ 2 root     root       4096 May 19  2024 private
drwxr-xr-x 3 root     root       4096 May 19  2024 samba
-rw-r----- 1 root     adm      179295 Jan 14 01:00 syslog
drwxr-xr-x 2 root     root       4096 Aug  1 00:07 sysstat
drwxr-xr-x 2 root     root       4096 May 19  2024 unattended-upgrades
-rw-r----- 1 root     adm           0 Jan 14 00:01 user.log
-rw-rw-r-- 1 root     utmp      28000 Jan 14 00:54 wtmp

$ ls -l /var/log.hdd
total 2784
drwxr-xr-x 2 root     root       4096 Jan 14 00:00 apt
-rw-r--r-- 1 root     root          0 Jan 13 00:00 armbian-hardware-monitor.log
-rw-r--r-- 1 root     root      25217 Jun 22  2024 armbian-hardware-monitor.log.1.gz
-rw-r--r-- 1 root     root      13541 May 19  2024 armbian-hardware-monitor.log.2.gz
-rw-r--r-- 1 root     root     293476 Jan 14 00:01 armbian-ramlog.log
-rw-r----- 1 root     adm      131656 Jan 14 00:00 auth.log
-rw-r----- 1 root     adm      452453 Jan 12 00:00 auth.log.1
-rw-r----- 1 root     adm       22732 Jan  5 00:00 auth.log.2.gz
-rw-r----- 1 root     adm       21450 Dec 29 00:00 auth.log.3.gz
[... 136 lines cut ...]
-rw-r----- 1 root     adm         143 Jan  4 00:00 messages.2.gz
-rw-r----- 1 root     adm         143 Dec 28 00:00 messages.3.gz
-rw-r----- 1 root     adm         143 Dec 21 00:00 messages.4.gz
drwxr-xr-x 2 pihole   pihole     4096 Jan 13 00:00 pihole
lrwxrwxrwx 1 pihole   pihole       23 May 19  2024 pihole-FTL.log -> /var/log/pihole/FTL.log
lrwxrwxrwx 1 pihole   pihole       26 May 19  2024 pihole.log -> /var/log/pihole/pihole.log
drwx------ 2 root     root       4096 May 19  2024 private
drwxr-xr-x 3 root     root       4096 May 19  2024 samba
-rw-r----- 1 root     adm      173202 Jan 14 00:00 syslog
-rw-r----- 1 root     adm      587292 Jan 12 00:00 syslog.1
-rw-r----- 1 root     adm       40287 Jan  5 00:00 syslog.2.gz
-rw-r----- 1 root     adm       38118 Dec 29 00:00 syslog.3.gz
-rw-r----- 1 root     adm       38115 Dec 22 00:00 syslog.4.gz
drwxr-xr-x 2 root     root       4096 Aug  1 00:07 sysstat
drwxr-xr-x 2 root     root       4096 May 19  2024 unattended-upgrades
-rw-r----- 1 root     adm           0 Jan 13 00:00 user.log
-rw-r----- 1 root     adm         236 Jun 22  2024 user.log.1
-rw-r----- 1 root     adm         125 May 19  2024 user.log.2.gz
-rw-rw-r-- 1 root     utmp      27600 Jan 13 23:50 wtmp
```

The files are included and exists on the img.

So please provide this to show if shrink-backup does includes the files.
If it does, it is very likely outside the scope of this script.

Awak00m, 2025-01-14:
>>So please provide this to show if shrink-backup includes the files or not.

My screenshots named "xxxx-before" are just what you want.
Or should I provide just text versions?

/var/log.hdd/ on SD before the first boot :
[image]
/var/log/ on SD before the first boot :
[image]

Awak00m, 2025-01-14:
Here is my "original" lsblk output:
sda           8:0    1 119,4G  0 disk
└─sda1        8:1    1 119,4G  0 part /media
mmcblk0     179:0    0  29,8G  0 disk
└─mmcblk0p1 179:1    0  29,4G  0 part /var/log.hdd
                                      /
zram0       253:0    0 242,7M  0 disk [SWAP]
zram1       253:1    0    50M  0 disk /var/log
zram2       253:2    0     0B  0 disk

This is what is processed by the script. Looks like a correct Armbian SD.
The above is yet before the loop creation, of course.

Here is a text version of loop0p1 /var/log :

-rw-r--r-- 1 root root              5241 янв 13 00:00 alternatives.log
drwxr-xr-x 2 root root              4096 янв 13 14:31 apt
-rw-r--r-- 1 root root            434284 янв 13 17:10 armbian-hardware-monitor.log
-rw-r----- 1 root adm              47201 янв 13 20:45 auth.log
-rw-r--r-- 1 root root             69555 янв 13 00:00 bootstrap.log
-rw-rw---- 1 root utmp              2688 янв 13 00:00 btmp
-rw-r----- 1 root adm              19432 янв 13 20:45 cron.log
-rw-r--r-- 1 root root            198226 янв 13 14:31 dpkg.log
-rw-r--r-- 1 root root                 0 янв 13 00:00 faillog
lrwxrwxrwx 1 root systemd-journal     20 янв 13 17:10 journal -> /var/log.hdd/journal
-rw-r----- 1 root adm             439612 янв 13 20:49 kern.log
-rw-rw-r-- 1 root utmp               292 янв 13 20:15 lastlog
drwx------ 2 root root              4096 янв 13 17:10 lost+found
drwx------ 2 root root              4096 янв  7 07:33 private
lrwxrwxrwx 1 root root                39 янв  7 07:33 README -> ../../usr/share/doc/systemd/README.logs
drwxr-xr-x 3 root root              4096 янв  7 07:33 runit
drwxr-x--- 3 root adm               4096 янв 12 00:00 samba
-rw-r----- 1 root adm             794881 янв 13 20:49 syslog
drwxr-xr-x 2 root root              4096 янв 13 20:19 v2ray
-rw-rw-r-- 1 root utmp             51840 янв 13 20:15 wtmp

Here is a text version of loop0p1 /var/log.hdd :

-rw-r--r-- 1 root root              5241 янв 13 00:00 alternatives.log
drwxr-xr-x 2 root root              4096 янв 13 14:31 apt
-rw-r--r-- 1 root root            386595 янв 13 17:10 armbian-hardware-monitor.log
-rw-r--r-- 1 root root             16158 янв 11 21:50 armbian-hardware-monitor.log.1.gz
-rw-r--r-- 1 root root             15343 янв 13 17:10 armbian-ramlog.log
-rw-r----- 1 root adm              44729 янв 13 17:10 auth.log
-rw-r----- 1 root adm              13301 янв 12 00:00 auth.log.1
-rw-r--r-- 1 root root             69555 янв 13 00:00 bootstrap.log
-rw-rw---- 1 root utmp              2688 янв 13 00:00 btmp
-rw-r----- 1 root adm              18163 янв 13 17:00 cron.log
-rw-r----- 1 root adm              22061 янв 12 00:00 cron.log.1
-rw-r--r-- 1 root root            198226 янв 13 14:31 dpkg.log
-rw-r--r-- 1 root root                 0 янв 13 00:00 faillog
drwxr-sr-x 3 root systemd-journal   4096 янв  8 21:28 journal
-rw-r----- 1 root adm             391649 янв 13 16:34 kern.log
-rw-r----- 1 root adm              62595 янв 11 22:57 kern.log.1
-rw-rw-r-- 1 root utmp               292 янв 13 16:19 lastlog
drwx------ 2 root root              4096 янв  7 07:33 private
lrwxrwxrwx 1 root root                39 янв  7 07:33 README -> ../../usr/share/doc/systemd/README.logs
drwxr-xr-x 3 root root              4096 янв  7 07:33 runit
drwxr-x--- 3 root adm               4096 янв 12 00:00 samba
-rw-r----- 1 root adm             720306 янв 13 17:10 syslog
-rw-r----- 1 root adm             133330 янв 12 00:00 syslog.1
drwxr-xr-x 2 root root              4096 янв 13 14:18 v2ray
-rw-rw-r-- 1 root utmp             48384 янв 13 17:10 wtmp

AFAICS, we have a little difference here:
lrwxrwxrwx 1 root systemd-journal     20 янв 13 17:10 journal -> /var/log.hdd/journal
and
lrwxrwxrwx 1 root root                39 янв  7 07:33 README -> ../../usr/share/doc/systemd/README.logs

So, we could conclude that relative links have been survived, while absolute - not. Is this a way to go?

UnconnectedBedna, 2025-01-14:
As I said, ram or not, does not matter.

One last time.

Provide information EXACTLY like I did in my first response.
Provide proof that the directories or content of /var/log and /var/log.hdd are missing from the image after creation.
I showed you exactly how to do it with my first response.

I am NOT going to guess and try to understand some strange images.
Use text, copy/paste and format as code.
https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

Provide the requested information if you claim something to be a bug or nothing will be done from my side.

I already explained that there is no journal in there on a normal running system, they are in /run on a live system.

Awak00m, 2025-01-15:
Sorry but we don't understand each other...
> Provide proof that the directories or content of /var/log and /var/log.hdd are missing from the image after creation.

I can't do it, because **everything is OK in both the image file and burned SD before the first boot**.
Therefore your actions from your first post prove nothing, as I completely agree and can confirm just the same.

Nevertheless, similar outputs for my **ORIGINAL LIVE SYSTEM** are below:

`lsblk`:
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0   1,8G  0 loop
└─loop0p1   259:0    0   1,8G  0 part /mnt/sda1
sda           8:0    1 119,4G  0 disk
└─sda1        8:1    1 119,4G  0 part /media
mmcblk0     179:0    0  29,8G  0 disk
└─mmcblk0p1 179:1    0  29,4G  0 part /var/log.hdd
                                      /
zram0       253:0    0 242,7M  0 disk [SWAP]
zram1       253:1    0    50M  0 disk /var/log
zram2       253:2    0     0B  0 disk
```
`df-h`:
```
udev               185M            0  185M            0% /dev
tmpfs               49M         2,2M   47M            5% /run
/dev/mmcblk0p1      29G         1,2G   28G            5% /
tmpfs              243M            0  243M            0% /dev/shm
tmpfs              5,0M            0  5,0M            0% /run/lock
tmpfs              243M            0  243M            0% /tmp
/dev/zram1          47M         4,4M   39M           11% /var/log
tmpfs               49M            0   49M            0% /run/user/0
/dev/sda1          117G          55G   57G           50% /media
/dev/loop0p1       1,8G         1,2G  452M           73% /mnt/sda1
```
`findmnt --raw | grep /var/log`:
```
/var/log.hdd /dev/mmcblk0p1[/var/log] ext4 rw,noatime,errors=remount-ro,commit=120
/var/log /dev/zram1 ext4 rw,nosuid,nodev,noexec,relatime,discard
```
Similar outputs for the **DUPLICATED LIVE SYSTEM** are below:
`lsblk`:
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1 119,4G  0 disk
└─sda1        8:1    1 119,4G  0 part /media
mmcblk0     179:0    0  29,8G  0 disk
└─mmcblk0p1 179:1    0   1,8G  0 part /var/log.hdd
                                      /
zram0       253:0    0 242,7M  0 disk [SWAP]
zram1       253:1    0    50M  0 disk /var/log
zram2       253:2    0     0B  0 disk
```
`df-h`:
```
udev               185M            0  185M            0% /dev
tmpfs               49M         2,6M   47M            6% /run
/dev/mmcblk0p1     1,8G         1,2G  451M           74% /
tmpfs              243M            0  243M            0% /dev/shm
tmpfs              5,0M            0  5,0M            0% /run/lock
tmpfs              243M            0  243M            0% /tmp
/dev/zram1          47M         2,8M   41M            7% /var/log
tmpfs               49M            0   49M            0% /run/user/0
/dev/sda1          117G          55G   57G           50% /media
```
`findmnt --raw | grep /var/log`:
```
/var/log.hdd /dev/mmcblk0p1[/var/log] ext4 rw,noatime,errors=remount-ro,commit=120
/var/log /dev/zram1 ext4 rw,nosuid,nodev,noexec,relatime,discard
```
[... 147 lines cut ...]
-rw-rw---- 1 root utmp              2688 янв 14 00:03 btmp
-rw-r----- 1 root adm              21491 янв 15 02:09 cron.log
-rw-r--r-- 1 root root            198226 янв 14 00:03 dpkg.log
-rw-r--r-- 1 root root                 0 янв 14 00:03 faillog
lrwxrwxrwx 1 root systemd-journal     20 янв 13 17:10 journal -> /var/log.hdd/journal
-rw-r----- 1 root adm             531009 янв 14 00:16 kern.log
-rw-rw-r-- 1 root utmp               292 янв 14 00:04 lastlog
drwx------ 2 root root              4096 янв 13 17:10 lost+found
drwx------ 2 root root              4096 янв  7 07:33 private
lrwxrwxrwx 1 root root                39 янв  7 07:33 README -> ../../usr/share/doc/systemd/README.logs
drwxr-xr-x 3 root root              4096 янв  7 07:33 runit
drwxr-x--- 3 root adm               4096 янв 12 00:00 samba
-rw-r----- 1 root adm             938774 янв 15 02:10 syslog
drwxr-xr-x 2 root root              4096 янв 13 20:19 v2ray
-rw-rw-r-- 1 root utmp             57216 янв 14 00:16 wtmp
```
In other words, we see that the **`journal` directory is missing, and it is a broken link, pointing to itself**.

I hope I was as clear as I can.

Obviously you have some different OS. Where you downloaded it from?
Could you test all the same with community version (I specified the link)?
Maybe the problem is just in the OS version?

And thank you for your time!

UnconnectedBedna, 2025-01-15:
> I can't do it, because everything is OK in both the image file and burned SD before the first boot.

Then... What does shrink-backup have to do with any of this?
If the files are the same on the img as on the system you run, what exactly are you trying to say here?

As I have explained multiple times, there IS NO journal in /var/log on a running DEBIAN BASED system, it gets moved when booting and shutting down.
Maybe you have configured something to create that symlink, and that is not correctly configured, I have no idea. But shrink-backup is definitely not responsible for that behavior.
When the script is done, it's done. And if the files are there before boot, they are there. If they gets removed or symlinked or whatever, it has NOTHING to do with shrink-backup.
Even if you have the autoexpansion enabled, it is running code that you already had from your armbian install, the systemd service: `armbian-resize-filesystem.service`

Awak00m, 2025-01-16:
> Then... What does shrink-backup have to do with any of this?

Good question! Should I provide you with a proof of the fact, that contents, produced by `dd` and by your script, differ **AFTER the first boot**?
Are you agree, that it would be a proof?
If so, I can confirm, that `dd if=/dev/mmcblk0p1 of=/dev/sda1 bs=1M` makes the exact copy. Just tried it. :100: 
> Also strange that the dates don't match up, do you create that symlink manually??

Obviously the "bad" symlink's date is different from the original directory's date, since a symlink is just a physical file created sometime during the system startup process. And of course, I never created it manually.
> normally IS NO journal in /var/log on a running DEBIAN BASED system

I can't tell, "how normal" is the "Armbian community" OS version. I just see, what I see. 
I posted the link to the origin.
And I checked all the same behavior on the original downloaded image with the only `armbian-config` been applied in order to turn WiFi on.
>If they gets removed or symlinked or whatever, it has NOTHING to do with shrink-backup.

In my opinion this statement is incorrect, but of course, who am I to argue with you...
Indeed, the problem is covered NOT in the symlink copying itself. It is covered in some other place.
> Just out of curiosity, what do you have in /run/log/journal on the running system?

```
/run/log/journal:
drwxr-s---+ 2 root systemd-journal 60 янв 16 10:01 0ee91095e12a4ef3b0e88b307b0ed90d
```
inside it we have:
`-rw-r-----+ 1 root systemd-journal 638976 янв 16 10:00 system.journal`
```
/var/log.hdd/journal:
drwxr-sr-x+ 2 root systemd-journal 4096 янв 16 00:00 0ee91095e12a4ef3b0e88b307b0ed90d
```
inside it we have:
```
-rw-r-----+ 1 root systemd-journal 1073576 янв 12 00:11 system@71a6efb12d304507bbf076017457af03-0000000000000001-00062b3458d>
-rw-r-----+ 1 root systemd-journal  957904 янв 12 00:36 system@71a6efb12d304507bbf076017457af03-0000000000000647-00062b72f9b>
-rw-r-----+ 1 root systemd-journal 1219288 янв 13 08:58 system@71a6efb12d304507bbf076017457af03-0000000000000ce6-00062b73540>
-rw-r-----+ 1 root systemd-journal  933664 янв 13 12:16 system@71a6efb12d304507bbf076017457af03-00000000000013e9-00062b8e747>
-rw-r-----+ 1 root systemd-journal 1085712 янв 13 14:11 system@71a6efb12d304507bbf076017457af03-0000000000001a66-00062b91375>
-rw-r-----+ 1 root systemd-journal  966112 янв 13 16:14 system@71a6efb12d304507bbf076017457af03-00000000000021b6-00062b92d3c>
-rw-r-----+ 1 root systemd-journal 1057040 янв 14 00:00 system@71a6efb12d304507bbf076017457af03-000000000000275c-00062b948ab>
-rw-r-----+ 1 root systemd-journal 1059904 янв 15 00:00 system@71a6efb12d304507bbf076017457af03-0000000000002e90-00062b9b0d2>
-rw-r-----+ 1 root systemd-journal 1056544 янв 15 03:51 system@71a6efb12d304507bbf076017457af03-000000000000345c-00062baf2b0>
-rw-r-----+ 1 root systemd-journal 1050624 янв 16 00:00 system@71a6efb12d304507bbf076017457af03-0000000000003b6b-00062bb2683>
-rw-r-----+ 1 root systemd-journal 2621440 янв 16 08:12 system.journal
```
> Sure, buy me one and I'll try it for you. xD

Buy what? Would you like to say that the same OS version has different structure on different hardware?
Yet I could agree, that the structure may vary between 32-bit and 64-bit CPU...
What you did show in your first answer, was definitely NOT a **fresh** Armbian community OS, right?

UnconnectedBedna, 2025-01-16:
> Good question! Should I provide you with a proof of the fact, that contents, produced by dd and by your script, differ AFTER the first boot?
> Are you agree, that it would be a proof?

Nope, **because a `dd` is a CLONE**, not a copy, not a backup, A CLONE BIT FOR BIT of the system you run. Completely different from rsync with shrink-backup **that is a COPY of SELECTED FILES**.
There seems to be something refusing to boot unless you shut down your system CORRECTLY. I have pointed you in a direction 4-5 times of what I think it is.

> since a symlink is just a physical file created sometime during the system startup process.

No they are not, they are exactly like files or directories on linux. When they are created, they are created. NOTHING in a linux filesystem is changed randomly at boot. A symlink is just a file with a different magic bit (or sticky bit or whatever it's called).
I created these symlinks a few years ago, very strange that they are not "recreated" every reboot, or you think I haven't rebooted since 2022?
```
$ ls -l                                                                                                                          ✔ 
total 0
lrwxrwxrwx 1 bedna media 31 Nov  6  2022  Albums -> /media/diverse/media/mp3/Albums
lrwxrwxrwx 1 bedna media 39 Nov  6  2022 'Sorterade mp3' -> '/media/diverse/media/mp3/Sorterade mp3/'
```

> Buy what? Would you like to say that the same OS version has different structure on different hardware?

Yes, of course... They are different releases... Did... did you not install a release for your specific hardware?!?
I didn't check your link since installing an OS has nothing to do with shrink-backup.

> What you did show in your first answer, was definitely NOT a fresh Armbian community OS, right?

It was not, but that does not matter at all. All shrink-backup does is copy selected files to an img.
I asked you for output to compare, and you delivered that.
After looking at files on the img and on your system, all files are there after the backup is done.
IDK how many times I have to hint and point you in the direction of strange configuration on your side.

But your question is a good one. Have YOU ensured a fresh install of this particular distro acts this way with a fresh install and a reboot? You are the one claiming there is a bug in shrink-backup, YOU have to provde proof. The test I did here obv worked as intended. So if something is not working on your side, the logical reason is something is different on YOUR side, or there would be more reports of the script not working.
I have still to see some proof that shrink-backup is causing any issues.

I will give you a hint, again, because I can be wrong. As you pointed out, the version I run is not the latest.
Arbian is forked from raspberry pi os and modified to become armbian, and the latest release of rpiOS DO utilize `/var/log/journal` while booted (bullseye does not). So guess what is inside `/run/log/journal`? Absolutely nothing.