UnconnectedBedna/shrink-backup #57: LOOP paths can not be set

fbernier, 2025-04-01:
**Describe the bug**
I'm running `sudo ./shrink-backup -atl /mnt/thelio/derp.img` on an OrangePi 5 Max running Armbian. This is the output:

```
fbernier@orangepi5-max:~/code/shrink-backup$ sudo ./shrink-backup -atl /mnt/thelio/derp.img
## Debugging requested, writing to log file: ./shrink-backup.log
## Zoom speed NOT requested...
## Scanning filesystem and calculating...
#####################################################################################
# A backup will be created at:                                                      #
# /mnt/thelio/derp.img                                                              #
# ext4 filesystem detected on root                                                  #
# --------------------------------------------------------------------------------- #
# Write to logfile:              true                                               #
# Zoom speed requested:          false                                              #
# Autocalculate img root size:   true                                               #
# Autoexpand filesystem at boot: true                                               #
# Use exclude.txt:               true                                               #
# Bootsector size:               15MiB                                              #
# Estemated root usage:          4832MiB                                            #
# Auto calculated root size:     5449MiB                                            #
# Total img size:                5465MiB                                            #
#####################################################################################
## Do you want to continue? [y/n] y

## Creating bootsector...
43007+0 records in
43007+0 records out
22019584 bytes (22 MB, 21 MiB) copied, 0.258881 s, 85.1 MB/s
## Resizing img file...
## Looping img file...
!! LOOP paths can not be set, retrying in 5 seconds...
!! LOOP paths can not be set, retrying in 5 seconds...
!! LOOP paths can not be set, retrying in 5 seconds...
!! LOOP PATHS CAN NOT BE SET!!!
!! Cleanup function called with non zero exit code, something went wrong!!!
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.26
```

This is the log output:

```
2025-04-01 11:42:33 [INFO]  - Debugging requested, writing to log file ./shrink-backup.log
2025-04-01 11:42:33 [INFO]  - Script started with: ./shrink-backup -atl /mnt/thelio/derp.img
2025-04-01 11:42:33 [INFO]  - Zoom speed NOT requested, setting SLEEPING=sleep 1 | ZOOM=false
2025-04-01 11:42:34 [INFO]  - -a selected by user, setting ADDED_SPACE to 0 (non-zero value)
2025-04-01 11:42:34 [DEBUG] - ADDED_SPACE=0
2025-04-01 11:42:34 [INFO]  - ext4 root filesystem detected
2025-04-01 11:42:34 [DEBUG] - FSTYPE=ext4
2025-04-01 11:42:35 [DEBUG] - LOCAL_DEV_PTUUID=7c5901dd-e168-8d4d-bf36-cba98647cc49 | LOCAL_DEV_PATH=/dev/mmcblk0
------------------------------------------------------------------------------
2025-04-01 11:42:35 [DEBUG] - INSTALL_METHOD=default
2025-04-01 11:42:35 [DEBUG] - IMG_FILE=/mnt/thelio/derp.img
2025-04-01 11:42:35 [DEBUG] - PARTITION_TABLE=gpt
2025-04-01 11:42:35 [DEBUG] - UPDATE=false
2025-04-01 11:42:35 [DEBUG] - AUTORESIZE_RUN=true
2025-04-01 11:42:35 [DEBUG] - PROMPTS=true
2025-04-01 11:42:35 [DEBUG] - EXCLUDE_FILE=true
2025-04-01 11:42:35 [DEBUG] - AUTOEXPAND=true
2025-04-01 11:42:35 [DEBUG] - RSYNC_DELETE=--delete
2025-04-01 11:42:35 [DEBUG] - F2FS=false
2025-04-01 11:42:35 [DEBUG] - TTY_AVAILABILITY=/dev/tty
------------------------------------------------------------------------------
2025-04-01 11:42:35 [INFO]  - -f selected by user, using ./exclude.txt
2025-04-01 11:42:35 [DEBUG] - ./exclude.txt exists
2025-04-01 11:42:35 [INFO]  - Running function: make_img
------------------------------------------------------------------------------
2025-04-01 11:42:35 [INFO]  - Running function: get_dev_variables
2025-04-01 11:42:35 [INFO]  - No boot partition detected
2025-04-01 11:42:35 [DEBUG] - LOCAL_DEV_ROOT_PATH=/dev/mmcblk0p1
2025-04-01 11:42:35 [DEBUG] - LOCAL_ROOT_UUID=c86e76c4-665f-48c5-bf6d-b17e035aa2a3 | LOCAL_ROOT_PARTUUID=18708472-0093-014f-887d-a51820368a0a
2025-04-01 11:42:35 [DEBUG] - LOCAL_ROOT_PARTN=1
2025-04-01 11:42:35 [INFO]  - Calculating size for dd to cover bootsector and adding 5MiB (5242880 bytes) to overlap into root (only used in img creation)
2025-04-01 11:42:35 [DEBUG] - LOCAL_ROOT_START=16777216 bytes | LOCAL_BOOTSECTOR=16776704 bytes | LOCAL_DDBOOTSECTOR=43007 512B blocks
2025-04-01 11:42:35 [INFO]  - Calculating recommended root size
2025-04-01 11:42:35 [DEBUG] - BLOCKSIZE=4096 bytes
2025-04-01 11:42:35 [INFO]  - ext4 filesystem detected, using resize2fs to set recommended root size
2025-04-01 11:42:35 [DEBUG] - LOCAL_AUTORESIZE_MIN=5714165760 bytes
2025-04-01 11:42:35 [INFO]  - Setting TOTAL (space needed for files on root) to autocalculated size
2025-04-01 11:42:35 [DEBUG] - TOTAL=5714165760 bytes
2025-04-01 11:42:35 [INFO]  - Calculating .img file size by adding LOCAL_BOOTSECTOR to TOTAL
2025-04-01 11:42:35 [DEBUG] - TRUNCATE_TOTAL=5730942464 bytes
2025-04-01 11:42:35 [INFO]  - Do you want to continue? [y/n]
2025-04-01 11:42:37 [INFO]  - Y or y pressed to confirm
------------------------------------------------------------------------------
2025-04-01 11:42:38 [INFO]  - Using dd to create bootsector
------------------------------------------------------------------------------
2025-04-01 11:42:38 [DEBUG] - Running: dd bs=512 count=43007 if=/dev/mmcblk0 of=/mnt/thelio/derp.img conv=noerror,sync status=progress
43007+0 records in
43007+0 records out
22019584 bytes (22 MB, 21 MiB) copied, 0.258881 s, 85.1 MB/s
------------------------------------------------------------------------------
2025-04-01 11:42:40 [INFO]  - Using truncate to resize img file to 5465MiB
2025-04-01 11:42:40 [DEBUG] - Running: truncate --size=5730942464 /mnt/thelio/derp.img
2025-04-01 11:42:41 [INFO]  - Running function: do_loop
2025-04-01 11:42:41 [DEBUG] - LOOP=/dev/loop0
2025-04-01 11:42:42 [DEBUG] - Running: losetup -P /dev/loop0 /mnt/thelio/derp.img
2025-04-01 11:42:42 [INFO]  - Running function: set_img_variables
2025-04-01 11:42:42 [INFO]  - No boot partition detected
2025-04-01 11:42:44 [DEBUG] - IMG_DEV_ROOT_PATH=
2025-04-01 11:42:44 [WARNING] - LOOP paths can not be set, retrying in 5 seconds
2025-04-01 11:42:44 [WARNING] - IMG_DEV_ROOT_PATH=
2025-04-01 11:42:49 [WARNING] - LOOP paths can not be set, retrying in 5 seconds
2025-04-01 11:42:49 [WARNING] - IMG_DEV_ROOT_PATH=
2025-04-01 11:42:54 [WARNING] - LOOP paths can not be set, retrying in 5 seconds
2025-04-01 11:42:54 [WARNING] - IMG_DEV_ROOT_PATH=
2025-04-01 11:42:59 [ERROR] - LOOP PATHS CAN NOT BE SET
2025-04-01 11:42:59 [ERROR] - Cleanup function called with non zero exit code: exit 1
2025-04-01 11:42:59 [DEBUG] - Removing loop in cleanup function: losetup -d /dev/loop0
2025-04-01 11:42:59 [INFO]  - Elapsed time: 00.26
2025-04-01 11:42:59 [DEBUG] - Exiting script
##############################################################################
```

I've been banging my head on this for a few hours now and would love some insights on how to debug this if you have any.

UnconnectedBedna, 2025-04-01:
Hello.

```
> sudo fdisk -lo start /mnt/thelio/derp.img 
> GPT PMBR size mismatch (122224639 != 11333422) will be corrected by write.
> Disk /mnt/thelio/derp.img: 5.4 GiB, 5802712576 bytes, 11333423 sectors
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: dos
> Disk identifier: 0x00000000
> 
> Start
>     1
```

Correlation vs causation.
The filesystem on the img has not yet been formatted so that the filesystem is incorrect at this stage is to be expected.

The problem is:
`2025-04-01 11:42:44 [DEBUG] - IMG_DEV_ROOT_PATH=`

This is an error that has been haunting shrink-backup for a long time (check closed issues with the exact same problem if you are curious), the closest to an explanation is running over the network for _some_ people.
Are you running this over network?

Can you confirm there is no boot partition on your OPI?

Can you confirm these are correct for your system? Ie, is the root partition on your OPI starting at 16777216bytes?
```
2025-04-01 11:42:35 [DEBUG] - LOCAL_ROOT_START=16777216 bytes | LOCAL_BOOTSECTOR=16776704 bytes | LOCAL_DDBOOTSECTOR=43007 512B blocks
```

`sudo fdisk -lo start /dev/mmcblk0` and then multiply that number by 512 should end up with 16777216

If you are running over network, can you verify if it fails the same way if you write the img to a usb stick?

fbernier, 2025-04-01:
yeah, running over the network. I'll definitely give a usb stick a try in a hour or so.

The root partition of my OPI does start at precisely 16777216 bytes and there's no boot partition.

UnconnectedBedna, 2025-04-01:
You comment arrived JUST as I made an edit above, so please try that too.

> One thing that suddenly hit me I have not thought of. Mount whatever you have at /mnt/thlio with -o sync and retry. If it is a network mount from another linux system, it might still be in THAT systems cache, ie the file is not yet done physically writing the data to the file.
> I am not sure the sync option that is included in the dd command will take effect on a remote mount unless also mounted with sync option.
> I could be wrong here, but it's worth a try.

Please let me know the result(s)

fbernier, 2025-04-01:
Spoke too soon... the resize took much longer so I though it was working but same error even with 30 seconds timeouts :/

UnconnectedBedna, 2025-04-01:
Closing this as solved

UnconnectedBedna, 2025-04-01:
I guess we reopen it again... xD

UnconnectedBedna, 2025-05-24:
@fbernier  this keeps bugging me, I keep thinking about how to possibly mitigate this and have yet more ideas, but since I don't have any way to force this to happen I have no way of testing it, but you do...

I was thinking of the usage of `oflag=sync` instead of `conv=fsync`, since most boot sectors are pretty small, the speed decrease is probably minimal, but it should be tested to see if it even mitigates the error or not or it makes no sense.

I understand you have a lot of stuff to do, but *if* you manage to find some time, I would really appreciate if we could do some testing with this and see if we can come up with a solution.
Thanks.

fkiraz, 2025-06-04:
I am on orange pi 5 plus, freshly installed Armbian 24 on EMMC, same problem. While script is running, loop0 seems wrong.

Command was `./shrink-backup -a /tmp/backup.img` with root

```
fdisk -l /dev/loop0

GPT PMBR size mismatch (488554495 != 4584302) will be corrected by write.
Disk /dev/loop0: 2.19 GiB, 2347163136 bytes, 4584303 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device       Boot Start     End Sectors  Size Id Type
/dev/loop0p1          1 4584302 4584302  2.2G ee GPT
```

UnconnectedBedna, 2025-06-04:
@fkiraz The fact it says it's a mismatch is probably not so strange since the root filesystem has not yet been formatted if it fails at the same point as op.
But all systems are different, do you have a boot partition for example?

A logfile and a `lsblk -f` to see your device setup would be useful.
Please create a new issue and provide information there.

closed 2025-08-02 by UnconnectedBedna

fbernier, 2025-08-27:
@UnconnectedBedna sorry for not getting back... I had a lot going on but I want to thank you for following through on this and solving it.

I just tried it and it _almost_ worked. There's just one failure towards the end(which doesn't prevent the script from executing to completion): `ln: failed to create symbolic link '/tmp/backup-ejy/etc/systemd/system/basic.target.wants/armbian-resize-filesystem.service': No such file or directory`

I do have this service on my system:

```
fbernier@orangepi5-max ~/c/shrink-backup (testing) [2]> sudo systemctl status armbian-resize-filesystem.service
[sudo] password for fbernier:
○ armbian-resize-filesystem.service - Armbian filesystem resize
     Loaded: loaded (/usr/lib/systemd/system/armbian-resize-filesystem.service; disabled; preset: enabled)
     Active: inactive (dead)
```

I'll try to solve it and make a PR when I have time but I'm not sure when that's gonna be so let me know if you want me to open an issue instead.

Thanks again