UnconnectedBedna/shrink-backup #58: Text Colors

MeestorX, 2025-05-17:
**Describe the bug**
If terminal window has a white background, the text is not readable

**To Reproduce**
Run a terminal window with a white background

**Expected behavior**
Text should be black on white

**Harware (please complete the following information):**
 - Hardware: Anything
 - OS: Any

**Additional context**
Please add a commandline switch that disables the textcoloring, or a light/dark mode.

UnconnectedBedna, 2025-05-18:
This is already on the testing branch but it's not easy to know unless you read commits since I forgot to add the info on the readme, witch I now corrected.

I would appreciate feedback if it works in your situation. 😃 

If you used git to install, please see https://github.com/UnconnectedBedna/shrink-backup/wiki/Installing#switch-branch on how to switch branch.

MeestorX, 2025-05-18:
TY! Working well. :-)

closed 2025-05-18 by MeestorX