UnconnectedBedna/shrink-backup #59: File too large/FAT32

MeestorX, 2025-05-19:
**Describe the bug**
Here's what I see when trying to write an image to a FAT32 USB stick. (It's blank and formatted FAT32)

```
!! Removing old img file...
## Creating bootsector...
19526144 bytes (20 MB, 19 MiB) copied, 1 s, 19.5 MB/s22019584 bytes (22 MB, 21 MiB) copied, 1.11597 s, 19.7 MB/s

43007+0 records in
43007+0 records out
22019584 bytes (22 MB, 21 MiB) copied, 2.01507 s, 10.9 MB/s
## Resizing img file...
truncate: failed to truncate '/media/usb/RP-Sat-sd.img' at 5952187904 bytes: File too large
!! TRUNCATE FAILED!!!
!! Cleanup function called with non-zero exit code, something went wrong!!!
## Exiting and cleaning up...
## Please stand by...
## Done.
```

**To Reproduce**
Try to create an img file written to a USB stick

**Expected behavior**
Should work?

**Harware (please complete the following information):**
 - Hardware: RockPi-S
 - OS: Debian Bookworm (Armbian 25.x.x)

**Additional context**
Add any other context about the problem here.
Run in debug mode by using `-l` option and please provide the output of shrink-backup.log

MeestorX, 2025-05-19:
Is writing to FAT32 supported?

UnconnectedBedna, 2025-05-20:
If a filesystem is mountable/writable is outside the scope of shrink-backup. So if you can mount and write files to it, shrink-backup should work.

> `truncate: failed to truncate '/media/usb/RP-Sat-sd.img' at 5952187904 bytes: File too large`

Seems the usb stick is full.
You can see how much space is left on it with `df` with the stick mounted. (witch I think you should have included in your initial bug report)

MeestorX, 2025-05-20:
```
df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            159M     0  159M   0% /dev
tmpfs            47M  4.7M   42M  11% /run
/dev/mmcblk1p1   59G  5.4G   52G  10% /
tmpfs           234M     0  234M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           234M     0  234M   0% /tmp
/dev/zram1       47M   27M   17M  62% /var/log
tmpfs            47M     0   47M   0% /run/user/0
/dev/sda1        32G   16K   32G   1% /media/usb
```
```
./shrink-backup -alt /media/usb/RP-Sat-sd.img
## Debugging requested, writing to log file: ./shrink-backup.log
## Zoom speed NOT requested...
## Scanning filesystem and calculating...
#####################################################################################
# A backup will be created at:                                                      #
# /media/usb/RP-Sat-sd.img                                                          #
# ext4 filesystem detected on root                                                  #
# --------------------------------------------------------------------------------- #
# Write to logfile:              true                                               #
# Zoom speed requested:          false                                              #
# rsync tty output:              true                                               #
# Autocalculate img root size:   true                                               #
# Autoexpand filesystem at boot: true                                               #
# Use exclude.txt:               true                                               #
# Bootsector size:               15MiB                                              #
# Estemated root usage:          5528MiB                                            #
# Auto calculated root size:     6399MiB                                            #
# Total img size:                6415MiB                                            #
#####################################################################################
## Do you want to continue? [y/n] y

## Creating bootsector...
20504576 bytes (21 MB, 20 MiB) copied, 1 s, 20.5 MB/s22019584 bytes (22 MB, 21 MiB) copied, 1.07155 s, 20.5 MB/s

43007+0 records in
43007+0 records out
22019584 bytes (22 MB, 21 MiB) copied, 1.98315 s, 11.1 MB/s
## Resizing img file...
truncate: failed to truncate '/media/usb/RP-Sat-sd.img' at 6727228928 bytes: File too large
!! TRUNCATE FAILED!!!
!! Cleanup function called with non-zero exit code, something went wrong!!!
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.09
```
```
cat shrink-backup.log
2025-05-20 04:49:27 [INFO]  - Debugging requested, writing to log file ./shrink-backup.log
2025-05-20 04:49:27 [INFO]  - Script started with: ./shrink-backup -alt /media/usb/RP-Sat-sd.img
2025-05-20 04:49:27 [INFO]  - Zoom speed NOT requested, setting SLEEPING=sleep 1 | ZOOM=false
2025-05-20 04:49:28 [INFO]  - -a selected by user, setting ADDED_SPACE to 0 (non-zero value)
2025-05-20 04:49:28 [DEBUG] - ADDED_SPACE=0
2025-05-20 04:49:28 [INFO]  - ext4 root filesystem detected
2025-05-20 04:49:28 [DEBUG] - FSTYPE=ext4
2025-05-20 04:49:28 [DEBUG] - LOCAL_DEV_PTUUID=4a64f710 | LOCAL_DEV_PATH=/dev/mmcblk1
------------------------------------------------------------------------------
2025-05-20 04:49:28 [DEBUG] - INSTALL_METHOD=default
2025-05-20 04:49:28 [DEBUG] - IMG_FILE=/media/usb/RP-Sat-sd.img
2025-05-20 04:49:28 [DEBUG] - PARTITION_TABLE=msdos
2025-05-20 04:49:28 [DEBUG] - UPDATE=false
2025-05-20 04:49:28 [DEBUG] - AUTORESIZE_RUN=true
2025-05-20 04:49:28 [DEBUG] - PROMPTS=true
2025-05-20 04:49:28 [DEBUG] - RSYNC_TTY=true
2025-05-20 04:49:28 [DEBUG] - EXCLUDE_FILE=true
[... 20 lines cut ...]
2025-05-20 04:49:29 [INFO]  - Setting TOTAL (space needed for files on root) to autocalculated size
2025-05-20 04:49:29 [DEBUG] - TOTAL=6710452224 bytes
2025-05-20 04:49:29 [INFO]  - Calculating .img file size by adding LOCAL_BOOTSECTOR to TOTAL
2025-05-20 04:49:29 [DEBUG] - TRUNCATE_TOTAL=6727228928 bytes
2025-05-20 04:49:29 [INFO]  - Do you want to continue? [y/n]
2025-05-20 04:49:31 [INFO]  - Y or y pressed to confirm
------------------------------------------------------------------------------
2025-05-20 04:49:32 [INFO]  - Using dd to create bootsector
------------------------------------------------------------------------------
2025-05-20 04:49:34 [DEBUG] - Running: dd bs=512 count=43007 if=/dev/mmcblk1 of=/media/usb/RP-Sat-sd.img conv=noerror,sync status=progress
43007+0 records in
43007+0 records out
22019584 bytes (22 MB, 21 MiB) copied, 1.98315 s, 11.1 MB/s
------------------------------------------------------------------------------
2025-05-20 04:49:36 [INFO]  - Using truncate to resize img file to 6415MiB
2025-05-20 04:49:36 [DEBUG] - Running: truncate --size=6727228928 /media/usb/RP-Sat-sd.img
------------------------------------------------------------------------------
2025-05-20 04:49:36 [ERROR] - TRUNCATE FAILED:
truncate: failed to truncate '/media/usb/RP-Sat-sd.img' at 6727228928 bytes: File too large
------------------------------------------------------------------------------
2025-05-20 04:49:36 [ERROR] - Cleanup function called with non zero exit code: exit 1
2025-05-20 04:49:36 [INFO]  - Elapsed time: 00.09
2025-05-20 04:49:36 [DEBUG] - Exiting script
##############################################################################
```

UnconnectedBedna, 2025-05-20:
I didn't actually check the size you were trying to squeeze onto the fat32, I assumed you knew there is a limitation.
Your img is 6415MiB, max size on fat32 is 4G.

Use exFAT instead.

closed 2025-05-20 by UnconnectedBedna

MeestorX, 2025-05-20:
Gotcha. TY.

MeestorX, 2025-05-20:
Reformatted to exFAT, but the write fails with broken pipe errors.
I can successfully create an image on the same device I'm making the image from, but not having any luck creating it on a USB stick.
e.g.:

```
2025-05-20 14:49:57 [INFO]  - Running function: do_mount
2025-05-20 14:49:58 [INFO]  - Creating temp directory
2025-05-20 14:49:58 [DEBUG] - Running: mktemp -d -t backup-XXX
2025-05-20 14:49:58 [DEBUG] - TMP_DIR=/tmp/backup-aaL
2025-05-20 14:50:00 [INFO]  - Mounting root partition from loop
2025-05-20 14:50:00 [DEBUG] - Running: mount /dev/loop0p1 /tmp/backup-aaL
2025-05-20 14:50:00 [INFO]  - Backing up files
2025-05-20 14:50:01 [INFO]  - Running function: do_rsync
2025-05-20 14:50:02 [DEBUG] - Backing up to /media/usb/media/usb/RP-Sat-sd.img
2025-05-20 14:50:02 [INFO]  - Creating temporary file to store rsync output
2025-05-20 14:50:02 [DEBUG] - Running: mktemp -t rsync-XXX
2025-05-20 14:50:02 [DEBUG] - tmp_file=/tmp/rsync-sOj
2025-05-20 14:50:02 [DEBUG] - Running: rsync -ahvHAX --exclude-from=./exclude.txt --exclude={/media/usb/*,/tmp/backup-aaL,/tmp/rsync-sOj} --info=progress2 --stats --delet>
------------------------------------------------------------------------------
2025-05-20 15:02:25 [ERROR] - RSYNC FAILED:
PIPESTATUS=0
^M          6.24G  98%    8.06MB/s    0:12:18 (xfr#69791, to-chk=58/82984)
var/log.hdd/faillog
^M          6.24G  98%    8.06MB/s    0:12:18 (xfr#69792, to-chk=57/82984)
var/log.hdd/kern.log
^M          6.24G  98%    8.06MB/s    0:12:18 (xfr#69793, to-chk=56/82984)
var/log.hdd/kern.log.1
^M          6.25G  98%    8.06MB/s    0:12:18 (xfr#69794, to-chk=55/82984)
var/log.hdd/lastlog
^M          6.25G  98%    8.05MB/s    0:12:19 (xfr#69795, to-chk=54/82984)
var/log.hdd/syslog
^M          6.25G  98%    8.05MB/s    0:12:19 (xfr#69796, to-chk=53/82984)
var/log.hdd/syslog.1
^M          6.25G  98%    2.08MB/s    0:00:32  ^M          6.27G  99%    6.46MB/s    0:00:07  ^M          6.27G  99%    5.88MB/s    0:00:07  rsync: [receiver] write faile>
rsync error: error in file IO (code 11) at receiver.c(381) [receiver=3.2.7]

rsync: [sender] write error: Broken pipe (32)
------------------------------------------------------------------------------
2025-05-20 15:02:25 [ERROR] - Script failed during rsync, cleanup exit 6
2025-05-20 15:02:25 [DEBUG] - Unmounting root partition in cleanup function: umount /tmp/backup-aaL
2025-05-20 15:02:25 [DEBUG] - Removing loop in cleanup function: losetup -d /dev/loop0
2025-05-20 15:02:26 [DEBUG] - Removing temp directory in cleanup function: rm -rf /tmp/backup-aaL
2025-05-20 15:02:26 [DEBUG] - Removing temp file in cleanup function: rm /tmp/rsync-sOj
2025-05-20 15:02:26 [INFO]  - Elapsed time: 12.53
2025-05-20 15:02:26 [DEBUG] - Exiting script
```