UnconnectedBedna/shrink-backup #60: Is the script trying to write and backup the log at the same time?

MeestorX, 2025-05-20:
**Describe the bug**
Getting a "no space left on device" where the /tmp/ folder is reportedly running out of space, even though there's plenty of space. Seems it's having trouble writing the log while it's also trying to back it up?
**To Reproduce**
Do a backup with -l enabled

**Expected behavior**
Should complete without error

**Harware (please complete the following information):**
 - Hardware: RockPi-S
 - OS: Debian Bookworm

**Additional context**
```
usr/local/bin/shrink-backup/shrink-backup.log
        237.99K   0%   27.96MB/s    0:00:00 (xfr#1, ir-chk=1066/20166)rsync: [receiver] write failed on "/tmp/backup-UgD/usr/local/bin/shrink-backup/shrink-backup.log": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(381) [receiver=3.2.7]
!! RSYNC FAILED!!!
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 00.20

root@rp-satellite-xx:/usr/local/bin/shrink-backup# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            159M     0  159M   0% /dev
tmpfs            47M  4.7M   42M  11% /run
/dev/mmcblk1p1   59G   12G   46G  21% /
tmpfs           234M     0  234M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           2.0G  4.0K  2.0G   1% /tmp
/dev/zram1       47M   32M   12M  74% /var/log
tmpfs            47M     0   47M   0% /run/user/0
```

UnconnectedBedna, 2025-05-20:
Look. Making more bug reports does not make it a bug in shrink-backup.
I told you in #59 it fails during rsync.
You probably have an unstable system/problems with cache/lack of memory or something similar, you have to figure out what it is. This has nothing to do with shrink-backup.

If you think it's a bug of some sort, witch I **HIGHLY** doubt, you should turn to rsync and report whatever bug you think this is.
But first, just google the error code, there are thousands of hits with this particular problem.

closed 2025-05-20 by UnconnectedBedna

MeestorX, 2025-05-20:
Just looking for some help with this script - should I use the discussion tab instead?

UnconnectedBedna, 2025-05-20:
Yes, that would be smarter, creating an issue when asking for help is pretty bad manners.

MeestorX, 2025-05-20:
Noted. TY