UnconnectedBedna/shrink-backup #62: Backup fails with Broken Pipe on Orange Pi with Snap installed

GJSchaller, 2025-06-30:
**Describe the bug**
Running shrink-backup on two nearly identical Pi setups (DNS1 & DNS2).  One of the devices is failing on backup, apparently due to Snap being installed - DNS1 uses this to push / sync PiHole settings to DNS2.

**To Reproduce**
Tested with Testing branch - run snap-backup /path/to/image.img, same issue occurs with -- fix as well.

**Expected behavior**
Backup should complete successfully.

**Harware (please complete the following information):**
 - Hardware: Orange Pi 3
 - OS: Debian for Orange Pi 3 from Orange Pi site

**Additional context**
This may be due to the Snap package installed on DNS1, which is not installed on DNS2.

[shrink-backup.log](https://github.com/user-attachments/files/20985110/shrink-backup.log)

GJSchaller, 2025-06-30:
I don't want to completely exclude Snap, obviously, but if there's a way to exclude the problematic path without killing the functionality of the backup image when I restore it, that works for me.

If there's a different / better way to install Snap, that also works.

UnconnectedBedna, 2025-06-30:
Since it came to light on the armbian foum post you made that you are using snap on one of the "identical" setups, you have to establish that the image actually does not get completely filled, and according to logs it does seem to get filled, or at least very close to:
```
          4.18G  80%   19.95MB/s    0:00:51  rsync: [receiver] write failed on "/tmp/backup-MIx/var/lib/snapd/cache/96eb00bfe832b7a3411e3ecc7008509075c65aff4ea30d8fe1b78568a44b7908afcd4f0db22370a640daf6f7af754c45": No space left on device (28)
```

And according to your log you make root partition 4774006784bytes by adding 1024MiB
```
2025-06-30 13:29:57 [DEBUG] - LOCAL_USED_SPACE=3700264960 bytes | ADDED_SPACE=1073741824 bytes
2025-06-30 13:29:57 [DEBUG] - TOTAL=4774006784 bytes
```
4774006784bytes = 4,44614028931Gigabytes

I am not 100% sure, but rsync might be reporting in Gibibytes here, and 4.18Gibibytes = 4.488240824Gigabytes, ie VERY close to what the root partition is.

You can use `sudo /root/shrink-backup-testing/shrink-backup --loop /mnt/backups/PiHole/dns1.img`
And then mount the root partition of the loop somewhere and check with `df` to find out if there are space left on the image or not.

Or you can try with adding more than 1024, like 2048 to see if it still fails just to make sure it's not actually related to you completely filling the image with snap stuff.

GJSchaller, 2025-06-30:
OK, adding extra space (8192) fixed this - I'll experiment with lowering that value until I hit an error, and go back up again.

I had thought Snaps was just an extra package, and wasn't aware how complex it was, or that I was cutting so close to the limit with space.  Thank you for your guidance and support!

UnconnectedBedna, 2025-06-30:
> I had thought Snaps was just an extra package, and wasn't aware how complex it was, or that I was cutting so close to the limit with space. Thank you for your guidance and support!

As I said, I have not used it myself, but it seems logical to me you don't want that data on your backup anyway since snap will most likely just mount over that data anyway when creating new mounts when you boot.
But you should look into how snaps work and not take my word for it.

closed 2025-06-30 by UnconnectedBedna