UnconnectedBedna/shrink-backup #63: Boot found in fstab but partition not mounted, please mount and retry script

mmattel, 2025-07-05:
**Describe the bug**
Running `sudo ./shrink-backup -ty /mnt/backup/rp_bkp.img 3072` -->

`Boot found in fstab but partition not mounted, please mount and retry script`

**Expected behavior**
The system is setup with Debian 12 and runs/boots flawless.
No clue what the script tries to tell me...?

**Harware (please complete the following information):**
 - Hardware: Raspberrypi 4
 - OS: Debian GNU/Linux 12 (bookworm)

**Additional context**
```
cat /etc/fstab

proc                  /proc  proc    defaults          0       0
PARTUUID=899c9e5d-01  /boot  vfat    defaults          0       2
PARTUUID=899c9e5d-02  /      ext4    defaults,noatime  0       1
```

robertstrom, 2025-07-07:
Hello,

I have been using PiShrink for several years now (started using it in 2020 when I first started using Pi's) and never knew there was anything better. Stumbled across this yesterday and have been playing around with it with mixed results.

I have a Pi5 that is running Armbian that i have been able to image and restore from image successfully, even testing the update capability ... successfully!, (the update capability is **amazing**).

Where things stray from success are when imaging a Pi5 that has Kali Linux installed.

I got the error mentioned in this issue title when I first attempted the image. I then switched to the **testing** branch and was able to image the system (unknown OS was reported).

I am not, however, able to restore the image successfully. The image is not booting. The attached image is what the Pi looks like after restoring the image to a MicroSD card (same exact brand, model, and size).

Command used to back up the Kali Linux Pi was:

```
sudo ./shrink-backup/shrink-backup -al /mnt/SMBMount/kali-rpi-case-2025-07-06.img
```

I also tried adding the `-e` argument to disable the auto resize on boot and that did not help any

Command used to restore the Kali Linux Pi was:

```
sudo dd if=/mnt/SMBMount/kali-rpi-case-2025-07-06.img of=/dev/sda bs=4M status=progress
```

 
[image]

Here's what the fidsk of that card looks like. Also showing that the testing branch is up to date.

Would love to use this to backup and restore my Kali Linux Pi devices. IS this possible? If so, how?

As a side note, I started trying to use **shrink-backup** saving to a network location via sshfs. That _**did NOT**_ go well at all. Extremely, extremely slow! I switched to mounting the NAS using SMB and it was much, much faster. I'd rather do the save via sshfs if it is possible to speed things up.

Thanks for an amazing script!

```
 sudo fdisk -l /dev/sda
Disk /dev/sda: 116.11 GiB, 124675686400 bytes, 243507200 sectors
Disk model: SD  Transcend   
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5a444a2f

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sda1         2048   524287   522240  255M  c W95 FAT32 (LBA)
/dev/sda2       524288 59661022 59136735 28.2G 83 Linux

~/shrink-backup   main  git pull
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 6 (delta 4), reused 0 (delta 0), pack-reused 0 (from 0)
Unpacking objects: 100% (6/6), 2.43 KiB | 310.00 KiB/s, done.
From https://github.com/UnconnectedBedna/shrink-backup
   9845e51..f788cb4  testing    -> origin/testing
Already up to date.

```

[image]

mmattel, 2025-07-07:
RPi 4, Debian 12, using the recent changes mounting the boot partition if not mounted in the `testing` branch:
```
git checkout testing
git pull
sudo ./shrink-backup -t /mnt/backup/rp-bkp.img 3072 
## Zoom speed NOT requested...
!! Boot found in fstab but partition not mounted...
## Do you want to mount boot partition? [y/n]
!! Mounting boot partition to /boot...
## Scanning filesystem and calculating...
## Raspberry pi detected
...
```
😃 
Note that you should add `shrink-backup.log` to `.gitignore` which is necessary when using sb via git clone...

UnconnectedBedna, 2025-07-07:
> Note that you should add shrink-backup.log to .gitignore which is necessary when using sb via git clone...

No, users making changes will have to reset hard or stash changes. Changes to that file is extremely rare so this is close to a non-issue.
I made a change 2 days ago on testing branch for the first time in 2 years (I think), and when testing is pushed to main in next release, I will make notes about it in the patch-notes.

> This RPi runs HomeAssistant via docker see error lines /var/lib/docker/overlay2....
> If I run from the shell:
> ls /var/lib/docker/overlay2... --> permission denied
> but if I do:
> sudo ls /var/lib/docker/overlay2... --> I get a proper result
> I guess that not only the script needs to be run with sudo, but, depending how the code is setup, the rsync commands need a preceeding sudo too... just my 2c

By starting the script with sudo, everything in the script runs with sudo (or root credentials rather).

> Note that you should add shrink-backup.log to .gitignore which is necessary when using sb via git clone...

No, files not tracked by git will not interfere with anything. The logfile is not tracked.

```
var/lib/docker/overlay2/02076b6bcf0d6ecc2c7cf9149fb09c8ce6dd1e6d65fe23735b0d0cc9ef4e7016/diff/usr/local/lib/python3.13/site-packages/hass_frontend/static/translations/page-onboarding/en-GB-6bc0129abf22deba973fc79c18590bbb.json
[sender] io timeout after 30 seconds -- exiting
rsync error: timeout in data send/receive (code 30) at io.c(201) [sender=3.2.7]
rsync error: received SIGUSR1 (code 19) at main.c(1613) [generator=3.2.7]
```

Stop the docker container and confirm rsync is still hanging.
Your transfer speed is extremely low, 2,68MB/s.
This looks like there is something happening outside the scope of shrink-backup.

<hr>

But nothing of this has anything to do with the original issue, that boot was unmounted.
What I am primarily interested in is if shrink-backup somehow caused the boot to become unmounted, and since you omitted the top of the logfile I can not confirm it being remounted again.
I know the mounting works, I tested it myself, what I want to know is if shrink-backup is unmounting it so dont unmount just to show it being remounted again.
Please confirm that boot is mounted, then rerun the script to see if it becomes unmounted.
You don't need to complete the entire backup if you don't want to. When the copy operation with rsync occurs, you can press ctrl+c.
And please provide entire log of the operation including a lsblk before you start the script to show boot is mounted.

mmattel, 2025-07-08:
I added a PR based on experiences to improve sb, see #65
Note that using the change in #65 made the script passing for the first time 😃 

```
lsblk

NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0 17,8G  0 loop 
├─loop0p1   259:0    0  256M  0 part /tmp/backup-EcG/boot
└─loop0p2   259:3    0 17,6G  0 part /tmp/backup-EcG
mmcblk0     179:0    0 57,6G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 57,4G  0 part /
```

```
sudo ./shrink-backup -tyl --fix /mnt/backup/rp-bkp.img 6142

...
Number of files: 648.805 (reg: 549.849, dir: 76.895, link: 21.997, dev: 44, special: 20)
Number of created files: 648.389 (reg: 549.437, dir: 76.891, link: 21.997, dev: 44, special: 20)
Number of deleted files: 1 (dir: 1)
Number of regular files transferred: 545.512
Total file size: 16,81G bytes
Total transferred file size: 16,40G bytes
Literal data: 16,40G bytes
Matched data: 0 bytes
File list size: 17,43M
File list generation time: 274,636 seconds
File list transfer time: 0,000 seconds
Total bytes sent: 16,44G
Total bytes received: 11,25M

sent 16,44G bytes  received 11,25M bytes  5,00M bytes/sec
total size is 16,81G  speedup is 1,02
## Rsync done...
## Please stand by...
## Finalizing filesystem...
rootfs: /lost+found not found.  CREATED.

      644476 inodes used (56.01%, out of 1150560)
         401 non-contiguous files (0.1%)
         267 non-contiguous directories (0.0%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 624334/53
     4541697 blocks used (98.70%, out of 4601574)
           0 bad blocks
           1 large file

      545512 regular files
       76893 directories
          43 character device files
           1 block device file
          14 fifos
        3925 links
       21997 symbolic links (20016 fast symbolic links)
           6 sockets
------------
      648391 files
## Remounting for autoexpansion...
## Mounting img root partition...
## Mounting img boot partition...
## Enabling autoexpansion for Raspberry pi...
## Raspberry pi filesystem autoresizing at boot...
## Backup done.
#####################################################################################
# Backup location: /mnt/backup/rp-bkp.img                                           #
# Write to logfile: true                                                            #
# Autoexpand filesystem at boot: true                                               #
# Use exclude.txt: true                                                             #
[... 78 lines cut ...]
2025-07-08 10:12:47 [DEBUG] - Running: mount /dev/loop0p2 /tmp/backup-EcG
2025-07-08 10:12:48 [INFO]  - Mounting boot inside root
2025-07-08 10:12:48 [DEBUG] - Running: mount /dev/loop0p1 /tmp/backup-EcG/boot
2025-07-08 10:12:48 [INFO]  - Running autoexpand function
2025-07-08 10:12:49 [INFO]  - Running function: autoexpansion_rpi
2025-07-08 10:12:49 [DEBUG] - Systemd basic.target.wants directory does not exist, running: mkdir /tmp/backup-EcG/etc/systemd/system/basic.target.wants
2025-07-08 10:12:49 [DEBUG] - Creating systemd unit expand-fs.service
2025-07-08 10:12:49 [DEBUG] - Enabling systemd service by creating symlink: ln -s /etc/systemd/system/expand-fs.service /tmp/backup-EcG/etc/systemd/system/basic.target.wants/expand-fs.service
2025-07-08 10:12:49 [DEBUG] - Creating expansion script /tmp/backup-EcG/expand-fs.sh
2025-07-08 10:12:49 [DEBUG] - Making expand-fs.sh executable
2025-07-08 10:12:49 [INFO]  - Raspberry pi filesystem autoresizing at boot
2025-07-08 10:12:51 [INFO]  - Img file created
2025-07-08 10:12:51 [DEBUG] - /mnt/backup/rp-bkp.img is 18234MiB with a root partition of 17974MiB including 6142MiB [extra space]
2025-07-08 10:12:51 [INFO]  - Backup done
-------------------------------------------------------------------------------------
2025-07-08 10:12:51 [DEBUG] - Cleanup function called with exit 0
2025-07-08 10:12:51 [DEBUG] - Unmounting boot partition in cleanup function: umount /tmp/backup-EcG/boot
2025-07-08 10:12:51 [DEBUG] - Unmounting root partition in cleanup function: umount /tmp/backup-EcG
2025-07-08 10:12:51 [DEBUG] - Removing loop in cleanup function: losetup -d /dev/loop0
2025-07-08 10:12:51 [DEBUG] - Removing temp directory in cleanup function: rm -rf /tmp/backup-EcG
2025-07-08 10:12:51 [DEBUG] - Removing temp file in cleanup function: rm /tmp/rsync-6M2
2025-07-08 10:12:51 [INFO]  - Elapsed time: 01.21
2025-07-08 10:12:51 [DEBUG] - Exiting script
#####################################################################################
```

UnconnectedBedna, 2025-07-08:
Thank you! 👍 

```
2025-07-08 09:11:31 [INFO]  - Separate boot partition detected
2025-07-08 09:11:31 [DEBUG] - BOOT_PATH=/boot
2025-07-08 09:11:31 [INFO]  - ext4 root filesystem detected
2025-07-08 09:11:31 [DEBUG] - FSTYPE=ext4
2025-07-08 09:11:32 [DEBUG] - LOCAL_DEV_PTUUID=899c9e5d | LOCAL_DEV_PATH=/dev/mmcblk0
2025-07-08 09:11:33 [INFO]  - Raspberry pi detected
```

There is no remounting of boot so the pr was a red herring, something else caused your system to unmount boot.
Closing this issue.