UnconnectedBedna/shrink-backup #64: kali linux-arm

UnconnectedBedna, 2025-07-07:
> Hello,
> 
> I have been using PiShrink for several years now (started using it in 2020 when I first started using Pi's) and never knew there was anything better. Stumbled across this yesterday and have been playing around with it with mixed results.
> 
> I have a Pi5 that is running Armbian that i have been able to image and restore from image successfully, even testing the update capability ... successfully!, (the update capability is **amazing**).
> 
> Where things stray from success are when imaging a Pi5 that has Kali Linux installed.
> 
> I got the error mentioned in this issue title when I first attempted the image. I then switched to the **testing** branch and was able to image the system (unknown OS was reported).
> 
> I am not, however, able to restore the image successfully. The image is not booting. The attached image is what the Pi looks like after restoring the image to a MicroSD card (same exact brand, model, and size).
> 
> Command used to back up the Kali Linux Pi was:
> 
> ```
> sudo ./shrink-backup/shrink-backup -al /mnt/SMBMount/kali-rpi-case-2025-07-06.img
> ```
> 
> I also tried adding the `-e` argument to disable the auto resize on boot and that did not help any
> 
> Command used to restore the Kali Linux Pi was:
> 
> ```
> sudo dd if=/mnt/SMBMount/kali-rpi-case-2025-07-06.img of=/dev/sda bs=4M status=progress
> ```
> 
>  
> [image]
> 
> Here's what the fidsk of that card looks like. Also showing that the testing branch is up to date.
> 
> Would love to use this to backup and restore my Kali Linux Pi devices. IS this possible? If so, how?
> 
> As a side note, I started trying to use **shrink-backup** saving to a network location via sshfs. That _**did NOT**_ go well at all. Extremely, extremely slow! I switched to mounting the NAS using SMB and it was much, much faster. I'd rather do the save via sshfs if it is possible to speed things up.
> 
> Thanks for an amazing script!
> 
> ```
>  sudo fdisk -l /dev/sda
> Disk /dev/sda: 116.11 GiB, 124675686400 bytes, 243507200 sectors
> Disk model: SD  Transcend   
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: dos
> Disk identifier: 0x5a444a2f
> 
> Device     Boot  Start      End  Sectors  Size Id Type
> /dev/sda1         2048   524287   522240  255M  c W95 FAT32 (LBA)
> /dev/sda2       524288 59661022 59136735 28.2G 83 Linux
> 
> 
> ~/shrink-backup   main  git pull
> remote: Enumerating objects: 8, done.
> remote: Counting objects: 100% (8/8), done.
> remote: Compressing objects: 100% (6/6), done.
> remote: Total 6 (delta 4), reused 0 (delta 0), pack-reused 0 (from 0)
> Unpacking objects: 100% (6/6), 2.43 KiB | 310.00 KiB/s, done.
> From https://github.com/UnconnectedBedna/shrink-backup
>    9845e51..f788cb4  testing    -> origin/testing
> Already up to date.
> 
> ```
> 
> 
> [image] 

 _Originally posted by @robertstrom in [#63](https://github.com/UnconnectedBedna/shrink-backup/issues/63#issuecomment-3043191812)_

I have a sneaking suspicion on what might have happened, the boot became unmounted again somewhere before the rsync operation happened, witch led to all files on boot being erased by rsync.
Looks a bit strange though because it seems to be formatting your boot: `mkfs.fat....`
You can use `shrink-backup --loop /mnt/SMBMount/kali-rpi-case-2025-07-06.img` and then mount the boot partition to see if my suspicion is correct.
**You did run the script with -l option, so please provide log file.**

Please also copy paste/text and use the code formatting feature on github instead of images in the future.

Besides that, shrink-backup has never been tested on kali-arm so there is no autoexpansion function for it, hence "unknown os", I checked their build script on gitlab and there does not seem to be any encryption or anything so it _should_ theoretically work. But maybe there is, because I just glanced at the repo, and if there is encryption, shrink-backup will not work. (but your fdisk output does not seem to indicate any encryption, lsblk would be better though)
I also saw they have their own expansion method so when I get time, I will look deeper into this.
Therefore I added the enhancement label instead of bug. I will need to do an installation of kali-arm to test functionality and create a function to use the correct method for autoexpansion. This will be done on a rpi4 though, not rpi5 since I don't own any.

As for sshfs that is part of  how that system works, it uses SSH encryption and therefore becomes way slower, no way around it.
I would recommend using NFS rather than samba though if you are mounting another linux system and not windows. (I assume your NAS is running linux and should support NFS)

UnconnectedBedna, 2025-07-07:
> I made changes to that file yesterday. Adding that to .gitignore would mean updating with `git pull` would not get those changes.

I made changes to that file 2 days ago: https://github.com/UnconnectedBedna/shrink-backup/commit/e84b8c35ec2b2f659e3ca3776912aa9ab44a4274
Changes to that file is **extremely rare**, and if no changes are made, changes users have made themselves will not be over written.
Information about changes will be in the patch notes when I push the next release.
Also, if you make changes in main branch to that file, switching to testing means all files change to what is in your testing branch, ie the default if unchanged before.

I would appreciate if you kept to #63 since this one is about kali-ARM.

mmattel, 2025-07-07:
> I would appreciate if you kept to...

you are right, sorry 🤦‍♂

robertstrom, 2025-07-08:
> [@robertstrom](https://github.com/robertstrom) I can confirm using shrink-backup with kali-ARM does not work. I have made [inquiries](https://forums.kali.org/t/how-does-the-root-partition-get-mounted-on-the-pre-built-arm-images/10180) about this issue and will come back with information when I know more.

Are you still wanting me to do any of what you asked for above (mounting the image file, sending the logs)?

You are amazingly responsive and detailed in your responses! Huge kudos to you for that and this great backup script!

Thanks,

Robert

UnconnectedBedna, 2025-07-09:
I have more information.

I managed to get a working backup. But I experienced the EXACT SAME THING as you did, and my suspicion about boot not being mounted WAS correct.

The thing is, it became unmounted DURING the rsync process when no unmount commands had yet been issued by shrink-backup so it's IMPOSSIBLE that shrink-backup did the unmounting.
Nothing of this makes any sense AT ALL and is VERY frustrating because it is clearly outside the scope of shrink-backup and might fool users to think they are safe when they are not because boot became unmounted randomly during the backup!

You can read a longer explanation about it in the earlier linked post at kali forums.

I will ad ANOTHER check of boot after the backup is done and report in red text that the boot partition became unmounted sometime during the backup and that user should remount and try again.

I have ran the script multiple times after it randomly became unmounted and they all succeeded without it unmounting again.

But bottom line, shrink-backup _works_ on kali-ARM, but that bug _can_ present itself at any moment, and I have no idea how to figure out how to stop it. All I can do is implement a check when backup is done and then inform users to rerun with -U to include all files from /boot/firmware.

Kali-ARM also have a systemd service that autoexpands the filesystem at boot and I will include that in the script so autoexpansion can be utilized.

Stay tuned.

UnconnectedBedna, 2025-07-10:
did you do a `git pull` too? Without that, you are not up to date with the new implementations where it does an extra check for the partition when the backup is done.

robertstrom, 2025-07-11:
Created a brand new image straight from the Kali Pi download using the xzcat command documented on their website. Booted that system and did git clone of the shrink-backup repo, did a `git pull origin testing` switched to the testing branch, ran shrink-backup, imaged the `.img` file using the command:

```
sudo dd if=/mnt/SMBMount/kali-rpi-case-2025-07-10.img of=/dev/sda bs=4M status=progress
```

Here is the log file from the backup using shrink-backup

```
#####################################################################################
2025-07-11 01:48:13 [INFO]  - Debugging requested, writing to log file ./shrink-backup/shrink-backup.log
2025-07-11 01:48:13 [INFO]  - Script started with: ./shrink-backup/shrink-backup -al /mnt/SMBMount/kali-rpi-case-2025-07-10.img
2025-07-11 01:48:13 [INFO]  - Zoom speed NOT requested, setting SLEEPING=sleep 1 | ZOOM=false
2025-07-11 01:48:14 [INFO]  - -a selected by user, setting ADDED_SPACE to 0 (non-zero value)
2025-07-11 01:48:14 [INFO]  - Separate boot partition detected
2025-07-11 01:48:14 [DEBUG] - BOOT_PATH=/boot/firmware
2025-07-11 01:48:14 [INFO]  - ext4 root filesystem detected
2025-07-11 01:48:14 [DEBUG] - FSTYPE=ext4
2025-07-11 01:48:14 [DEBUG] - LOCAL_DEV_PTUUID=f6ec0804 | LOCAL_DEV_PATH=/dev/mmcblk0
2025-07-11 01:48:14 [INFO]  - Kali-ARM detected
-------------------------------------------------------------------------------------
2025-07-11 01:48:15 [DEBUG] - INSTALL_METHOD=default
2025-07-11 01:48:15 [DEBUG] - OS=kali
2025-07-11 01:48:15 [DEBUG] - IMG_FILE=/mnt/SMBMount/kali-rpi-case-2025-07-10.img
2025-07-11 01:48:15 [DEBUG] - PARTITION_TABLE=msdos
2025-07-11 01:48:15 [DEBUG] - UPDATE=false
2025-07-11 01:48:15 [DEBUG] - AUTORESIZE_RUN=true
2025-07-11 01:48:15 [DEBUG] - PROMPTS=true
2025-07-11 01:48:15 [DEBUG] - RSYNC_TTY=true
2025-07-11 01:48:15 [DEBUG] - EXCLUDE_FILE=false
2025-07-11 01:48:15 [DEBUG] - AUTOEXPAND=true
2025-07-11 01:48:15 [DEBUG] - RSYNC_DELETE=--delete
2025-07-11 01:48:15 [DEBUG] - F2FS_CONVERSION=false
2025-07-11 01:48:15 [DEBUG] - TTY_AVAILABILITY=/dev/tty
2025-07-11 01:48:15 [DEBUG] - ADDED_SPACE=0
-------------------------------------------------------------------------------------
2025-07-11 01:48:15 [INFO]  - -f NOT selected by user, using default exclude directories
2025-07-11 01:48:15 [INFO]  - Running function: check_dependencies
2025-07-11 01:48:15 [INFO]  - Running function: make_img
-------------------------------------------------------------------------------------
2025-07-11 01:48:15 [INFO]  - Running function: dev_variables
2025-07-11 01:48:15 [DEBUG] - BOOT_PARTITION=true | LOCAL_DEV_BOOT_PATH=/dev/mmcblk0p1 | LOCAL_DEV_ROOT_PATH=/dev/mmcblk0p2
2025-07-11 01:48:15 [DEBUG] - LOCAL_BOOT_UUID=B9E0-4008 | LOCAL_ROOT_UUID=e0671880-863e-426e-a118-51b8ee235c3e | LOCAL_ROOT_PARTUUID=f6ec0804-02
2025-07-11 01:48:15 [DEBUG] - LOCAL_ROOT_PARTN=2
2025-07-11 01:48:15 [INFO]  - Calculating size for dd to cover bootsector and adding 5MiB (5242880 bytes) to overlap into root (only used in img creation)
2025-07-11 01:48:15 [DEBUG] - LOCAL_ROOT_START=268435456 bytes | LOCAL_BOOTSECTOR=268434944 bytes | LOCAL_DDBOOTSECTOR=534527 512B blocks
2025-07-11 01:48:15 [INFO]  - Calculating recommended root size
2025-07-11 01:48:15 [DEBUG] - BLOCKSIZE=4096 bytes
2025-07-11 01:48:15 [INFO]  - ext4 filesystem detected, using resize2fs to set recommended root size
2025-07-11 01:48:15 [DEBUG] - LOCAL_AUTORESIZE_MIN=14617870336 bytes
2025-07-11 01:48:15 [INFO]  - Setting TOTAL (space needed for files on root) to autocalculated size
2025-07-11 01:48:15 [DEBUG] - TOTAL=14617870336 bytes
2025-07-11 01:48:15 [INFO]  - Calculating .img file size by adding LOCAL_BOOTSECTOR to TOTAL
2025-07-11 01:48:15 [DEBUG] - TRUNCATE_TOTAL=14886305280 bytes
2025-07-11 01:48:15 [INFO]  - Do you want to continue? [y/n]
2025-07-11 01:48:22 [INFO]  - Y or y pressed to confirm
-------------------------------------------------------------------------------------
2025-07-11 01:48:23 [INFO]  - Using dd to create bootsector
2025-07-11 01:48:23 [DEBUG] - Running: dd bs=512 count=534527 if=/dev/mmcblk0 of=/mnt/SMBMount/kali-rpi-case-2025-07-10.img conv=noerror,sync status=progress
-------------------------------------------------------------------------------------
2025-07-11 01:48:31 [DEBUG] - dd completed:
534527+0 records in
534527+0 records out
273677824 bytes (274 MB, 261 MiB) copied, 7.70994 s, 35.5 MB/s
-------------------------------------------------------------------------------------
2025-07-11 01:48:33 [INFO]  - Using truncate to resize img file to 14196MiB
2025-07-11 01:48:33 [DEBUG] - Running: truncate --size=14886305280 /mnt/SMBMount/kali-rpi-case-2025-07-10.img
2025-07-11 01:48:34 [INFO]  - Running function: do_loop
2025-07-11 01:48:34 [DEBUG] - LOOP=/dev/loop0
2025-07-11 01:48:35 [DEBUG] - Running: losetup -P /dev/loop0 /mnt/SMBMount/kali-rpi-case-2025-07-10.img
[... 28 lines cut ...]
2025-07-11 01:48:47 [INFO]  - Creating temp directory
2025-07-11 01:48:47 [DEBUG] - Running: mktemp -d -t backup-XXX
2025-07-11 01:48:47 [DEBUG] - TMP_DIR=/tmp/backup-0RC
2025-07-11 01:48:49 [INFO]  - Mounting root partition from loop
2025-07-11 01:48:49 [DEBUG] - Running: mount /dev/loop0p2 /tmp/backup-0RC
2025-07-11 01:48:50 [INFO]  - Mounting boot inside root
2025-07-11 01:48:50 [INFO]  - Boot directory did not exist on img
2025-07-11 01:48:50 [DEBUG] - Running: mkdir -p /tmp/backup-0RC/boot/firmware
[... cut ...]
2025-07-11 01:48:51 [DEBUG] - tmp_file=/tmp/rsync-lNv
2025-07-11 01:48:51 [DEBUG] - Running: rsync -ahvHAX --exclude={/lost+found,/proc/*,/sys/*,/dev/*,/tmp/*,/run/*,/mnt/*,/media/*,/var/swap,/snap/*,/mnt/SMBMount/*} --info=progress2 --stats --delete --force --partial --delete-excluded --timeout=30 / /tmp/backup-0RC
```

I did **absolutely NOTHING else to the image** before running shrink-backup so this is just the base Kali image.

Let me know what else I can do to help troubleshoot.

Thanks!

Robert

robertstrom, 2025-07-11:
I updated the testing branch

```
git pull origin testing
``` 

made sure that the testing branch was active

```
git checkout testing
```

ran shrink-backup and restored the backup to another MicroSD card. Unfortunately it did not boot (I tried a few times and waited for a while each time).

I know you're not a fan of pictures but I have no other way to communicate what I am seeing when the system does not boot.

[image]

I've got to get to working now. Will post more information later.

UnconnectedBedna, 2025-07-11:
> I know you're not a fan of pictures but I have no other way to communicate what I am seeing when the system does not boot.

It's fine, but from now on you can call that "failing like in original post" (unless you get another error message when booting). xD

And that is what I got when the boot was empty/partial. It's looking for boot on all available devices and partitions, that is what the text means.
So my suspicion is that it was STILL unmounted during rsync and therefore deleted files on img boot partition.
The script must also have failed to detect this.
So for me to be able to do anything about this, you need to provide information about this.
Did it really not warn you about this at the end of the script with capital red text?
```
!! BOOT PARTITION BECAME UNMOUNTED DURING BACKUP!!"
!! RE-RUN SHRINK-BACKUP WITH -U OPTION!!
```

Look in log file, it is labeled as `ERROR - Boot partition became unmounted during backup`

Can you please check with the method I mentioned earlier? Loop img, mount boot and make sure it looks correct and all files are there.

If you make a new backup, please provide log file and after backup is done, `lsblk` to make sure boot partition is still mounted and provide output here.

I will make the error message even more descriptive so update with `git pull`, make sure you are on testing branch and re-run the script.

robertstrom, 2025-07-11:
Here's the end of the output when the script was run

```

Number of files: 478,998 (reg: 394,864, dir: 38,750, link: 45,384)
Number of created files: 478,995 (reg: 394,864, dir: 38,747, link: 45,384)
Number of deleted files: 401 (reg: 399, dir: 2)
Number of regular files transferred: 394,851
Total file size: 12.26G bytes
Total transferred file size: 12.25G bytes
Literal data: 12.25G bytes
Matched data: 0 bytes
File list size: 15.01M
File list generation time: 0.001 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 12.28G
Total bytes received: 7.89M

sent 12.28G bytes  received 7.89M bytes  17.25M bytes/sec
total size is 12.26G  speedup is 1.00
## Rsync done...
## Please stand by...
## Finalizing filesystem...
ROOTFS: /lost+found not found.  CREATED.

      478996 inodes used (53.64%, out of 892928)
         138 non-contiguous files (0.0%)
         339 non-contiguous directories (0.1%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 433863/50
     3358000 blocks used (94.04%, out of 3570960)
           0 bad blocks
           1 large file

      394851 regular files
       38750 directories
           0 character device files
           0 block device files
           0 fifos
          13 links
       45384 symbolic links (45074 fast symbolic links)
           0 sockets
------------
      478998 files
## Remounting for autoexpansion...
## Mounting img root partition...
## Mounting img boot partition...
## Enabling autoexpansion for Kali-ARM...
## Kali-ARM filesystem autoresizing at boot...
## Backup done.
#####################################################################################
# Backup location: /mnt/SMBMount/kali-rpi-case-2025-07-11.img                       #
# Write to logfile: true                                                            #
# Autoexpand filesystem at boot: true                                               #
# Use exclude.txt: false                                                            #
# Boot partition: true                                                              #
# Bootsector size: 255MiB                                                           #
# Estimated root usage: 12788MiB                                                    #
# Image size: 14205MiB with a root partition of 13949MiB                            #
#####################################################################################
## Exiting and cleaning up...
## Please stand by...
## Done.
## Elapsed time: 13.45

```

Nothing red, everything was green and purple and white

UnconnectedBedna, 2025-07-11:
Can you please check with the method I mentioned earlier? Loop img, mount boot and make sure it looks correct and all files are there.
You can compare your device boot to the mounted img boot with diff:
```
diff -r /boot/firmware /path/to/mounted_img_boot_partition
```
If it returns nothing it means they are the same, otherwise it will print differences and that means the boot partition on img is incorrect.

If you make a new backup, please provide log file and after backup is done, `lsblk` to make sure boot partition is still mounted and provide output here.

robertstrom, 2025-07-11:
I reran the backup and did a restore and mounted the BOOT partition on the MicroSD card and it was empty.

Here is the log file

```
#####################################################################################
2025-07-11 18:06:43 [INFO]    - Debugging requested, writing to log file ./shrink-backup/shrink-backup.log
2025-07-11 18:06:43 [INFO]    - Script started with: ./shrink-backup/shrink-backup -al /mnt/NFSMount/kali-rpi-case-2025-07-11.img
2025-07-11 18:06:43 [INFO]    - Zoom speed NOT requested, setting SLEEPING=sleep 1 | ZOOM=false
2025-07-11 18:06:44 [INFO]    - -a selected by user, setting ADDED_SPACE to 0 (non-zero value)
2025-07-11 18:06:44 [INFO]    - Separate boot partition detected
2025-07-11 18:06:44 [DEBUG] - BOOT_PATH=/boot/firmware
2025-07-11 18:06:44 [INFO]    - ext4 root filesystem detected
2025-07-11 18:06:44 [DEBUG] - FSTYPE=ext4
2025-07-11 18:06:44 [DEBUG] - LOCAL_DEV_PTUUID=5a444a2f | LOCAL_DEV_PATH=/dev/mmcblk0
2025-07-11 18:06:44 [INFO]    - Kali-ARM detected
-------------------------------------------------------------------------------------
2025-07-11 18:06:45 [DEBUG] - INSTALL_METHOD=default
2025-07-11 18:06:45 [DEBUG] - OS=kali
2025-07-11 18:06:45 [DEBUG] - IMG_FILE=/mnt/NFSMount/kali-rpi-case-2025-07-11.img
2025-07-11 18:06:45 [DEBUG] - PARTITION_TABLE=msdos
2025-07-11 18:06:45 [DEBUG] - UPDATE=false
2025-07-11 18:06:45 [DEBUG] - AUTORESIZE_RUN=true
2025-07-11 18:06:45 [DEBUG] - PROMPTS=true
2025-07-11 18:06:45 [DEBUG] - RSYNC_TTY=true
2025-07-11 18:06:45 [DEBUG] - EXCLUDE_FILE=false
2025-07-11 18:06:45 [DEBUG] - AUTOEXPAND=true
2025-07-11 18:06:45 [DEBUG] - RSYNC_DELETE=--delete
2025-07-11 18:06:45 [DEBUG] - F2FS_CONVERSION=false
2025-07-11 18:06:45 [DEBUG] - TTY_AVAILABILITY=/dev/tty
2025-07-11 18:06:45 [DEBUG] - ADDED_SPACE=0
-------------------------------------------------------------------------------------
2025-07-11 18:06:45 [INFO]    - -f NOT selected by user, using default exclude directories
2025-07-11 18:06:45 [INFO]    - Running function: check_dependencies
2025-07-11 18:06:45 [INFO]    - Running function: make_img
-------------------------------------------------------------------------------------
2025-07-11 18:06:45 [INFO]    - Running function: dev_variables
2025-07-11 18:06:45 [DEBUG] - BOOT_PARTITION=true | LOCAL_DEV_BOOT_PATH=/dev/mmcblk0p1 | LOCAL_DEV_ROOT_PATH=/dev/mmcblk0p2
2025-07-11 18:06:45 [DEBUG] - LOCAL_BOOT_UUID=C6CB-422C | LOCAL_ROOT_UUID=17ebfac2-5412-46d3-9cfd-4316d8f97e9f | LOCAL_ROOT_PARTUUID=5a444a2f-02
2025-07-11 18:06:45 [DEBUG] - LOCAL_ROOT_PARTN=2
2025-07-11 18:06:45 [INFO]    - Calculating size for dd to cover bootsector and adding 5MiB (5242880 bytes) to overlap into root (only used in img creation)
2025-07-11 18:06:45 [DEBUG] - LOCAL_ROOT_START=268435456 bytes | LOCAL_BOOTSECTOR=268434944 bytes | LOCAL_DDBOOTSECTOR=534527 512B blocks
2025-07-11 18:06:45 [INFO]    - Calculating recommended root size
2025-07-11 18:06:45 [DEBUG] - BLOCKSIZE=4096 bytes
2025-07-11 18:06:45 [INFO]    - ext4 filesystem detected, using resize2fs to set recommended root size
2025-07-11 18:06:45 [DEBUG] - LOCAL_AUTORESIZE_MIN=30385135616 bytes
2025-07-11 18:06:45 [INFO]    - Setting TOTAL (space needed for files on root) to autocalculated size
2025-07-11 18:06:45 [DEBUG] - TOTAL=30385135616 bytes
2025-07-11 18:06:45 [INFO]    - Calculating .img file size by adding LOCAL_BOOTSECTOR to TOTAL
2025-07-11 18:06:45 [DEBUG] - TRUNCATE_TOTAL=30653570560 bytes
2025-07-11 18:06:45 [INFO]    - Do you want to continue? [y/n]
2025-07-11 18:06:48 [INFO]    - Y or y pressed to confirm
-------------------------------------------------------------------------------------
2025-07-11 18:06:49 [INFO]    - Using dd to create bootsector
2025-07-11 18:06:49 [DEBUG] - Running: dd bs=512 count=534527 if=/dev/mmcblk0 of=/mnt/NFSMount/kali-rpi-case-2025-07-11.img conv=noerror,sync status=progress
-------------------------------------------------------------------------------------
2025-07-11 18:06:59 [DEBUG] - dd completed:
534527+0 records in
534527+0 records out
273677824 bytes (274 MB, 261 MiB) copied, 9.37449 s, 29.2 MB/s
-------------------------------------------------------------------------------------
2025-07-11 18:07:01 [INFO]    - Using truncate to resize img file to 29233MiB
2025-07-11 18:07:01 [DEBUG] - Running: truncate --size=30653570560 /mnt/NFSMount/kali-rpi-case-2025-07-11.img
2025-07-11 18:07:02 [INFO]    - Running function: do_loop
2025-07-11 18:07:02 [DEBUG] - LOOP=/dev/loop0
2025-07-11 18:07:03 [DEBUG] - Running: losetup -P /dev/loop0 /mnt/NFSMount/kali-rpi-case-2025-07-11.img
2025-07-11 18:07:04 [INFO]    - Using sfdisk to remove root partition
2025-07-11 18:07:04 [DEBUG] - Running: sfdisk --delete -f /dev/loop0 2
2025-07-11 18:07:06 [INFO]    - Using parted to recreate root partition
2025-07-11 18:07:06 [DEBUG] - Running: parted -s -a none /dev/loop0 unit B mkpart primary 268435456 100%
[... 101 lines cut ...]
2025-07-11 18:32:14 [DEBUG] - /mnt/NFSMount/kali-rpi-case-2025-07-11.img is 29233MiB with a root partition of 28977MiB
2025-07-11 18:32:14 [INFO]    - Backup done
-------------------------------------------------------------------------------------
2025-07-11 18:32:14 [DEBUG] - Cleanup function called with exit 0
2025-07-11 18:32:14 [DEBUG] - Unmounting boot partition in cleanup function: umount /tmp/backup-Lzm/boot/firmware
2025-07-11 18:32:14 [DEBUG] - Unmounting root partition in cleanup function: umount /tmp/backup-Lzm
[... cut ...]
2025-07-11 18:32:14 [DEBUG] - Exiting script
#####################################################################################

```

**NOTE:** when I went to do another backup I got this message

```
┌──(root㉿kali-rpi5-case)-[/home/rstrom]
└─# ./shrink-backup/shrink-backup -al /mnt/NFSMount/kali-rpi-case-2025-07-11.img
## Debugging requested, writing to log file: ./shrink-backup/shrink-backup.log
## Zoom speed NOT requested...
!! Boot found in fstab but partition not mounted...
## Do you want to mount boot partition? [y/n] 
```

UnconnectedBedna, 2025-07-11:
OMFG!!! 🤦

I entered the WRONG FKN VARIABLE in the script when checking if boot is mounted after backup is done... 🤦🤦🤦

Still don't know WHY your boot becomes umnounted so often, it does not happen for me THAT often, _only_ every 4-5 time or so, but for you it seems to happen constantly.
And again, it's not shrink-backup doing it, it's something shrink-backup is doing so boot becomes unmounted though, but it's NOT an umount command...
Really frustrating.

I unmounted the boot partition while script was running and it now gives the warning correctly.
Update shrink-backup and retry, sorry for my mistake.

robertstrom, 2025-07-12:
> OMFG!!! 🤦

> I entered the WRONG FKN VARIABLE in the script when checking if boot is mounted after backup is done... 🤦🤦🤦

LMAO !!

> Still don't know WHY your boot becomes umnounted so often

It appears that it is getting unmounted every time. I've tested the updated `shrink-backup` script several times now (at least three times and restored more than that). Every time that it gets to the end it tells me that the `Boot partition (/dev/loop0p1) became unmounted during backup`

Then I run `sudo ./shrink-backup/shrink-backup -Ul /mnt/NFSMount/kali-rpi-case-2025-07-11.img` and it is fixed / the script completes without any errors and the restore is successful.

I have successfully restored to MicroSD card and multiple different USB thumb drives. Partition expansion has been successful in every case.

I have also done an apt update / apt upgrade on the Kali install and then run `sudo ./shrink-backup/shrink-backup -alU /mnt/NFSMount/kali-rpi-case-2025-07-11.img` and then had to rerun `sudo ./shrink-backup/shrink-backup -Ul /mnt/NFSMount/kali-rpi-case-2025-07-11.img` to fix the image, but the updating of the image was successful and the restore after that was successful.

YEAH!!!

Thanks so much for your persistence in resolving this with me! It's not as clean as either of us would like but it gets the job done with the tiniest bit of extra effort.

I've been saving the image to my NAS (QNAP). Network is GB wired for both the RPi and the NAS. I have tried mounting the NAS via both SMB and NFS and had the same results in both scenarios. I am going to test saving the image to another USB drive that is attached directly to the RPi and see if the issue still exists. I will report back the results.

Thanks again!!

robertstrom, 2025-07-14:
Got it / testing

Curious .... what is the reason for getting rid of the second check for the mounting of the boot partition? I thought it worked great. If the partition is mounted no one will see it, if it's not mounted then it's extremely helpful. What is the harm in having the additional check?

Thanks!

UnconnectedBedna, 2025-07-14:
> what is the reason for getting rid of the second check for the mounting of the boot partition?

Why check if _X_ happened if there are safeguards making sure _X_ can never happen?
I didn't remove the first check, I removed the second after implementing a "safeguard" it can not become unmonted in the first place, why clutter up with extra code not needed?

I made a special fork and asked you to test to _make sure_ the safeguard is working properly as intended.
If the new method does not work, I remove the fork and keep the old code as is.

robertstrom, 2025-07-15:
K, I have tested a few times and it appears to be working. I will do some more testing and report back to you in a few days

UnconnectedBedna, 2025-07-15:
Thank you.
I will make some changes and then implement this into testing branch.

UnconnectedBedna, 2025-07-17:
Thank you so much for your dedication to QA!

closed 2025-08-02 by UnconnectedBedna