UnconnectedBedna/shrink-backup #68: ubuntu linux server arm

robertstrom, 2025-07-24:
**Is your feature request related to a problem? Please describe.**<br>

I currently have one RPi running Ubuntu 24.04.2 LTS and I have gotten around to using shrink-backup to create a backup of the RPi configuration.

I ran shrink-backup and was told

```
!! Boot found in fstab but partition not mounted, please mount and retry script!!!
```

I had recently updated but not rebooted so I rebooted and then applied some more patches that were needed and then rebooted again and ran shrink-backup and got the same message.

I then looked that the supported operating systems and noticed that Ubuntu was not on the list.

This is the result of running the `mount`  command

```
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=1643644k,nr_inodes=410911,mode=755,inode64)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=387420k,mode=755,inode64)
/dev/sda2 on / type ext4 (rw,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev,inode64)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k,inode64)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=32,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=1788)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,nosuid,nodev,relatime,pagesize=2M)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,nr_inodes=1048576,inode64)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
/var/lib/snapd/snaps/snapd_24724.snap on /snap/snapd/24724 type squashfs (ro,nodev,relatime,errors=continue,threads=single,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/snapd_24787.snap on /snap/snapd/24787 type squashfs (ro,nodev,relatime,errors=continue,threads=single,x-gdu.hide,x-gvfs-hide)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
tmpfs on /run/user/1001 type tmpfs (rw,nosuid,nodev,relatime,size=387416k,nr_inodes=96854,mode=700,uid=1001,gid=1001,inode64)
```

This is the result of running the `lsblk` command

```
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0    7:0    0  44.3M  1 loop /snap/snapd/24724
loop1    7:1    0  42.9M  1 loop /snap/snapd/24787
sda      8:0    0 119.2G  0 disk 
├─sda1   8:1    0   512M  0 part 
└─sda2   8:2    0 118.7G  0 part  /
```

This is the log file

```
2025-07-24 03:46:50 [INFO]  - Debugging requested, writing to log file ./shrink-backup/shrink-backup.log
2025-07-24 03:46:50 [INFO]  - Script started with: ./shrink-backup/shrink-backup -al /mnt/SMBMount/pihole.img
2025-07-24 03:46:50 [INFO]  - Zoom speed NOT requested, setting SLEEPING=sleep 1 | ZOOM=false
2025-07-24 03:46:51 [INFO]  - -a selected by user, setting ADDED_SPACE to 0 (non-zero value)
2025-07-24 03:46:51 [DEBUG] - ADDED_SPACE=0
2025-07-24 03:46:51 [INFO]  - ext4 root filesystem detected
2025-07-24 03:46:51 [DEBUG] - FSTYPE=ext4
2025-07-24 03:46:51 [DEBUG] - LOCAL_DEV_PTUUID=fccefbd8 | LOCAL_DEV_PATH=/dev/sda
------------------------------------------------------------------------------
2025-07-24 03:46:51 [DEBUG] - INSTALL_METHOD=default
2025-07-24 03:46:51 [DEBUG] - IMG_FILE=/mnt/SMBMount/pihole.img
2025-07-24 03:46:51 [DEBUG] - PARTITION_TABLE=msdos
[... 173 lines cut ...]
2025-07-24 03:58:18 [DEBUG] - Running: mount /dev/loop2p2 /tmp/backup-W0L
2025-07-24 03:58:19 [ERROR] - Boot found in fstab but partition not mounted, exit 1
2025-07-24 03:58:19 [ERROR] - Cleanup function called with non zero exit code: exit 1
2025-07-24 03:58:19 [DEBUG] - Unmounting root partition in cleanup function: umount /tmp/backup-W0L
2025-07-24 03:58:19 [DEBUG] - Removing loop in cleanup function: losetup -d /dev/loop2
2025-07-24 03:58:19 [DEBUG] - Removing temp directory in cleanup function: rm -rf /tmp/backup-W0L
2025-07-24 03:58:19 [INFO]  - Elapsed time: 00.39
2025-07-24 03:58:19 [DEBUG] - Exiting script
##############################################################################
```

**Describe the solution you'd like**<br>

It would be nice if shrink-backup would also work with Ubuntu ARM. As you know, I am totally willing to work with you to test and troubleshoot.

Let me know if you need more information.

Your consideration of this feature enhancement would be greatly appreciated. Thanks!

**Describe alternatives you've considered**<br>

Shutting the system down and using dd to backup the media and PiShrink to shrink the image

**Additional context**<br>
Add any other context or screenshots about the feature request here.

UnconnectedBedna, 2025-07-24:
This issue has already been fixed on the testing branch but not been released to "stable" yet. Please see #64 for more information.

As for the request to test it on ubuntu-ARM, I made this issue into a feature request and will look into it.

Since you are using snaps, you probably want to exclude those files, witch is also done by default on the testing branch so you won't need to do anything about it.
To test for yourself (I honestly think it already works with the fix), [switch to testing branch](https://github.com/UnconnectedBedna/shrink-backup/wiki/Installing#switch-branch) and retry shrink-backup.

robertstrom, 2025-07-24:
I did actually want the auto expansion. What I found odd was that at the end of the backup the script gave me the message 

```
!! No autoexpand option available for this OS!
```

So I did not expect the expansion, but I booted the system and checked the available space and I was pleased to see that the auto expansion did happen.

I did do a little research about the way that auto expansion worked for Ubuntu ARM and found exactly what you posted above about the `/etc/cloud/cloud.cfg`

I have never not wanted auto expansion so I have not tested that at all.

So the only _**issue**_ (that is not really a big issue) is the misleading message abot no autoexpansion when it really is happening.

One other thing ... shrink-backup also says that the OS is not recognized.

UnconnectedBedna, 2025-07-24:
Not recognized = not tested/supported.
But I will look into it. I also want to see if shrink-backup can work on ubuntu-core-arm, but I think that is encrypted ootb, but I want to make sure.
I also have to test it for the latest version, so if all goes well it will be included in next release.

commit 7c4544b4 (2025-07-31): Ubuntu-server-arm

Resolves #68 
Also added information about `--fix` in window before starting backup.
closed 2025-08-02 by UnconnectedBedna